sidoh/esp8266_milight_hub #6: Pipeline Issues with python-limitlessled

aneisch, 2017-02-19:
An issue exists where pipelines are used, brightness/color change does not function as expected. Instructions seem to be dropped when concurrent operations are called. This issue exists when triggering scenes from Home Assistant.

sidoh, 2017-02-19:
Any theories as to why this might be happening? I'm not very familiar with the library HA uses. Is it possible that it's grouping several commands in a packet?

aneisch, 2017-02-19:
I don't have any ideas yet. I'll do some captures and reading into the
library to see what's going on.

aneisch, 2017-02-19:
The expectation is that group one would be brought to 92 brightness, color 255,255,255, and group two would be brought to 255 brightness, color 255,217,201.

[out.pcap.zip](https://github.com/sidoh/esp8266_milight_hub/files/786188/out.pcap.zip)

I don't know how to decode the PCAP but am looking into the protocol.

sidoh, 2017-02-19:
Okay certainly seems like it's just sending a bunch of packets of length 2. It probably should work.

From looking at the pcap, it's not sending any color packets. Color packets start with `0x40`. These all seem to be turning on group 1/2 and setting the brightness.

aneisch, 2017-02-19:
Amended comment to add sniffed RF.

aneisch, 2017-02-19:
Oh, and I apologize, I misread my config. This particular scene has both groups at 255,255,255 color. I assume we don't see color changes because we set them both to white.

sidoh, 2017-02-19:
Got it, makes sense. Does it consistently miss the same actions, or is it kind of random?

aneisch, 2017-02-19:
Seems to be kind of random. Sometimes one group will go to 100%, others it will stay low. When switching between this "full" scene, and a "dim" scene (which modifies the color AND brightness of groups) the colors do not change properly, seemingly at random.

sidoh, 2017-02-19:
Alright, that makes sense. I guess maybe some kind of ordering issue? Could you perhaps add some `Serial.println` debugging to `MiLightUdpServer#handleCommand` to ensure the server is receiving the commands?

sidoh, 2017-02-20:
Just noticed HASS has a docker image, which makes it super easy for me to test stuff with HA.

I'm noticing, as you did, that sometimes commands seem to be dropped. 

I'll poke around.

aneisch, 2017-02-20:
It does! The configuration for limitless is very basic, you'll only need a few lines in your config. Thanks!

sidoh, 2017-02-20:
How many groups are you controlling at a time? I'm noticing that it seems to mess up more when dealing with multiple groups.

aneisch, 2017-02-20:
I currently have two groups in use. The scenes in HA are configured to alter the state of both groups. How it chunks/sends the commands I am unsure. Hoping to get some logging put in today.

sidoh, 2017-02-21:
Unfortunately it seems like this is because the ESP isn't receiving some of the UDP packets. Even if I disable everything in the UDP server handler except serial printing, I'm seeing a drop rate of 20-30%. I stumbled across [at least one](https://circuitmaker.com/blog/networked-lighting-part-2-the-esp8266-experience) other report of this being the case.

I was able to reproduce this with `nc`:

```
$ cat /tmp/a | tr '\n' '\0' | xargs -n 1 -0 -I {} bash -c 'echo -n -e "{}" | nc -w 0 -u esp 8899 && sleep 0.025'
```

( HA seems to wait approximately 25ms between commands ).

I'm guessing this is because HA is sending quite a few UDP packets (20-30) as quickly as it can, and it's overwhelming the ESP. Obviously since there's no acknowledgement of receipt in UDP, a dropped packet isn't going to be resent. 

Increasing the sleep time to 50ms seems to make it work.

I'm pretty sure decreasing the number of resends will fix this. I had it set pretty high because it seemed like it was necessary for the HTTP API to work. Now that I'm mucking around in the UDP protocol more, I realize that UDP packets are sent 3 times, which causes more resends.

If I decrease the number of resends to 10 (from 50), a 25ms sleep works.

Probably the right fix is to:

1. Decrease the number of resends
2. Have HTTP API increase it or re-call the same command a few times.

sidoh, 2017-02-21:
Merged some changes. Could you try again?

aneisch, 2017-02-21:
I'd be glad to, but won't have the opportunity until late tonight or tomorrow.

aneisch, 2017-02-22:
Finally got around to flashing this and it's much improved. I'd say about 90% there. The only issues I still notice are:

1. The lights do not always respond properly at first (probably attributed to your findings above)
2. If lights are instructed to change colors, they are consistently changed to an incorrect one.

sidoh, 2017-02-23:
Glad to hear it's improved. Thanks for testing.

Could you share your HomeAssistant scene config so I can try to reproduce/test/iterate? 

By at first do you mean they're less responsive when changing from off to on? I noticed this too.

Might need to increase the number of resends a bit. I could make it configurable without flashing the firmware too.

re: color being wrong - interesting. The UDP protocol's byte for hue is shifted from the byte in the 2.4 GHz protocol, but I thought I'd accounted for that. I'll test with HA later today.

aneisch, 2017-02-23:
I've pulled out the relevant stuff from my config. Full config can be found at https://github.com/aneisch/home-assistant if anything else is needed. 

[hass_configs.zip](https://github.com/sidoh/esp8266_milight_hub/files/795198/hass_configs.zip)

Yes, that's it exactly. And like I said, it only seems to be an issue [the color] when changing from UDP. I haven't tested from REST because I'm not sure what color value I can use to map properly. If you click on the light in Hass, a window will pop up and allow you to select colors (FYI).

sidoh, 2017-02-23:
Perfect. Thanks very much! I'll dig in.

sidoh, 2017-02-23:
Fixed the colors, I think. I had the right offset between the protocols, but they also flipped the spectrums (went R->B->G in UDP as opposed to R->G->B in the 2.4 GHz protocol).

I'm hopeful that increasing the number of repeated 2.4 GHz packets will help. I'll try to find a number of repeats that's a better balance between UDP reliability and 2.4 GHz reliability. I'd also like to expose a setting that allows the user to tune it for themselves.

aneisch, 2017-02-24:
Let's call this closed. I can't make it _not_ work now. Impressive work. Very impressive! Thank you!

closed 2017-02-24 by aneisch

sidoh, 2017-02-24:
Happy to hear it! Obviously let me know if you encounter other problems :)

aneisch, 2017-03-01:
Group 1 seems to be uncontrollable from HA now, but group 2 remains responsive. There's certainly some weirdness here, and I don't know that HA isn't to blame. My web portal (on the D1) seems to also no longer function (I can access, but any commands sent have no effect on any groups). Also getting weird returns from curl. I think I'm going to re-flash.

```
[aneisch@nuc-nuc]$ curl -vvv -X PUT --data-binary '{"status":"off","hue":100,"level":50}' -X PUT http://10.0.1.47/gateways/0x1/2
* About to connect() to 10.0.1.47 port 80 (#0)
*   Trying 10.0.1.47...
* Connected to 10.0.1.47 (10.0.1.47) port 80 (#0)
> PUT /gateways/0x1/2 HTTP/1.1
> User-Agent: curl/7.29.0
> Host: 10.0.1.47
> Accept: */*
> Content-Length: 37
> Content-Type: application/x-www-form-urlencoded
> 
* upload completely sent off: 37 out of 37 bytes
< HTTP/1.1 400 Bad Request
< Content-Type: text/plain
< Content-Length: 22
< Connection: close
< Access-Control-Allow-Origin: *
< 
* Closing connection 0
Unknown device type: 2
```

sidoh, 2017-03-01:
Bah.

I changed the routes to better support the different types of bulbs (which have different radio configs). It should be:

```
curl -vvv -X PUT --data-binary '{"status":"off","hue":100,"level":50}' -X PUT http://10.0.1.47/gateways/0x1/rgbw/2
```

Sorry for the sloppy release. Iterating kinda quickly. :)

Super bizarre that a single group would be unresponsive. Do they belong to the same UDP server? Can you verify things in the UI (might need to re-upload it) work as expected?

aneisch, 2017-03-01:
Ha, no apology needed! So the curl command shut off group 1. Going to re-up the UI.

sidoh, 2017-03-01:
Sweet.

So one interesting thing I noticed while playing around more with HA is that it overwrote any settings for `port` or `version` you specified in the config.

I spent about an hour scratching my head and digging around in HA/limitlessled source (my Python is awful), but figured it out only to realize it was fixed in the dev branch already. Not sure if it's in master already, but here's the fix:

https://github.com/home-assistant/home-assistant/commit/58e707a264c47e3b1e3191ba11e28f7a4c78feca#diff-b25001516ea483503dbc3147514f5344

aneisch, 2017-03-01:
Ah, good news. Re-uploading index seemed to fix it. Also, I see new packet_repeats and http_repeat_factor options, any recommendation for numbers?

sidoh, 2017-03-01:
I use 10 for repeats, and 5 for http repeat factor and it works really well. Did they not default to something?

aneisch, 2017-03-01:
Doesn't look like it, but maybe if I'd have reset the Arduino.

aneisch, 2017-03-01:
Well, it looks like I'm a few pushes behind, I hadn't noticed there were changes made in the last few days. I'll be flashing new firmware this evening.

sidoh, 2017-03-02:
Did you end up figuring out what was going on with HA? The web UI shouldn't affect that since it's basically just a consumer of the HTTP API.

aneisch, 2017-03-02:
Not sure if it was just a transient issue, or a bug that was fixed in one our your recent pushes, but everything seems stable now. I'll re-open this if that changes.

aneisch, 2017-03-02:
Huh, and just like that, same behavior. Group 2 controllable through HASS, Group 1 unresponsive. Both groups respond from the emulator's dashboard.

Looks like this is actually a limitlessled pipeline issue. The pipeline for group 1 (andrew_bedroom) never seems to be finish.
```
17-03-01 23:02:24 INFO (MainThread) [homeassistant.loader] Loaded light.limitlessled from homeassistant.components.light.limitlessled
17-03-01 23:02:25 INFO (MainThread) [homeassistant.components.light] Setting up light.limitlessled
17-03-01 23:02:39 INFO (Thread-14) [limitlessled.pipeline] Starting a new pipeline on group 1 (andrew_bedroom) @ 10.0.1.47
17-03-01 23:02:39 INFO (Thread-14) [limitlessled.pipeline]  -> Running stage 'on()' on group 1 (andrew_bedroom) @ 10.0.1.47
17-03-01 23:02:39 INFO (Thread-15) [limitlessled.pipeline] Starting a new pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:39 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'on()' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:39 INFO (Thread-14) [limitlessled.pipeline]  -> Running stage 'white()' on group 1 (andrew_bedroom) @ 10.0.1.47
17-03-01 23:02:39 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'transition(0, color=Color(R=255, G=217, B=201), brightness=1.0)' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:40 INFO (Thread-14) [limitlessled.pipeline]  -> Running stage 'transition(3, color=Color(R=255, G=255, B=255), brightness=0.0784313725490196)' on group 1 (andrew_bedroom) @ 10.0.1.47
17-03-01 23:02:41 INFO (Thread-15) [limitlessled.pipeline] Finished pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:42 INFO (Thread-15) [limitlessled.pipeline] Starting a new pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:42 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'white()' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:43 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'transition(0, color=Color(R=255, G=255, B=255), brightness=1.0)' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:44 INFO (Thread-15) [limitlessled.pipeline] Finished pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:46 INFO (Thread-15) [limitlessled.pipeline] Starting a new pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:46 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'transition(0, color=Color(R=255, G=217, B=201), brightness=1.0)' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:02:47 INFO (Thread-15) [limitlessled.pipeline] Finished pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:04:30 INFO (Thread-15) [limitlessled.pipeline] Starting a new pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:04:30 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'transition(0, brightness=0.0)' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:04:31 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'off()' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:04:31 INFO (Thread-15) [limitlessled.pipeline] Finished pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:04:35 INFO (Thread-15) [limitlessled.pipeline] Starting a new pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:04:35 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'on()' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:04:36 INFO (Thread-15) [limitlessled.pipeline]  -> Running stage 'transition(0, color=Color(R=255, G=217, B=201), brightness=1.0)' on group 2 (andrew_monitor_backlight) @ 10.0.1.47
17-03-01 23:04:37 INFO (Thread-15) [limitlessled.pipeline] Finished pipeline on group 2 (andrew_monitor_backlight) @ 10.0.1.47
```

aneisch, 2017-03-02:
```
[aneisch@nuc-nuc ~]$ grep "group 1" hass.log 
17-03-01 23:02:39 INFO (Thread-14) [limitlessled.pipeline] Starting a new pipeline on group 1 (andrew_bedroom) @ 10.0.1.47
17-03-01 23:02:39 INFO (Thread-14) [limitlessled.pipeline]  -> Running stage 'on()' on group 1 (andrew_bedroom) @ 10.0.1.47
17-03-01 23:02:39 INFO (Thread-14) [limitlessled.pipeline]  -> Running stage 'white()' on group 1 (andrew_bedroom) @ 10.0.1.47
17-03-01 23:02:40 INFO (Thread-14) [limitlessled.pipeline]  -> Running stage 'transition(3, color=Color(R=255, G=255, B=255), brightness=0.0784313725490196)' on group 1 (andrew_bedroom) @ 10.0.1.47
[aneisch@nuc-nuc ~]$ date
Wed Mar  1 23:08:04 CST 2017
```

sidoh, 2017-03-02:
Aah, tricky. Good find.

Phew!