sidoh/esp8266_milight_hub #12: failed upload index.html

turuwhiska, 2017-03-10:
i have thi error when i try upload de ui Error updating web file
i use the command curl -vvv -X POST -F 'image=@web/index.html' http://<ip_of_esp>/web

sidoh, 2017-03-10:
Are you replacing `<ip_of_esp>` with an IP?

sidoh, 2017-03-10:
Looks like this is you:

http://blog.christophermullins.com/2017/02/11/milight-wifi-gateway-emulator-on-an-esp8266/#comment-1136

so sounds like you are.

The logs you included from the Serial console make it sound like your ESP has different flash sizes than the firmware you pushed to it. Are you using one of the pre-compiled firmware binaries? What ESP breakout model are you using?

turuwhiska, 2017-03-10:
yes i remplace  <ip_of_esp> , im using esp8266-12 , i tested the two pre-compilated files
its posible compile for esp-12?

turuwhiska, 2017-03-10:
ist posible use in esp-01? how can make? dont have a makefile thank so much for response from argentina

sidoh, 2017-03-10:
Should be possible to compile it for esp-12. Just add a new environment in `platformio.ini`. Here's a copy from another project where I used an esp-12f:

```
[env:esp12f]
platform = espressif8266
board = esp12e
framework = arduino
build_flags = -Wl,-Tesp8266.flash.4m1m.ld
```

I needed to specify the custom `build_flags`. I think the platformio defaults were wrong, or at least didn't work with the device I had. YMMV.

Don't think this will work with esp-01. The nrf24l01 is an SPI device, and the esp-01 doesn't expose the SPI GPIOs (12-15, esp-01 only exposes GPIO 0/2).

turuwhiska, 2017-03-10:
ah perfect i will test how you compile? make its not posible?

sidoh, 2017-03-10:
Probably easiest to use [platformio](http://platformio.org/). Basically just open up the project, add the section to `platformio.ini`, and run compile :)

sidoh, 2017-03-11:
@turuwhiska were you able to get this working? I can upload an esp-12 binary if it'd help.

turuwhiska, 2017-03-11:
i dont test becouse dont have internet for download platformio. (the life in patogonia)  if you cau upload the bin for esp-12 perfect!!!, thank so much

turuwhiska, 2017-03-11:
Super , if you can upload the Bin file foro esp-12 perfect . I dont have internet foro Download platformio . The Life in Patagonia jaja

turuwhiska, 2017-03-12:
can you upload the bin file?

terrag42, 2017-03-12:
@sidoh I tried to build from source and got the error:
`src\main.cpp:2:25: fatal error: WiFiManager.h: No such file or directory`
Should the WiFiManager library be included in the "libs" folder?

turuwhiska, 2017-03-14:
i installed atom and platformio but i can install the board sdk. internet in patagonia argentina its a probtlem,
if you can upload the file i thank you so much

sidoh, 2017-03-18:
Sorry for the delay. Here it is:

https://github.com/sidoh/esp8266_milight_hub/blob/master/dist/firmware-esp12.bin

turuwhiska, 2017-03-18:
thanks so much . i tested the new .bin and had the same problem
*WM: AutoConnect                                                                
*WM: Connecting as wifi client...                                               
*WM: Using last saved values, should be faster                                  
*WM: Connection result:                                                         
*WM: 3                                                                          
*WM: IP Address:                                                                
*WM: 192.168.43.152                                                             
Opening settings file failed 

and when i try upload de webui 

Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file

turuwhiska, 2017-03-18:
and i used sudo ./esptool.py --port /dev/ttyUSB0 erase_flash

sidoh, 2017-03-18:
Hmm... are you able to link to the device you bought?

turuwhiska, 2017-03-18:
What would that be?

sidoh, 2017-03-18:
I mean the ESP. Can you link to where you got it from?

All I can figure is that the flash sizes are different from the one I have.

turuwhiska, 2017-03-18:
sure. Is it possible to add the ui in the bin?

sidoh, 2017-03-18:
Yeah - definitely something I want to do. That's the idea behind #2, although the approach I was planning on taking here wouldn't solve the issue you're having (seems like the bin is still not compatible with your device).

I'll test with an esp12 myself soon. Trying to get the new RGB+CCT protocol finished.

turuwhiska, 2017-03-18:
ok thank so much!!!

sidoh, 2017-03-18:
Can you show me the esp12 you have? Link to where you bought it from or maybe take a picture?

turuwhiska, 2017-03-18:
im sorry is the model esp8266-07
http://www.prometec.net/producto/esp8266-esp-07/

turuwhiska, 2017-03-18:
its posible use in esp-07?

fireboy1919, 2017-03-20:
BTW, this also didn't work for me until I did this:
curl -vvv -X POST -F 'image=@web/index.html' http://<ip_of_esp>/web -u "":""

I was getting unauthorized errors.

sidoh, 2017-03-20:
Here's firmware for the esp07:

https://github.com/sidoh/esp8266_milight_hub/blob/master/dist/firmware-esp07.bin

@fireboy1919, are you running with the latest version? I thought I fixed that.

turuwhiska, 2017-03-20:
Thanks i Will test now

https://github.com/sidoh/esp8266_milight_hub/blob/master/dist/firmware-esp07.bin

turuwhiska, 2017-03-20:
i tested the new bin for esp07
the same error :
*WM: Connecting as wifi client...                                               
*WM: Using last saved values, should be faster                                  
*WM: Connection result:                                                         
*WM: 3                                                                          
*WM: IP Address:                                                                
*WM: 192.168.43.152                                                             
Opening settings file failed                                                    
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file                                                         
Error updating web file

sidoh, 2017-03-20:
I tested with a fresh esp07 and it worked just fine. How are you flashing the firmware? Can you paste the command you're using? I used this:

```
python /opt/esptool/esptool.py --port /dev/cu.SLAB_USBtoUART --baud 115200 write_flash -fm=dio -fs=1MB 0x00000 .pioenvs/esp07/firmware.bin
```

Note that you need to use `-fs=1MB` for esp07 instead of `-fs=4MB` because it only has 1MB of flash.

turuwhiska, 2017-03-20:
yes!!!!!!!! perfect thanks so much!!!!!work!!
close the inssue?

sidoh, 2017-03-20:
Awesome, glad that part's working. Sure, sounds like we can close.

closed 2017-03-20 by sidoh