sidoh/esp8266_milight_hub #14: Add support for RGB+CCT bulbs

sidoh, 2017-03-18:
I think I managed to reverse the protocol. Will need to add support for the new packet structure:

http://blog.christophermullins.com/2017/03/18/reverse-engineering-the-new-milightlimitlessled-2-4-ghz-protocol/

Lstt2005, 2017-03-20:
Dear friend! I was also try to investigate new MIlight B2 like this – [https://www.aliexpress.com/item/Milight-B2-4-Zone-CCT-Adjust-Smart-Panel-Remote-Controller-for-led-strip-light-lamp-or/32790035244.html?ws_ab_test=searchweb0_0,searchweb201602_6_10065_10068_433_434_10136_10137_10138_10060_10062_10141_10056_10140_10055_10054_10059_9911_10099_10103_10102_10096_120_10052_10053_10050_10107_10142_10051_10106_10084_10083_10119_10080_10082_10081_10110_10111_10112_10113_10114_10078_10079_10073_10070_10122_10123_10120_10124-9911_10120,searchweb201603_13,afswitch_1_afChannel,ppcSwitch_5,single_sort_0_default&btsid=857396d0-41cf-4863-80c5-29187dd5beda&algo_expid=d3e93619-973d-411e-bb60-417f9524e141-1&algo_pvid=d3e93619-973d-411e-bb60-417f9524e141](url)

But i was beginner in C++ (Arduino or Plaformino), can you rewrite you algo with unscramble on c+??
I also have a Salea Logic files with saved inital power on for remote control and some keys pressed..

sidoh, 2017-03-20:
Hard for me to tell what version of the protocol that remote uses by looking at it. It's possible it's using the older 7 byte protocol, which doesn't have any scrambling.

Are you able to convert the SPI dumps from your logic analyzer to the 7 byte / 9 byte protocol?

Lstt2005, 2017-03-20:
Can you help me, wich utility i can converted? This Remote uses scrambled 9 byte protocol, as you described early.

Lstt2005, 2017-03-20:
[image]
[image]
[image]
This is scrambled 9 bytes payload

sidoh, 2017-03-20:
Ah yeah, that's all you need. Just some of the 9-byte captures.

This decodes to:

```
$ echo '2D 81 1B 61 45 25 B4 47 FE' | ruby scripts/decode_packet.rb
2D 21 91 81 01 C0 9F 00 26
```

Seems to indicate you pressed (and held) the "ON" button (going off of the fact that `b[4] & 0xF == 0x01`, and `b[7] == 0`).

This is interesting because the value in index 1 changes from 0x20 to 0x21. Assuming this indicates remote type.

I'm already working on porting the code to C++ (in [this](https://github.com/sidoh/esp8266_milight_hub/tree/new_protocol) branch). It mostly works, but I suspect the remote you have uses a different channel set / syncwords. Configs for currently supported remotes are [here](https://github.com/sidoh/esp8266_milight_hub/blob/new_protocol/lib/MiLight/MiLightRadioConfig.h#L50).

Lstt2005, 2017-03-20:
Thanks, i will try later. This remotes also use the same channels like you describes and same syncwords uses 0x7236 , 0x1809..

Lstt2005, 2017-03-20:
Say, can you add support this remotes, as shown in picrures? And if Yes, i can help you in full investigate packets from remotes, as you wish..

sidoh, 2017-03-20:
Oh, looks like the little wifi box I have ([this one](http://www.ebay.com/itm/Mi-Light-2-4G-Milight-Wifi-Box-LED-strip-light-Controller-of-RGBW-Control-Syste-/262879423109?hash=item3d34d5aa85:g:8VIAAOSw4A5YuZEn)) supports the same protocol.

The differences here are:

1. `b[1]` is 0x21 instead of 0x20. I'm assuming that `b[1]` indicates remote type or something like that.
2. The command 0x02 is brightness. (It's color for the remote types I have).
3. The argument for brightness is scrambled using different additive offsets. Would need to scrape some packets to reverse these.

It appears that everything else is the same.

I'll make another ticket for this, but without having bulbs it'll be hard to test.

sidoh, 2017-03-20:
Created #16 for this.

sidoh, 2017-03-26:
Added with #22

closed 2017-03-26 by sidoh