sidoh/esp8266_milight_hub #15: Support official UDP discovery

fireboy1919, 2017-03-20:
From the Limitless LED page, you can discover a hub in the milight app like this:
- Step 1:Send UDP message to the LAN broadcast IP address of "10.10.100.255" and port of 48899 => "Link_Wi-Fi"
- All Wifi bridges on the LAN will respond with their details. Response is "10.10.100.254, ACCF232483E8"

- Step 2 (optional for changing settings on the wifi bridge): Then send "+ok" to the LimitlessLED Wifi Bridge. Send UDP message to the response IP address returned from step 1 "10.10.100.254" => "+ok"
- Step 3 (optional for changing settings on the wifi bridge): After that you may send AT commands (ending with \r\n) to the module.

-----

Obviously, there's no way to get the app to figure out how to work on ports other than the default, but essentially with this one addition, the esp8266 milight hub would be a superior drop-in replacement for the original hub.

sidoh, 2017-03-20:
Good idea. :)

sidoh, 2017-03-24:
So I poked around with this a little bit, and I think it's harder than I'd hoped. 

As far as I can tell, the only way for a hub to show up in the new version of the app (the one for v6 hubs) is to run through the installation procedure, which involves connecting to an AP broadcasted by the device, etc. It's not enough to respond to this broadcast packet. In fact, it doesn't look like the app even sends it except during the setup of a new hub.

Even if I have the ESP spoof the MAC address of the official hub there are some extra undocumented packets being sent that seem necessary for the app to consider itself "connected" with the hub.

I added support for:

1. UDP sessions / heartbeats
2. The discovery mechanism you mentioned
3. Most of the commands for the v6 bulbs

Lstt2005, 2017-03-26:
Chris! Can you explain, what mean: "You can add an arbitrary number of UDP gateways through the REST API or through the web UI. Each gateway server listens on a port and responds to the standard set of commands supported by the Milight protocol. This should allow you to use one of these with standard Milight integrations (SmartThings, Home Assistant, OpenHAB, etc.)."
What format commands will be send over UDP? Where we use saved on gateways ID's?

sidoh, 2017-03-26:
It implements parts of the protocol documented [here](http://www.limitlessled.com/dev/). It's meant to emulate [these things](http://futlight.com/productdetails.aspx?id=235&typeid=100).

sidoh, 2017-04-03:
Should be implemented in 1.1.0.

closed 2017-04-03 by sidoh