sidoh/esp8266_milight_hub #16: Add support for v2 CCT bulbs

sidoh, 2017-03-20:
Raised by @lstt2005 in #14. The protocol has some minor differences from the one used by the RGB+CCT remotes:

1. `b[1]` is 0x21 instead of 0x20. I'm assuming that b[1] indicates remote type or something like that.
2. The command 0x02 is brightness. (It's color for the remote types I have).
3. The argument for brightness is scrambled using different additive offsets. Would need to scrape some packets to reverse these.

Commands for on, off, and color temperature appear to use the same codes and RF configs.

Will need to:

1. Add a way to specify `b[1]`.
2. Reverse additive offsets for brightness command.
3. Add a new remote type in the UI / RF configs.

Lstt2005, 2017-03-20:
`
[image]

`

Lstt2005, 2017-03-20:
[FUT091_DECODE.xlsx](https://github.com/sidoh/esp8266_milight_hub/files/855984/FUT091_DECODE.xlsx)

sidoh, 2017-03-20:
Nice, thanks for sharing the packet captures.

I'm still iterating on the code for the RGB+CCT bulbs, so probably not safe to develop against yet. When I merge it in, though, I'd be happy to take a look at a PR.

Lstt2005, 2017-03-20:
Ok, say if I can help you, than I can..

Lstt2005, 2017-03-24:
I sniff my FUT081 remote with you New_protocol brach:
1. ON - 4 times
Packet received: AAA90B580D987A3A579E308D
Packet transformed: 09AD019BE1C5A59EC710
Failed CRC: expected 11686, got 50960
Radio is available
Packet received: AAA90B580D987A3A579E308D
Packet transformed: 09AD019BE1C5A59EC710
Failed CRC: expected 11686, got 50960

Radio is available
Packet received: AAA906EB430C06A14F5A823A
Packet transformed: 09762D0C035628AF15C4
Failed CRC: expected 58100, got 5572
Radio is available
Packet received: AAA906EB430C06A14F5A823A
Packet transformed: 09762D0C035628AF15C4
Failed CRC: expected 58100, got 5572
Radio is available
Packet received: AAA90F13A82D1C84F1614C45
Packet transformed: 098F5C418B13F2682823
Failed CRC: expected 44901, got 10275
Radio is available
Packet received: AAA90F13A82D1C84F1614C45
Packet transformed: 098F5C418B13F2682823
Failed CRC: expected 44901, got 10275
Radio is available
Packet received: AAA906BB336C66D1180AE633
Packet transformed: 09D6CD6C63B6880175C6
Failed CRC: expected 22276, got 30150
Radio is available
Packet received: AAA906BB336C66D1180AE633
Packet transformed: 09D6CD6C63B6880175C6
Failed CRC: expected 22276, got 30150
2. OFF 4 times
Packet received: AAA908EA2FBA5902767D0F94
Packet transformed: 097145DFA509E4E60B9F
Failed CRC: expected 64946, got 2975
Radio is available
Packet received: AAA908EA2FBA5902767D0F94
Packet transformed: 097145DFA509E4E60B9F
Failed CRC: expected 64946, got 2975

Radio is available
Packet received: AAA90B783DA84A11686E1792
Packet transformed: 09EDC15B21856861879E
Failed CRC: expected 61089, got 34718
Radio is available
Packet received: AAA90B783DA84A11686E1792
Packet transformed: 09EDC15B21856861879E
Failed CRC: expected 61089, got 34718
Radio is available
Packet received: AAA90E22591CADABC250E577
Packet transformed: 0947A489535B3DA470EA
Failed CRC: expected 22568, got 28906
Radio is available
Packet received: AAA90E22591CADABC250E577
Packet transformed: 0947A489535B3DA470EA
Failed CRC: expected 22568, got 28906
Radio is available
Packet received: AAA9085A8F1AF9B2D4DDBD99
Packet transformed: 09A1158FF5D9B4B2DB9B
Failed CRC: expected 41260, got 56219
Radio is available
Packet received: AAA9085A8F1AF9B2D4DDBD99
Packet transformed: 09A1158FF5D9B4B2DB9B
Failed CRC: expected 41260, got 56219

sidoh, 2017-03-24:
I'm pretty sure the wifi box I have can emulate the remote you have. The hardest part is I don't have bulbs to test with.

I also want to find a saner way to manage all of the different remote types before adding more of them.

It's pretty weird that it's failing CRC. Basically means the packets are getting thrown out. Mine doesn't do that. Are you listening on the rgb+cct channel?

Lstt2005, 2017-03-24:
No, it's sniff ONLY on CCT channel...

sidoh, 2017-03-24:
Oh, interesting. That's why it's failing CRC. The "cct" config is expecting the 7 byte protocol, so it's reading the CRC checksum at the wrong offset.

This is definitely the 9 byte 2.4 GHz protocol, though. I can decode the packets and see the values I expect:

```
$ echo '8F5C418B13F2682823' | ruby scripts/decode_packet.rb
8F 21 11 C0 01 00 07 00 9B

$ echo 'A1158FF5D9B4B2DB9B' | ruby scripts/decode_packet.rb
A1 21 11 C0 01 05 0D 00 5C
```

Lstt2005, 2017-03-24:
Yes, this remote send data lengt (via SPI sniff) 9 bytes:
[image]

Lstt2005, 2017-03-27:
Today is receive FutLight B2 - http://futlight.com/productdetails.aspx?id=263&typeid=100 . Interesting - when i sniff it, ID is changing - 
Packet received (9 bytes):
Raw packet: 0D A1 6A B0 A3 C6 38 A5 7D 

Decoded:
Key      : 0D
b1       : 21
ID       : 0271
Command  : 03
Argument : 47
Sequence : C3
Group    : 02
Checksum : 49

Packet received (9 bytes):
Raw packet: E8 F4 B7 E9 3C DA 50 3B 95 

Decoded:
Key      : E8
b1       : 21
ID       : 0275
Command  : 03
Argument : 4E
Sequence : C2
Group    : 02
Checksum : 3A

Packet received (9 bytes):
Raw packet: 49 6D 36 FC 6F 1C 86 71 E1 

Decoded:
Key      : 49
b1       : 21
ID       : 0271
Command  : 03
Argument : 55
Sequence : C1
Group    : 02
Checksum : 21

Packet received (9 bytes):
Raw packet: 53 A0 78 C2 59 0E E7 6E 60 

Decoded:
Key      : 53
b1       : 21
ID       : 027D
Command  : 03
Argument : 58
Sequence : C0
Group    : 02
Checksum : 19

Packet received (9 bytes):
Raw packet: 1E 85 35 2C 70 FE 60 2F 11 

Decoded:
Key      : 1E
b1       : 21
ID       : 0271
Command  : 03
Argument : 5E
Sequence : BF
Group    : 02
Checksum : 09

Packet received (9 bytes):
Raw packet: 82 29 19 50 54 4E FB 13 84 

Decoded:
Key      : 82
b1       : 21
ID       : 0271
Command  : 03
Argument : 6A
Sequence : BE
Group    : 02
Checksum : F8

Packet received (9 bytes):
Raw packet: 84 50 53 45 D8 75 9B D7 BF 

Decoded:
Key      : 84
b1       : 21
ID       : 027D
Command  : 03
Argument : 71
Sequence : BD
Group    : 02
Checksum : F4

Packet received (9 bytes):
Raw packet: 92 19 09 60 44 96 09 03 92 

Decoded:
Key      : 92
b1       : 21
ID       : 0271
Command  : 03
Argument : 82
Sequence : BC
Group    : 02
Checksum : FE

Packet received (9 bytes):
Raw packet: 5D 51 DA 40 13 73 80 15 92 

Decoded:
Key      : 5D
b1       : 21
ID       : 0271
Command  : 03
Argument : 82
Sequence : BB
Group    : 02
Checksum : EC

Packet received (9 bytes):
Raw packet: C6 DD CD A4 08 5A C3 C7 E0 

Decoded:
Key      : C6
b1       : 21
ID       : 0271
Command  : 03
Argument : 82
Sequence : BA
Group    : 02
Checksum : C0

Packet received (9 bytes):
Raw packet: 9B 58 F0 5A D2 C8 A6 E6 6A 

Decoded:
Key      : 9B
b1       : 21
ID       : 026D
Command  : 02
Argument : 2A
Sequence : B9
Group    : 02
Checksum : 5B

Packet received (9 bytes):
Raw packet: AF 3C D4 5E B6 B3 BB CA DA 

Decoded:
Key      : AF
b1       : 21
ID       : 0275
Command  : 02
Argument : 23
Sequence : B8
Group    : 02
Checksum : 37

Packet received (9 bytes):
Raw packet: 48 94 97 09 1B 7A DD 1B 81 

Decoded:
Key      : 48
b1       : 21
ID       : 0275
Command  : 02
Argument : 0E
Sequence : B7
Group    : 02
Checksum : CE

Packet received (9 bytes):
Raw packet: 4E 55 45 1C 7F 33 47 3F 6C 

Decoded:
Key      : 4E
b1       : 21
ID       : 0271
Command  : 02
Argument : 23
Sequence : B6
Group    : 02
Checksum : D4

Packet received (9 bytes):
Raw packet: 20 BC 7F 11 03 92 E7 03 63 

Decoded:
Key      : 20
b1       : 21
ID       : 0275
Command  : 02
Argument : 2E
Sequence : B5
Group    : 02
Checksum : D4

Packet received (9 bytes):
Raw packet: 86 1D 0D 64 47 58 0D 07 57 

Decoded:
Key      : 86
b1       : 21
ID       : 0271
Command  : 02
Argument : 40
Sequence : B4
Group    : 02
Checksum : B7

Packet received (9 bytes):
Raw packet: 2A 81 31 48 6A 39 70 2B 8A 

Decoded:
Key      : 2A
b1       : 21
ID       : 0271
Command  : 01
Argument : 05
Sequence : B3
Group    : 02
Checksum : 9E

Packet received (9 bytes):
Raw packet: F8 E4 E7 B9 6E D3 10 6B 90 

Decoded:
Key      : F8
b1       : 21
ID       : 0275
Command  : 01
Argument : 07
Sequence : B2
Group    : 02
Checksum : 11
With no algorithm..Periodically..

sidoh, 2017-03-27:
Interesting.

I had to do some guesswork reversing the parameters for the ID bytes because they're fixed and I only have four devices. 

Can you paste more examples? Basically the closer you are to having all possible values for the first byte the better, but 20-50 is probably sufficient. Only need the `Raw Packet` part. 

Can use this command to dump all captured packets to a file automatically:

```
while [ 1 ]; do curl -s http://<esp>/gateway_traffic/rgb_cct | grep '^Raw' | cut -d':' -f2; done > packet_capture.txt
```

Lstt2005, 2017-03-27:
> while [ 1 ]; do curl -s http://<esp>/gateway_traffic/rgb_cct | grep '^Raw' | cut -d':' -f2; done > packet_capture.txt

Is it work in Windows command promt? :)

sidoh, 2017-03-27:
haha, no, it wouldn't.

Pasting all of the output is fine.

Lstt2005, 2017-03-27:
You mean like this ?
Packet received (9 bytes):
Raw packet: F8 E4 E7 B9 6E D3 10 6B 90

Decoded:
Key : F8
b1 : 21
ID : 0275
Command : 01
Argument : 07
Sequence : B2
Group : 02
Checksum : 11

sidoh, 2017-03-27:
Yeah, that's fine. Whatever's easier. Just want more data :)

Lstt2005, 2017-03-27:
[FUT_B2_RAW.xlsx](https://github.com/sidoh/esp8266_milight_hub/files/873530/FUT_B2_RAW.xlsx)
Enough?

sidoh, 2017-03-27:
That's helpful, thanks :)

sidoh, 2017-03-28:
Assuming I did it right, I think I found the correct values. I narrowed it down to a single option that keeps the IDs for the remotes I have and your remote fixed. I'll update in the morning.

Lstt2005, 2017-03-28:
Ok, thank you! Just found on Taobao - https://world.taobao.com/item/520922289204.htm?fromSite=main. Want buy it and make 2 DIY lamps and try it.  It's summary 108w - 36w RGB, 36W CoolWhite and 36W WarmWhite..
[image]

sidoh, 2017-03-28:
Updated (in v1.1.0 branch). Can you see if it's working?

Lstt2005, 2017-03-28:
Yes! It's work! 
[~$FUT_B2_RAW_agter_correct.xlsx](https://github.com/sidoh/esp8266_milight_hub/files/876484/FUT_B2_RAW_agter_correct.xlsx)

sidoh, 2017-03-28:
Sweet. 

So to be clear, the ESP is still not working with your bulbs, correct?

Lstt2005, 2017-03-28:
It's work, Chris!!!

sidoh, 2017-03-28:
Oh, sweet. I actually figured it wouldn't work because it's still sending a different value for the 2nd byte.

Was there anything else you had in mind for this ticket?

Lstt2005, 2017-03-29:
No, i think i have no question. Thanks.

sidoh, 2017-03-29:
Cool. Just wanted to make sure before closing. Thanks for raising the issue!

closed 2017-03-29 by sidoh

cookejames, 2017-04-01:
I have [this bulb](https://www.aliexpress.com/item/NEW-Milight-Led-Bulb-4W-5W-6W-9W-GU10-E27-E14-RGBW-RGBWW-Lamps-Wireless-Wifi/32790230448.html) (the 4W GU10 RGWCCT) which only responds to the RGBW type and not the RGB+CCT type. I've tried a couple of your different branches with no difference. Have I been missold and this is a RGBW bulb or do you think this is a different protocol or version?

I'm afraid I don't have a remote or hub to confirm if this bulb can do the CCT mode at all. I do have one of [these bulbs](http://www.ebay.co.uk/itm/182438943215) in warm white which is quite a nice light but physically it is much larger than a normal GU10 which is why I was excited to try the CCT bulb.

sidoh, 2017-04-01:
Interesting. If responds to the RGBW radio configs and protocol, I think it's probably RGBW-only. It's possible this is a different distributor that changed the hardware/firmware a little.

Are you able to take off the cap/housing to see what kinds of LEDs are in there? My RGB+CCT bulbs look like this:

[image]

cookejames, 2017-04-02:
Here it is with the top off and it looks like it has all the correct LEDs for CCT operation. Now I understand how it does the white colour change I've looked at the bulb when it is on and only the three cold white LEDs are working.

What would be the best way to investigate the protocol? Buy a remote so it can be sniffed?
[image]
[image]

sidoh, 2017-04-02:
Looks like I can't see the images.

Do you know what model of remote is supposed to work with your bulb?

cookejames, 2017-04-02:
Sorry I've uploaded the images to imgur instead so you should be able to see them. The seller I bought the bulbs from lists a remote in [this listing](https://www.aliexpress.com/item/NEW-Milight-Led-Bulb-4W-5W-6W-9W-GU10-E27-E14-RGBW-RGBWW-Lamps-Wireless-Wifi/32790230448.html) but that picture looks like a normal RGBW remote. However on eBay [others are selling it](http://www.ebay.co.uk/itm/MiLight-2-4g-remote-RGB-CCT-GU10-4W-led-bulb-lamp-Dimmable-RGBW-RGBWW-spot-light-/232286218967?var=&hash=item3615566ad7:m:mgu3fRyp8DdvmeyjYhHSFng) with what looks like a CCT remote.

sidoh, 2017-04-02:
I have no idea how the reseller/manufacturer arrangement works with these devices, but the website futlight.com seems to be the nicest and most organized list of devices. They mention which remotes each type of bulb is compatible with:

http://futlight.com/productdetails.aspx?id=223&typeid=101

Looks like these are compatible with two different types of RGBW remotes, and the RGB+CCT remote. I'm pretty sure the remote being advertised here is the same one I developed against.

Are you positive it doesn't work with RGB+CCT? I could imagine that if you pair the bulb with RGBW, it would stop listening on RGB+CCT bands. If you haven't already, you might try unpairing it with the RGBW.

cookejames, 2017-04-02:
I'm certain it doesn't work but your idea about pairing is a good one, I paired this before you had CCT support so I will try that tonight.

sidoh, 2017-04-02:
Cool. Unfortunately I don't think any of the bulbs I have support more than one protocol so I can't test it myself.

cookejames, 2017-04-09:
Sorry I didn't reply sooner. I ended up doing as you suggested and unpaired the bulb and that did the trick. It seems the bulb listens for the protocol at pairing and then only listens for that protocol. My RGBWW bulbs that were previously working using v5 once unpaired and paired using v6 now only respond to v6. The CCT bulb is working perfectly. Thanks for your hard work - hopefully I'll have some time to contribute back soon as I would like to add MQTT support.