sidoh/esp8266_milight_hub #17: Is it possible implement mqtt client ?

Lstt2005, 2017-03-20:
Is it possible implement mqtt client, based on pubsubclient - https://github.com/knolleary/pubsubclient

sidoh, 2017-03-20:
How are you imagining this would be used?

Lstt2005, 2017-03-20:
1. For example, can control MiLight via  Linear MQTT Dashboard https://play.google.com/store/apps/details?id=com.ravendmaster.linearmqttdashboard&hl=ru
2. SignUp on https://customer.cloudmqtt.com/login - and we have asolutly free mqtt cloud for 10 device.
3. Try IOBroker - http://forum.iobroker.net - greatest open sorce project for Home automation! - https://github.com/ioBroker
It consist of 75-90 drivers for all services (like Telegram chat bot, MQTT broker and client etc) and  ton's of Hardware 
For example, control MiLight via Telegram chat Bot (based on IOBroker) through MQTT client, based diretcly on ESP.

sidoh, 2017-03-21:
It always makes me feel like I'm missing something, but I'm not sold on MQTT being a good general-purpose protocol for home automation. It seems like it's great in situations where one end of communication isn't always available, but feels like it adds more moving pieces and another thing to break/debug.

It sounds like you're saying this would make it easier to integrate with HA platforms? I'm all for that, but most of the ones I'm familiar with already have support for the UDP protocol.

Lstt2005, 2017-03-21:
Partly agree .. However, it is very convenient when it is possible to control the end device (for example, ECP8266) directly from the smartphone, taking into account, of course, the MQTT server. It does not need, for example, a controller (OpenHab, MySensors)
In addition, I have already mentioned the links above to the ready-made MQTT clients, which can directly manage the ESP.
If you have the opportunity to create a branch with MQTT support, I would undertake to take on the function of an active stager.  :)
Sorry for my Russian English:)

Lstt2005, 2017-03-21:
For example, the POWERFULL PROJECT - https://github.com/arendst/Sonoff-MQTT-OTA-Arduino - it's use and MQTT protocol also..

sidoh, 2017-03-21:
Ah, I see. It sorta seems like the MQTT tooling you're describing is serving the role of a controller. At the point you have a bunch of devices and want to control them in a central location, why not use a hub (controller) like SmartThings, HomeAssistant, OpenHab, etc.?

Don't mean to be dense - just wanting to understand :)

Lstt2005, 2017-03-21:
No, not exactly. I mean, to implement in your library only the ability to connect to an external server (either local or cloud). And then it would be possible to manage ESP8266 both through the central controller of Smart Home (OpenHab, etc.), and directly, via a mobile client on the android, in case of connecting an ESP to an external MQTT server.
In Russia, for example, not all providers provide the service "White IP", in addition, double NAT is used, so sometimes it is difficult to connect from the outside to the controller in the house ..

sidoh, 2017-03-22:
Oh, I see what you mean.

At the risk of getting too philosophical, allowing external access seems like it ought to be a function of a hub/controller. SmartThings, for example, enables this by establishing an outbound connection to ST cloud servers. I guess you could probably do the same thing with a local controller with some SSH proxy stuff?

I'm happy to look into this, but I think there are quite a few things I'd want to get to before it.

sidoh, 2017-04-18:
Took a stab at MQTT subscriber here:

https://github.com/sidoh/esp8266_milight_hub/tree/mqtt

Needs some documentation and I haven't verified that it works yet, but should be pretty close to working.

sidoh, 2017-04-19:
Okay, verified that this works. Still want to spruce some things up and test a little more, but should be working.

sidoh, 2017-04-20:
@Lstt2005, are you able to test MQTT with the mqtt branch? Instructions in the README within that branch. Would need to update the web UI as well.

Lstt2005, 2017-04-20:
Yes! Works great! Thanks!

sidoh, 2017-04-21:
Awesome. :)

Let me know if you notice issues.

Lstt2005, 2017-04-21:
Ok, thanks!

closed 2017-04-22 by sidoh