sidoh/esp8266_milight_hub #19: failed upload index.html

Lstt2005, 2017-03-23:
Hello! I try to upload to index.html to ESP [image]
After, i try view content on it and i see - 
[image]

Lstt2005, 2017-03-23:
[image]

sidoh, 2017-03-23:
`/download_update` is only supported in a branch I haven't merged yet.

Did you try restarting the ESP after uploading the UI?

Lstt2005, 2017-03-23:
Yes, of course. I flash whole my NODE MCU with blank.bin 4Mbytes, try upload via PlatformIO, tru flash you distrib in github - same result...(

sidoh, 2017-03-23:
I've seen it not work right after uploading, and for whatever stupid reason it's there after restarting the ESP.

If it's consistently not working, I think it can only mean the flash size settings in the distributed firmware are a mismatch for the device you're pushing it to.

Lstt2005, 2017-03-23:
I have the same Lolin
[image]

sidoh, 2017-03-23:
Can you share the command you're using to flash it?

Lstt2005, 2017-03-23:
In PlatformIO
[03/23/17 20:53:25] Processing nodemcuv2 (platform: espressif8266, board: nodemcuv2, framework: arduino)
 Verbose mode can be enabled via `-v, --verbose` option
Collected 32 compatible libraries
Looking for dependencies...
Library Dependency Graph
|   |-- <MiLight>
|   |   |-- <RF24> v1.1.7
|   |   |   |-- <SPI> v1.0
|   |-- <Settings>
|   |   |-- <ArduinoJson> v5.8.3
|   |   |-- <Helpers>
|   |-- <ESP8266WebServer> v1.0

|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ArduinoJson> v5.8.3
|   |-- <Vector>
|   |-- <Helpers>
|   |-- <ESP8266WiFi> v1.0
|-- <WifiManager> v0.12
|   |-- <DNSServer> v1.1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WebServer> v1.0
|   |   |-- <ESP8266WiFi> v1.0
|   |-- <ESP8266WiFi> v1.0
|-- <Helpers>
|-- <ArduinoJson> v5.8.3
|-- <SPI> v1.0
|-- <Settings>
|   |-- <ArduinoJson> v5.8.3
|   |-- <Helpers>
Linking .pioenvs\nodemcuv2\firmware.elf
Checking program size
text       data     bss     dec     hex filename
358054     8948   30488  397490   610b2 .pioenvs\nodemcuv2\firmware.elf
Building .pioenvs\nodemcuv2\firmware.bin
Looking for upload port...
Auto-detected: COM3
Uploading .pioenvs\nodemcuv2\firmware.bin
Uploading 371152 bytes from .pioenvs\nodemcuv2\firmware.bin to flash at 0x00000000
................................................................................ [ 22% ]
................................................................................ [ 44% ]
................................................................................ [ 66% ]
................................................................................ [ 88% ]
...........................................                                      [ 100% ]
 [SUCCESS] Took 42.24 seconds

Lstt2005, 2017-03-23:
[image]

Lstt2005, 2017-03-23:
I solve the problem. 
1. Run Arduino 1.8.1.
2. Loand ANY sketch
3. Make directory DATA in the folder of sketch.
4. Copy index.html on DATA folder.
5. IN configuration Select "Flash Size 4M (1M SPIIFS)" - ONLY! (not 3M SPIFFS)!
6. In Arduino in Menu select Tools->ESP8266 Sketch Data Upload
7. Restart NodeMCU.
8. Run   curl -u':' -X PUT --data-binary '{"admin_username":""}' 192.168.0.111/settings
9. Open in browser http://192.168.0.111 and make some username and password - submit

If i use balnk user and password, in next time i must input username and password in browser.
[image]

sidoh, 2017-03-24:
> In configuration Select "Flash Size 4M (1M SPIIFS)" - ONLY! (not 3M SPIFFS)!

Yeah, I've found that the ESP is super finicky about the flash size parameters. My guess is that you'd see something like this in serial logs:

```
Error updating web file
```

I think Sketch Data Upload is just building and flashing a SPIFFS image, which shouldn't be necessary as long as the flash params are set right. I've found that I'm reliably able to use SPIFFS on a fresh chip (and I've tested with ~10 across 5 different models) if I:

1. Erase flash, regardless of whether or not it's a fresh chip
1. Make sure I'm flashing it with firmware compiled with the correct settings
1. Flash with `esptool.py`

Seems like most people who use this run into trouble getting SPIFFS to work correctly. This is basically the source of all of the problems with settings not saving, trouble uploading the web UI, etc. Probably worth looking into at least documenting the procedures a little better.

Lstt2005, 2017-03-24:
Maybe in PlatformIO configuration parametrs for esp8266 something wrong?

sidoh, 2017-03-24:
Yeah, sounds plausible. I've had trouble with default parameters for esp-07 and esp-12 in the past.

Lstt2005, 2017-03-24:
1. Just add in platrom.ini

[env:nodemcuv2]
platform = espressif8266
framework = arduino
board = nodemcuv2
build_flags = -Wl,-Tesp8266.flash.4m1m.ld
upload_speed = 921600

2. Create a folder data in main project folder.
3. Place in folder data file index.html
4. In treminal run: pio run -e nodemcuv2 -t uploadfs
5. Profit!

I recommend to you modify this intruction in read.me - it's much easier, then use curl etc....
Much more in http://docs.platformio.org/en/stable/platforms/espressif8266.html#uploading-files-to-file-system-spiffs

sidoh, 2017-03-24:
I use the default build_flags for the modemcuv2 with what appears to be the same hardware and have no issues. I'm not convinced it makes sense to just drop in the settings you pasted and call it fixed.

Right, I'm familiar with platformio's SPIFFS tools. The reason I'm avoiding relying on them here is because I don't want to force people to set up a development environment to use this. As long as you're using the right flags when flashing the firmware (the reason I'd suggest using `esptool.py` is because you have more explicit control over this), I don't actually think using the SPIFFS tools are easier than using curl (maybe not the case if using Windows).

Lstt2005, 2017-03-24:
Yes, there are, for example, hundreds beginners on esp8266.ru - they using only Windows...For Windows users much easer use Arduino IDE or PlatforIO and nothing else...:)

sidoh, 2017-03-24:
Alright, fair enough. Probably a good idea to at least mention it as an option. I'll get around to this at some point, but feel free to open a PR if you want. :)

Lstt2005, 2017-03-24:
No, for me it's just a hobby, nothing else....

sidoh, 2017-03-24:
Me too :)

I'm going to leave this open as a reminder to flesh out the docs around getting the ESP set up. Thanks for raising!

Lstt2005, 2017-03-24:
Ok, thank you!

sidoh, 2017-03-26:
Updated the documentation to suggest using `uploadfs` instead of the HTTP endpoint:

https://github.com/sidoh/esp8266_milight_hub/commit/9e76eeb16799fd6f640aca29d2ec56c19fb2c9ef

Thanks for pushing on that. I initially felt like it was more work, but with all the problems people have had uploading the UI, I think this is better.

closed 2017-03-26 by sidoh