sidoh/esp8266_milight_hub #24: General request future (like software WGT)

Lstt2005, 2017-03-29:
Maybe add to library software WGT (and maybe add this option in Web UI)? For example, On, Off, sec for restart (0-xxx)?

sidoh, 2017-03-29:
I'm not familiar with the acronym WGT. Can you tell me what it means?

Lstt2005, 2017-03-29:
WDT, of course :)

sidoh, 2017-03-29:
Haha. What is WDT? I promise I'm not trying to be dense :)

Lstt2005, 2017-03-29:
:) WatchDogTimer - in ESP - ESP.restart() restarts the CPU. https://github.com/esp8266/Arduino/blob/d2b370b22d1593226fad3ae82d7feac795dfb51d/doc/libraries.md
Because ESP8266 NOT stable device sometimes:)

sidoh, 2017-03-29:
Gotcha. So the idea is to provide a way to restart the ESP via an API, and also maybe a way for regular restarts (like restart every 24 hrs or something)?

Lstt2005, 2017-03-29:
Yes I agree. But for beginners, I think that it would be very convenient if such an automatic restart possible with the indication of a time interval or in the UI, or the parameter setting via the API (min, hours, etc.) would be realized..

sidoh, 2017-03-29:
Sure, makes sense. If we added it to the API, of course makes sense to expose it in the UI as well.

So I'm imagining a command to force a reset, and an optional setting to schedule a reset every X minutes. Obviously would find a way to incorporate both of those things into the UI as well.

Does that sound in line with what you're imagining?

sidoh, 2017-03-30:
I added a setting for auto restarts and a restart button in the v1.1.0 branch.

Lstt2005, 2017-03-30:
Ok, thank! Today i test it.

sidoh, 2017-03-30:
Sweet!

Did this get at everything you had in mind?

Lstt2005, 2017-03-30:
For now, i don't new ideas. Thanks.

sidoh, 2017-03-30:
Sweet. Thanks for the idea! :)

closed 2017-03-30 by sidoh