sidoh/esp8266_milight_hub #25: Add support for "Disco" modes for HTTP.

DarkFox, 2017-03-29:
As far as I can see (Looking through the code), the Disco modes are only supported via the official UDP implementation, and possibly only as cycling through them, by simulating a button push.

sidoh, 2017-03-29:
Support for modes and mode speeds is added in [v1.1.0](https://github.com/sidoh/esp8266_milight_hub/tree/v1.1.0). :)

Unfortunately cycling through modes is the only way to do it for some (most?) bulb types. The underlying RF protocol doesn't support anything else. RGB+CCT bulbs allow you to set it directly.

DarkFox, 2017-03-29:
Ah! I was looking in the wrong branch then. 👍 

As for setting directly, that sounds odd. I've got a handful of RGB-only LED STRIP controllers, that seem to support setting it directly without problem (Via official bridge and RGBLEDController app). It also seems to be described at the bottom of this post: https://hackaday.io/project/5888-reverse-engineering-the-milight-on-air-protocol/log/18529-command-and-control

Though I suppose it could've just not been implemented in the library?

sidoh, 2017-03-29:
Wouldn't have expected you to find my unannounced dev branch, haha. I'll merge it into master when I wrap up a few more features I want to knock out.

Haha. Nah, I'm super happy to add features people find useful :)

Just meant to say I don't think most of the RF protocols used to communicate with the bulbs support commands of the form "set mode to X." The ones I've seen aside for the new one for RGB+CCT bulbs have "next mode" and "previous mode."

sidoh, 2017-03-31:
Noticed that the v6 app had a way to directly set modes for RGBW bulbs (the v5 app does not). I'll merge into the `v1.1.0` branch soon.

sidoh, 2017-04-03:
Should be implemented in v1.1.0. Let me know if you notice any issues.

closed 2017-04-03 by sidoh