sidoh/esp8266_milight_hub #27: Some question about UDP

Lstt2005, 2017-03-30:
As i write earlier, i found github Android project Bilight - https://github.com/JasperG/bilight
When i experiment with it, sometimes, i see this messages in ESP8266 console:
"
Received request with untracked session ID: 0
Received request with untracked session ID: 0
Received request with untracked session ID: 0
Received request with untracked session ID: 0
Received request with untracked session ID: 0
Received request with untracked session ID: 0
Received request with untracked session ID: 0
Received request with untracked session ID: 0
Received request with untracked session ID: 0
"
Sometimes be, sometimes no. Question - is it application problem or some bug in ESP8266 library?

sidoh, 2017-03-30:
That means the app is sending a command with a session ID the ESP doesn't recognize. Sessions are only important in order for the ESP to know which IP and port to send a response packet to. If the app works well without response packets, it shouldn't matter.

This could happen if, for example, the app was started and then the ESP was rebooted. The app should probably handle getting a new session, but looks like maybe it's not.

Lstt2005, 2017-03-30:
Ok, i see.

Lstt2005, 2017-03-30:
It's work situation, when some server with different ID work on one UDP port?
[image]

sidoh, 2017-03-30:
That wouldn't work, no. There has to be a way to select a device ID, and the UDP protocol doesn't expose a way for that to happen. So you gotta go off of something like the port.

Lstt2005, 2017-03-30:
I have a great desire to experimet with Bilight and want control more, than 1 ID with the same port. Can we try enter multiple Device ID in Gateway Servers?

sidoh, 2017-03-30:
I guess I mean to say that as far as I know, there is literally not a way to do this within the confines of the official UDP protocol.

Lstt2005, 2017-03-30:
You did not understand me a little. I mean, if you can add more than one ID function to the library, if there are more, then send several packets to other stored IDs

sidoh, 2017-03-30:
Hmm, I see. I guess this feels a little unclean to me, but I'm not sure why. I guess maybe because it seems like clients ought to be able to address a specific identifier directly.

Lstt2005, 2017-03-30:
Although, perhaps you're right! In fact, nothing prevents you from pair a virtual ID to an unlimited number of devices and managing them via UDP.

Lstt2005, 2017-03-31:
Today try new branch - discovery - work! :)
Sending response to 192.168.0.119:52123
[MiLightUdpServer port 5987] - Handling packet: D0 00 00 00 02 00 02
Packet size: 7
Sending response to 192.168.0.119:52123
[MiLightUdpServer port 5987] - Handling packet: D0 00 00 00 02 00 02
Packet size: 7
Sending response to 192.168.0.119:52123
[MiLightUdpServer port 5987] - Handling packet: D0 00 00 00 02 00 02
Packet size: 7
Sending response to 192.168.0.119:52123
[MiLightUdpServer port 5987] - Handling packet: D0 00 00 00 02 00 02
Packet size: 7
Sending response to 192.168.0.119:52123

sidoh, 2017-03-31:
:) awesome

Lstt2005, 2017-03-31:
one question - if i try new china LEDs - can i implemet new sync words or different packet size..?
I mean this - https://detail.1688.com/offer/522143371857.html 
[image]

sidoh, 2017-03-31:
Yeah, I think it probably is a knockoff or redistributor of MiLight. I have no way of knowing for sure, though.

It's pretty easy to add different syncwords and packet structures at this point, so if it does need that, easy enough to add.

Lstt2005, 2017-03-31:
Ok, i must receive within 1 mouth, and try it.

Lstt2005, 2017-03-31:
MiLight 3.0 Work!
[image]
[image]
But, as I wrote earlier, adjusting the color temperature works the other way around.

sidoh, 2017-03-31:
Yeah I was noticing that too. I basically chose the direction that seemed to match the choice made by the folks who worked on MiLight stuff for HomeAssistant. It's clear this is flipped from the official MiLight app, though. I just pushed a fix.

sidoh, 2017-04-02:
The white temperature ordering should be correct in the latest dev release. 

Did we answer all the questions you had for this ticket?

Lstt2005, 2017-04-02:
No, thanks!

sidoh, 2017-04-02:
👍

closed 2017-04-02 by sidoh