sidoh/esp8266_milight_hub #31: Milight app v3 some qustion

Lstt2005, 2017-04-03:
It's normally, that saved value in Milight app do not apply ?
[image]

Lstt2005, 2017-04-04:
[image]
When i chose this remote control (virtual FUT091), which we test in Web interface, it dont work with my FUT035 controller, and i see in debug:
:ON:
session: 0002, sequence: 229, group: 1, checksum: 60
Command cmdType: 31, cmdHeader: 00000304, cmdArg: 03000000
V6MiLightUdpServer - Unhandled command: 31 00 00 03 04 03 00 00
[MiLightUdpServer port 5987] - Handling packet: 80 00 00 00 11 00 02 4C E6 00 31 00 00 03 04 04 00 00 00 01 00 3D
Packet size: 22
:OFF:
session: 0002, sequence: 230, group: 1, checksum: 61
Command cmdType: 31, cmdHeader: 00000304, cmdArg: 04000000
V6MiLightUdpServer - Unhandled command: 31 00 00 03 04 04 00 00
[MiLightUdpServer port 5987] - Handling packet: 10 00 00 00 0A 02 AD A7 11 18 FE 34 D3 95 31

sidoh, 2017-04-04:
Right, I don't think that remote is handled right now. Do you need it?

Lstt2005, 2017-04-04:
[image]
FUT091 - we begin our work together with him :) Please, can you add this remote in UDP code?

sidoh, 2017-04-04:
Yeah, should be straightforward enough. There are some other things I want to get to first, though (namely wrapping up the LT8900 stuff)

Lstt2005, 2017-04-04:
Ok, wait..Thanks!

Lstt2005, 2017-04-04:
Whats about 
"It's normally, that saved value in Milight app do not apply ?"
[image]

sidoh, 2017-04-04:
Yep, I'll take a look when I finish up the other stuff =)

Lstt2005, 2017-04-04:
Do you know, how mute sound in app v3 :)?

sidoh, 2017-04-04:
lol never noticed there is sound

Lstt2005, 2017-04-04:
I mean sound, when rotate the wheel with brightness, for example..

sidoh, 2017-04-05:
I don't think mine has sound lol :)

re: saved modes, this is yet another undocumented feature of the protocol. Just took a crack at implementing it. Can you try the udp_presets branch?

Lstt2005, 2017-04-05:
When i try udp_preset branch i receive:
Compiling .pioenvs\esp12\src\main.o
Archiving .pioenvs\esp12\libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\esp12\FrameworkArduino\Esp.o
Compiling .pioenvs\esp12\FrameworkArduino\FS.o

Compiling .pioenvs\esp12\FrameworkArduino\HardwareSerial.o
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\esp12\FrameworkArduino\IPAddress.o
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\esp12\FrameworkArduino\MD5Builder.o
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\esp12\FrameworkArduino\Print.o
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
*** [.pioenvs\esp12\src\main.o] Error 1
*** [.pioenvs\esp12\FrameworkArduino\Esp.o] Error 1
*** [.pioenvs\esp12\FrameworkArduino\FS.o] Error 1
:(((

sidoh, 2017-04-05:
I think I wasn't being very clear about what's broken.

It's just `platformio.ini`. If you replace it with the version from 1.1.0 it should work just fine.

Lstt2005, 2017-04-05:
udp_presets branch works correctly!

Lstt2005, 2017-04-05:
Some little bug: when we long press ON button, we see in debug: V6MiLightUdpServer - Unhandled command: 31 00 00 08 84 01 00 00 - i think is command Light ON on 100%
When i sniff FUT091 i see:
Packet received (9 bytes):
Raw packet: C8 14 2A 1C 1E 08 4F 9D C8 

Decoded:
Key      : C8
b1       : 21
ID       : 1141
Command  : 81
Argument : 00
Sequence : A5
Group    : 00
Checksum : 89
Packet received (9 bytes):
Raw packet: 79 3D F7 DD C1 21 73 43 F9 

Decoded:
Key      : 79
b1       : 21
ID       : 1141
Command  : 81
Argument : 00
Sequence : A4
Group    : 00
Checksum : D9
Packet received (9 bytes):
Raw packet: DF 0C 91 FB 03 62 74 98 1F 

Decoded:
Key      : DF
b1       : 21
ID       : 1141
Command  : 81
Argument : 00
Sequence : A3
Group    : 00
Checksum : 2A

sidoh, 2017-04-05:
Nice.

Makes sense, I can poke at that later today.

sidoh, 2017-04-06:
So I don't think this behavior for FUT092. Holding down on or off doesn't seem to do anything.

Can you remind me what bulbs you're using with FUT092?

Lstt2005, 2017-04-06:
I use FUT035 LED strip controller - http://futlight.com/productdetails.aspx?id=124&typeid=146
When holding down ON - command for FULL brightness, holding down off - NIGHT MODE (about 2-4 % brightness).
BTW - Futlight released new remot control -8 zone ! - http://futlight.com/productdetails.aspx?id=278&typeid=143
Maybe with new protocols command:)

sidoh, 2017-04-06:
Oh ok. Those are controlled by the RGB radio right?

I think there's an 8 zone controller in the v6 wifi box. I poked at the protocol and it looks pretty similar to the protocol for FUT092. Just different offsets in on/off commands to accommodate more groups.

Lstt2005, 2017-04-06:
>Oh ok. Those are controlled by the RGB radio right?
What you mean?

I use FUT035 LED strip controllers (CCT not RGB) - http://futlight.com/productdetails.aspx?id=124&typeid=146
When holding down ON (on FUT091) - command for FULL brightness, holding down off - NIGHT MODE (about 2-4 % brightness).

sidoh, 2017-04-06:
Which mode in esp8266_milight_gateway do you use to control these devices?

Lstt2005, 2017-04-06:
I mean in Milight app v3 i use 1st remote control (RGG+CCT) - it's has separate button - Night mode, but don't have mode Full brighness, wich work (on LED controller), when i holding down ON button on real FUT091...

sidoh, 2017-04-06:
Gotcha.  It's interesting that these worth with FUT091/FUT092 at all. Neither of them are listed as compatible on FUT035's product page (the one you linked). It says it's compatible with FUT005, FUT006, and FUT007.

Lstt2005, 2017-04-06:
Yes, you are right. But if we see on http://futlight.com/productdetails.aspx?id=233&typeid=100 (FUT091) we see compatible LED controller (FUT035, FUT036 and it's on fact and another different bulbs). I think desription on FUT035 not full..

sidoh, 2017-04-06:
I see, makes sense.

Can you give the [held_commands](https://github.com/sidoh/esp8266_milight_hub/tree/held_commands) branch a try? Basically I tried to make it so that if you hold a button in the app, that's reflected in the RF packets. Haven't had a chance to test it out myself.

Lstt2005, 2017-04-06:
Yes, it's work! Thanks!

sidoh, 2017-04-06:
Sweet. I'll merge into 1.2.0 later.

sidoh, 2017-04-07:
Thanks as always for finding bugs and providing suggestions, @Lstt2005!

Does that cover it for this ticket? Anything else you had in mind?

Lstt2005, 2017-04-07:
Can you plane support FUT091 in Milight App? :)

sidoh, 2017-04-07:
Oh, right. Can certainly look into it.

Are there things this remote does that FUT092 doesn't? It's pretty easy to add support, just don't wanna add bloat :)

Lstt2005, 2017-04-07:
No, there certainly are no additional functions .. Just when you select multiple WiFiBox, it's more convenient to click on its icon, and then the only CCTv2 mode control tab will open immediately. A trifle, but very convenient, when the number of management points (Wifi) is large .. :)
If you can invented for support multiple gateways in MiLIght app V3 (in tabs of gateways) works with one ESP It's WILL BE MIRACLE!

sidoh, 2017-04-07:
So I poked at this again for a bit, and I'm now at least pretty convinced it's possible, but it'll take some fairly significant changes to the way UDP packets are handled.

Basically rather than varying the port, I think you could have a UDP server with multiple device IDs respond with several different MAC addresses. Both the discovery packet response and the session start request have a MAC address in them, so you should know which device the client is trying to address. 

It'll take a bit of time, but should be doable.

Lstt2005, 2017-04-07:
>UDP server with multiple device IDs 

You mean C++ code inside you library on external server?

sidoh, 2017-04-07:
Inside the C++ code.

The result, if it works like I'm hoping it will, is that you'd see a bunch of devices show up in the device list, each responding to commands with a different device ID.

Lstt2005, 2017-04-07:
It would be just amazing !!!!!
BTW, yesterday - the day before yesterday Milight v3  was updated on GooglePlay.

sidoh, 2017-04-07:
Sweet. Anything new?

Lstt2005, 2017-04-07:
Write:1.Modified some bugs to be compatible with some smartphone which cannot work before.
Visually nothing change.

Lstt2005, 2017-04-07:
> The result, if it works like I'm hoping it will, is that you'd see a bunch of devices show up in the device list, each responding to commands with a different device ID.

It would be possible to manually set the device names in list, like as : kitchen, dinig room, etc? Maybe in code directly?

sidoh, 2017-04-07:
Probably not unless the Milight app has support for it.

Lstt2005, 2017-04-07:
It turns out that this can be changed in the application:
[image]

sidoh, 2017-04-07:
Ah, sweet! Where is that screen?

sidoh, 2017-04-07:
Oh i see it. You left-swipe on the line item and an edit button shows up. Pretty well hidden haha

sidoh, 2017-04-09:
Alright, I poked at it a little more thoroughly, and I'm feeling disheartened again. 

For background, the connection process looks like this:

P1. (Client -> Server): send search packet via UDP broadcast on port 5987
P2. (Server -> Client): send search response
P3. (Client -> Server): send session start request
P4. (Server -> Client): send session ID, to be used with all further requests

The challenge is:

1. In order for a single device to serve multiple IDs, there has to be a way for the client to specify which one it means to control.
1. I've tried two rough approaches with this.
  A. The search response packet contains a field for port. Changing this field in the response causes the client to send the session start request (P3) to the new port, but further requests are on port 5987, and the session started on the other port is never used again as far as I can tell.
  B. I had thought that P3 includes some scrutable identifier (this is why I felt positive in my last update), like the mac address of the device it was trying to start a session with, but it's the same packet every time. Not sure why I thought that it was any other way.
1. I could be wrong about some of my assumptions above, but if they're all true, I don't think it's possible to do this. :(

Have you used HomeAssistant or something like that, @Lstt2005? It should integrate nicely with this project and I think has most of the same controls that the Milight app does.

Lstt2005, 2017-04-09:
ok. i see.

closed 2017-04-22 by sidoh