sidoh/esp8266_milight_hub #32: Addition for dual white bulbs formatter

WoodsterDK, 2017-04-03:
What about making a frame that drives the bulbs all the way up in light levels and color all warm.
Then a known state can be achieved, and it will be possible to dim and color set to a known state.

sidoh, 2017-04-03:
I think it already does this, right?

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/CctPacketFormatter.cpp#L16
https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/PacketFormatter.cpp#L79

WoodsterDK, 2017-04-03:
Seems like it, well done.☺

sidoh, 2017-04-03:
Sweet.

This is part of the reason I refactored PacketFormatter to support buffering multiple packets. Good catch. :)

WoodsterDK, 2017-04-18:
Hi, just had a look at it - and noticed, that the level always get reset by going down, before going up.
This is the absolute level - but look wise might seem strange to go down before going up if going from level 6 to 8.
What about implementing a relative level command, that steps the desired levels, without taking the zero level into account.
Today I'm handling this in an upper layer - then a smooth transition can be made.
When my upper layer initializes, then the standard level command will be called to put the bulbs at a well known level, and afterwards the "relative level command" will be used.

sidoh, 2017-04-18:
It basically seems like this amounts to calling `increase_brightness` or `decrease_brightness` multiple times? I think you could just do that with the REST API. Maybe could implement handling a list of commands instead of a single command.

I'm looking into MQTT now in case that makes this approach seem any more appealing.

WoodsterDK, 2017-04-18:
I know - but performance wise it will look better, as it does not add any unnecessary overhead.

sidoh, 2017-04-18:
The multiple command thing wouldn't really add much overhead. We're probably talking sub-millisecond. Imagining you'd send something like this:

```json
{"commands":["increase_brightness","increase_brightness"]}
```

I assume MQTT would have reasonable latency even without this new feature since everything is over pre-established connections.

WoodsterDK, 2017-04-18:
Ah... I haven't figured out that you could do it like this.... thought it had to be multiple REST commands:
{"commands":"increase_brightness"}
{"commands":"increase_brightness"}
Then of course it does not add extra....

sidoh, 2017-04-18:
Yeah, multiple HTTP requests would be pretty slow for this kind of thing. Multiple MQTT messages might not be so bad, though.

You can't do the multiple commands thing yet, but it shouldn't be so hard to implement. Just opened #55 for that. If you don't mind, I'm gonna close this one.

closed 2017-04-18 by sidoh