sidoh/esp8266_milight_hub #33: Add default configuration to the platformio.ini

WoodsterDK, 2017-04-04:
Making a section, that fills in desired default settings to the flash image.
E.g. pin number definitions and radiointerface type.

sidoh, 2017-04-04:
Great idea.

I guess there are two ways I can think of to solve this:

1. Settings are currently stored as JSON in SPIFFS, so we could just have different settings templates that users paste into `data` and flash a SPIFFS image
2. Compiler flag magic.

What are your thoughts? I guess I'm partial to #1 because I'm bad at compiler flag magic, and it seems clean, but I suppose it's a worse user experience.

WoodsterDK, 2017-04-04:
1 seems fine as a start.... it will be easy to switch to desired configuration according to the setup.

sidoh, 2017-05-06:
@WoodsterDK, another option might be to supply some JSON blobs that are uploaded via the web UI. I just added the UI feature in v1.3.0 for #79. Thoughts on that as an approach? Maybe a little less cumbersome than (1).

WoodsterDK, 2017-05-06:
Super option.