sidoh/esp8266_milight_hub #34: Fix Windows build compatability

sidoh, 2017-04-04:
The changes to `platformio.ini` to support tracking version and continuous deployment broke compatibility with Windows

Lstt2005, 2017-04-04:
Many mistale in branch PL1167...

sidoh, 2017-04-04:
Can you clarify what you mean?

This branch is a work in progress. Obviously you're welcome to try it, but don't count on it working. :)

Lstt2005, 2017-04-04:
|-- <Settings>
|   |-- <ArduinoJson> v5.8.4
|   |-- <Helpers>
|-- <GithubClient>
|   |-- <ESP8266WiFi> v1.0
|-- <Helpers>
Compiling .pioenvs\esp07\src\main.o

xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\esp07\FrameworkArduino\Esp.o
Compiling .pioenvs\esp07\FrameworkArduino\FS.o
Compiling .pioenvs\esp07\FrameworkArduino\HardwareSerial.o
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\esp07\FrameworkArduino\IPAddress.o
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\esp07\FrameworkArduino\MD5Builder.o
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\esp07\FrameworkArduino\Print.o
xtensa-lx106-elf-g++: error: unrecognized command line option '--always)"'
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
*** [.pioenvs\esp07\src\main.o] Error 1
*** [.pioenvs\esp07\FrameworkArduino\Esp.o] Error 1
PIO Buildsrc\main.cpp0001:1
LFUTF-8C++4 updates

sidoh, 2017-04-04:
Oh, the build you mean.

Yeah, sorry that's so broken. For now you can probably just revert `platformio.ini` to what it was before v1.1.0. I'll work on finding a way to make it compatible with Windows.

Lstt2005, 2017-04-04:
But platformio.ini in brach esp8266_milight_hub-1.2.0 - works well:
[common]
board_f_cpu = 160000000L
lib_deps_builtin =
  SPI
lib_deps_external =
  RF24
  WiFiManager
  ArduinoJson
build_flags = !echo -D MILIGHT_HUB_VERSION="\"$(git describe --always)\""
//build_flags = -D MILIGHT_HUB_VERSION=\"1.1.0\" -D MILIGHT_UDP_DEBUG
# -D MILIGHT_UDP_DEBUG

[env:nodemcuv2]
upload_speed = 921600
platform = espressif8266
framework = arduino
board = nodemcuv2
lib_deps =
  ${common.lib_deps_builtin}
  ${common.lib_deps_external}
build_flags = ${common.build_flags} -D FIRMWARE_VARIANT="\"nodemcuv2\""

sidoh, 2017-04-04:
Right, that's because WoodsterDK commented out a bunch of stuff that's needed by Travis for continuous deploys.

sidoh, 2017-04-05:
@Lstt2005 are you able to try out building under this branch?

https://github.com/sidoh/esp8266_milight_hub/tree/fix_build

Took a stab at fixing windows builds.

sidoh, 2017-04-05:
@Lstt2005, could you try building here? I don't have a Windows machine so it's tricky for me to test :(

Lstt2005, 2017-04-05:
I can try after 8-10 hours..

sidoh, 2017-04-05:
Haha, cool. Didn't mean to pester you. No rush. Thanks man :)

Lstt2005, 2017-04-05:
Still to many erros :

Error: Traceback (most recent call last):
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\__main__.py", line 106, in main
    cli(None, None, None)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\user\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 113, in cli
    pg = ProjectGenerator(project_dir, ide, board[0])
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 36, in __init__
    self._gather_tplvars()
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 146, in _gather_tplvars
    self._tplvars.update(self.get_project_env())
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\util.py", line 134, in __call__
    value = self.func(*args)
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 52, in get_project_env
    for k, v in config.items(section):
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\util.py", line 52, in items
    items.append((option, self.get(section, option)))
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\util.py", line 59, in get
    return self.VARTPL_RE.sub(self._re_sub_handler, value)
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\util.py", line 65, in _re_sub_handler
    return self.get(section, option)
  File "c:\users\user\.platformio\penv\lib\site-packages\platformio\util.py", line 56, in get
    value = ConfigParser.get(self, section, option, **kwargs)
  File "C:\Users\User\.platformio\python27\Lib\ConfigParser.py", line 618, in get
    raise NoOptionError(option, section)
NoOptionError: No option 'build_flags' in section: 'common'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

sidoh, 2017-04-05:
Are you on the "fix_build" branch? 

Just to make sure I'm clear, this is just one error that you've pasted, right? Were there others?

Lstt2005, 2017-04-05:
Yes, branch "fix_build":
Error: [04/05/17 18:25:33] Processing nodemcuv2 (platform: espressif8266; lib_deps: SPI, , RF24, WiFiManager, ArduinoJson; build_flags: !python .get_version.py -D FIRMWARE_VARIANT=nodemcuv2; board: nodemcuv2; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option

 [SUMMARY] 

Environment d1_mini  	[SKIP]

Environment esp12    	[SKIP]

Environment esp07    	[SKIP]

fatal: Not a git repository (or any of the parent directories): .git

Traceback (most recent call last):

File ".get_version.py", line 3, in <module>

version = check_output(["git", "describe", "--always"]).rstrip()

File "C:\Python27\Lib\subprocess.py", line 219, in check_output

raise CalledProcessError(retcode, cmd, output=output)

subprocess.CalledProcessError: Command '['git', 'describe', '--always']' returned non-zero exit status 128

OSError: 'python .get_version.py -D FIRMWARE_VARIANT=nodemcuv2' exited 1:

File "C:\users\user\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 159:

env.SConscript("$BUILD_SCRIPT")

File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:

return _SConscript(self.fs, *files, **subst_kw)

File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:

exec _file_ in call_stack[-1].globals

File "C:\users\user\.platformio\platforms\espressif8266\builder\main.py", line 347:

target_elf = env.BuildProgram()

File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:

return self.method(*nargs, **kwargs)

File "C:\users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 57:

env.ProcessFlags(env.get("BUILD_FLAGS"))

File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:

return self.method(*nargs, **kwargs)

File "C:\users\user\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 115:

parsed_flags = env.ParseFlags(str(flags))

File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 800:

do_parse(arg)

File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 676:

arg = self.backtick(arg[1:])

File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 594:

raise OSError("'%s' exited %d" % (command, status))

 [ERROR] Took 0.86 seconds 

Environment nodemcuv2	[ERROR]

 [ERROR] Took 0.86 seconds

sidoh, 2017-04-05:
Aah, this is helpful. Thanks!

How are you starting the build?

Can you try adding the `-v` flag and pasting the output?

sidoh, 2017-04-05:
Okay, I tried a different approach which should work better on Windows, or at least not crash the build. Can you let me know if it works?

Lstt2005, 2017-04-05:
Yes, works fine!

sidoh, 2017-04-05:
Excellent! thanks for testing :)

sidoh, 2017-04-05:
Merged this into master and released v1.1.1 with the fix. Will also merge this into the PL1167 and udp_presets branches.

closed 2017-04-05 by sidoh