sidoh/esp8266_milight_hub #37: Is it any chance make support miltiple gateways in MiLIght app V3 ?

Lstt2005, 2017-04-05:
Can any chance make support (maiby in library?) for support miltiple gateways in MiLIght app V3 (in tabs of gateways) works with one ESP - maybe on other UPD ports? Very cool and comfortable application for bulbs and LED strips..

Lstt2005, 2017-04-05:
May be this repo help - https://github.com/jjensn/milight-decompile  ?

sidoh, 2017-04-05:
Hah, interesting. It could!

Unrelated, but [there](https://github.com/jjensn/milight-decompile/blob/master/com/irainxun/wifilight/fragment/FUT028ColorFragment.java) goes the theory that they're doing anything sane to construct the UDP packets. Messy byte arrays like the rest of us scrubs.

Lstt2005, 2017-04-05:
https://github.com/jjensn/milight-decompile/blob/555a38808e78106db9954988e20a1ca42c3c4ef8/com/irainxun/wifilight/apsetting/net/UdpUnicast.java
https://github.com/jjensn/milight-decompile/blob/555a38808e78106db9954988e20a1ca42c3c4ef8/com/irainxun/wifilight/apsetting/net/UdpBroadcast%24ReceiveData.java
https://github.com/jjensn/milight-decompile/blob/555a38808e78106db9954988e20a1ca42c3c4ef8/com/irainxun/wifilight/apsetting/utils/Constants.java

sidoh, 2017-04-09:
Alright, I poked at it a little more thoroughly, and I'm feeling disheartened again. 

For background, the connection process looks like this:

P1. (Client -> Server): send search packet via UDP broadcast on port 5987
P2. (Server -> Client): send search response
P3. (Client -> Server): send session start request
P4. (Server -> Client): send session ID, to be used with all further requests

The challenge is:

1. In order for a single device to serve multiple IDs, there has to be a way for the client to specify which one it means to control.
1. I've tried two rough approaches with this.
  A. The search response packet contains a field for port. Changing this field in the response causes the client to send the session start request (P3) to the new port, but further requests are on port 5987, and the session started on the other port is never used again as far as I can tell.
  B. I had thought that P3 includes some scrutable identifier (this is why I felt positive in my last update), like the mac address of the device it was trying to start a session with, but it's the same packet every time. Not sure why I thought that it was any other way.
1. I could be wrong about some of my assumptions above, but if they're all true, I don't think it's possible to do this. :(

I poked around the decompiled source for a bit, but it's pretty hard to follow.

Have you used HomeAssistant or something like that, @Lstt2005? It should integrate nicely with this project and I think has most of the same controls that the Milight app does.

Lstt2005, 2017-04-09:
Thank you very much for your attempts to understand. Apparently, you will have to go in a different direction ..

sidoh, 2017-04-09:
Haha, nothing to be sorry for. I had fun digging into it.

Could always write a phone app that leverages the multiple ports!

Lstt2005, 2017-04-09:
I agree. Simply this ready-to-use application was very pleased by the design and functionality, so it's native. Unfortunately, in the Android studio I'm just beginner ..

sidoh, 2017-04-09:
I agree - was pleasantly surprised with it as well!

Lstt2005, 2017-04-09:
I>I could be wrong about some of my assumptions above, but if they're all true, I don't think it's possible to do this. :(

Can I help you on this situation than anything (with my beginner level)?

sidoh, 2017-04-09:
If you could find the place in the decompiled app where it's sending/receiving each of the above packets (P1..P4), we could probably tell for sure whether it's possible or not.

Lstt2005, 2017-04-09:
Ok, i will try...

Lstt2005, 2017-04-09:
>Have you used HomeAssistant or something like that, @Lstt2005?

They NO support multiples ID ..

sidoh, 2017-04-09:
But they have support for multiple ports :)

This is basically the entire point of esp8266_milight_hub being able to spin up multiple UDP servers on different ports, each bound to its own device ID.

Lstt2005, 2017-04-18:
> If you could find the place in the decompiled app where it's sending/receiving each of the above packets (P1..P4), we could probably tell for sure whether it's possible or not.

Is it pcap file maybe help you ?

Android ip 192.168.0.115, milight gateway 192.168.0.138, i just launch Milight and press @Searching for device"...

sidoh, 2017-04-18:
Packet captures are basically what I've been working with here. Appreciate you providing one, but looks pretty much like the ones I've been dealing with.

Lstt2005, 2017-04-20:
Maybe i found, what you say..) I analyze pakets and found, that in each session, Milight App send number of udp port, on wich it connect to ESP8266 to 5987 port in hex. Please, see attachment in doc.
[PacketAnalize.docx](https://github.com/sidoh/esp8266_milight_hub/files/945217/PacketAnalize.docx)

sidoh, 2017-04-21:
Yeah that's what I was noticing as well. This is the response to the broadcasted search request.

The problem I noticed was that the client seemed to start the session on the discovery port (5987), and use the session ID it was assigned there when sending packets to the custom port. This means you can't tell which session is attached to which port.

Lstt2005, 2017-04-21:
((((

closed 2017-04-21 by Lstt2005

sidoh, 2017-04-21:
:(

Hopefully it becomes possible (or more doable) in the future!