sidoh/esp8266_milight_hub #45: compilation error

turuwhiska, 2017-04-10:
hi i tried compile in atom platformio and have this error:
src/main.cpp:5:16: fatal error: fs.h: No such file or directory
#include <fs.h>
^
compilation terminated.
*** [.pioenvs/nodemcu/src/main.o] Error 1

sidoh, 2017-04-10:
It looks like you might be on an older version. Can you make sure the code is up to date?

turuwhiska, 2017-04-10:
Ah ok, it's possible. I am using the compiled version of the sources for debian 32 bit
tomorrow i will test in windows the last version 
thanks so much

sidoh, 2017-04-10:
Sure thing.

Pre-compiled images for each version can be found on the releases page too, if that suits your needs:

https://github.com/sidoh/esp8266_milight_hub/releases

turuwhiska, 2017-04-10:
yes of course i test the bin for esp-07 and perfect!!!

sidoh, 2017-04-11:
Were you able to get it to compile?

sidoh, 2017-04-13:
I'm going to close this ticket, but feel free to reopen if you're still having trouble.

closed 2017-04-13 by sidoh