sidoh/esp8266_milight_hub #46: Is it possible to manually change the MAC ID ?

Lstt2005, 2017-04-10:
Is it possible to store different mac address in string with configuration in index.html with each udp server? (For BilIght android app)?

sidoh, 2017-04-11:
Certainly possible. I'm a little hesitant to make things over-configurable, though. Don't want to make things unnecessarily confusing.

What do you think about responding with something like:

```
00:00:00:00:11:22
```

When `0x1122` is the device ID bound to the given port? Do you feel like that'd accomplish what you want?

Lstt2005, 2017-04-11:
Yes, I think that would suit me. Thank you!

Lstt2005, 2017-04-11:
And whether it is possible to track the variant, for example on port 5987, if a connection is established, then the next request from the application (MiLight)  is responsible for assigning a MAC address, for example ID + 1, to the next request from the ID + 2 application, etc., to try to deceive the application (Milight) (only on port 5987)?

sidoh, 2017-04-11:
Yeah I guess that's possible, but it seems like that'd be super unreliable and confusing. What should happen if, for example, one session expires or disconnects?

Lstt2005, 2017-04-11:
Yes, I agree (( .. I want to try to scan the application for requests and answers, it may be possible to find all the same semblance of the Id session ..

sidoh, 2017-04-11:
Go for it! Would love it if you figured out a way to do it. :)

I just pushed a change to v1.2.0 that does this.

Lstt2005, 2017-04-11:
>I just pushed a change to v1.2.0 that does this.
It's work with Bilight - https://github.com/JasperG/bilight (https://play.google.com/store/apps/details?id=com.jaspergoes.bilight) !

sidoh, 2017-04-11:
sweet. :)

sidoh, 2017-04-12:
Gonna close since it sounds like this worked, but feel free to re-open if I missed something.

closed 2017-04-12 by sidoh

alexeinz, 2017-04-25:
I wonder if the change works, Openhab also requires a specific mac address for discovery and I cant seem to make it work
00000000B55D for example doesnt work

sidoh, 2017-04-25:
Got it. I was a little worried about that. It seemed nicer to advertise meaningful information (i.e., the device ID) in the response rather than meaningless information (i.e., the ESP's hardware address), but certainly not at the expense of breaking stuff.

I suppose we could expose a setting. @Lstt2005, how useful is this feature to you? Would it break your workflow we responded with the hardware address instead of the device ID?

alexeinz, 2017-04-25:
atleast for openhab, each gateway requires a separate mac address, there seem to be no other way to add the device in openhab ( unless using mqtt,but that would complicate things quite a bit )

sidoh, 2017-04-25:
@alexeinz, do you happen to know what's causing OpenHAB to break? Asking because I'm concerned the issue is something else. I'd be a tiny bit surprised if OpenHAB was actually verifying the hardware address of the responding IP via ARP or something. It seems a little brittle.

sidoh, 2017-04-25:
> atleast for openhab, each gateway requires a separate mac address, there seem to be no other way to add the device in openhab ( unless using mqtt,but that would complicate things quite a bit )

Do you have multiple gateways running with the same device ID?

alexeinz, 2017-04-26:
no,the gateways all have separate device ID ( the device ID of the wifi boxes that the bulbs originally paired to ) 
Even though entered in Gateway servers, they are not discovered by openhab ( I do adjust the ports ) because openhab requires associated mac address,resolvable. See screenshot below.

It did manage to discover one gateway when I added the hardware mac address of the esp,but unreliably and that means i have to fully pair all bulbs to esp, which kind of defeats the purpose of the Gateway servers setting

[image]

[image]

sidoh, 2017-04-26:
Ah, so it looks like this mechanism is optional. Does it not work if you just leave off the Bridge ID?

alexeinz, 2017-04-26:
Well it is optional, but then it forces the bridge into v2 compatibility mode, and v6 devices will not function at all

sidoh, 2017-04-26:
Sorry that I'm fumbling around a bit here. I've not used OpenHAB before.

That's unfortunate. Is it not possible to just add a gateway with an IP and a port, and tell OpenHAB that it's a v6 hub? That's basically how it works for HomeAssistant.

alexeinz, 2017-04-26:
Yes, unfortunately this seems to be the functionality,
i think there is also open issue talking about same problem.
That said, this specific thread mentioned earlier creation of virtual mac addresses- tied into the virtual gateways... I had the impression this was implemented or you have decided against it in the end?

sidoh, 2017-04-26:
To confirm - it's not possible to add a device without its version being set by the discovery mechanism?

It's worth noting that Milight's approach to discovery is extremely confused. The discovery protocol outlined in the [documentation](http://www.limitlessled.com/dev/) is actually only used by the official Milight app when configuring wifi settings on a new device, and is not implemented in esp8266_milight_hub. Under normal circumstances, the Milight app discovers new devices by broadcasting a few packets to port 8899, so I forwent implementing the "real" discovery mechanism.

It's certainly possible to implement the discovery protocol, and to reply with multiple arbitrary packets. Since I'm pretty fuzzy on how OpenHAB works, though, I could use some help in knowing what packets need to get sent.

Poking through OpenHAB's milight code, it looks like it's determining the version [here](https://github.com/openhab/openhab2-addons/blob/master/addons/binding/org.openhab.binding.milight/src/main/java/org/openhab/binding/milight/internal/protocol/MilightDiscover.java). Would it be sufficient to reply with multiple search results on port 48899, one for each UDP server?

For example, if you had an ESP with IP 10.0.0.100, and UDP servers on ports 800,801,802, bound to device IDs 0x1,0x2,0x3 respectively, do you think it'd be sufficient to reply with three packets as follows?

```
10.0.0.10,000000000001,HF-LPB100
```
```
10.0.0.10,000000000002,HF-LPB100
```
```
10.0.0.10,000000000003,HF-LPB100
```

alexeinz, 2017-04-26:
Yes, it seems that this one here is the key to discover v3 and v6 
I am not sure what is the 3rd argument though.. would that be the host-id?

If you could make the change.. I can test it easily when I am back home in the evening.

            // Now loop forever, waiting to receive packets and printing them.
            while (!willbeclosed) {
                packet.setLength(buffer.length);
                datagramSocket.receive(packet);
                // **example: 10.1.1.27,ACCF23F57AD4,HF-LPB100**
                String[] msg = new String(buffer, 0, packet.getLength()).split(",");
                if (msg.length >= 2 && msg[1].length() == 12) {
                    // Stop resend timer if we got a packet.
                    if (resendTimer != null) {
                        resendTimer.cancel(true);
                        resendTimer = null;
                    }
                    // Determine version: Version 6 sends a third argument != ""
                    int version = msg.length >= 3 && msg[2].trim().length() > 0 ? 6 : 3;
                    // Notify all observers
                    discoverResult.bridgeDetected(((InetSocketAddress) packet.getSocketAddress()).getAddress(), msg[1],
                            version);
                } else {
                    logger.error("Unexpected data received {}", msg[0]);
                }

sidoh, 2017-04-26:
According to the documentation, the third argument is just the string literal `HF-LPB100`. I guess maybe the old protocol doesn't include the third argument.

I'm happy to give it a go, but I'm a little worried that it's not going to work. See #71 for more detail.

alexeinz, 2017-04-26:
yes , this should work , that string is the identifier
i believe here is complete info on what it expects to see to identify hub type
 https://github.com/home-assistant/home-assistant/issues/4676

Sent from my cell

> On Apr 26, 2017, at 13:45, Chris Mullins <notifications@github.com> wrote:
> 
> HF-LPB100