sidoh/esp8266_milight_hub #48: Integration with Smartthings

fireboy1919, 2017-04-13:
I have written an integration for this project that works with this project over here:  https://github.com/fireboy1919/MiThings

And it works!  Works great, actually.  Unfortunately, though, Smartthings hub is super buggy, and I had to modify one thing.

Specifically, Smartthings doesn't support PUT requests.  I don't know why.  Same exact command as post, but with a different verb, and it doesn't go out.  I have a bug request in and will be ready if they ever decide to fix it.

In the meantime, I added an extra line to my compiled version of this project to support POST requests for sending bulb commands.  Seems to work fine.

How would you feel about adding or changing that command to a POST rather than a PUT?

sidoh, 2017-04-13:
I think I've used PUT with SmartThings before. Is the issue that HubAction doesn't support it?

PUT feels like the "right" HTTP verb given the behavior. I'm not opposed to adding POST, but I'd prefer not to replace PUT.

fireboy1919, 2017-04-13:
I agree that it is the right verb.  Yes, the issue is that HubAction doesn't go out at all when used with PUT.

I've tried it on both the device and smart apps; same problem.  HubAction is GET or POST only right now.

sidoh, 2017-04-13:
Gotcha, makes sense. I've been pretty unimpressed with HubAction :(

Adding support for POST in existing PUT routes sounds fine to me.

fireboy1919, 2017-04-13:
Oh, and btw, this is the only working Smartthings integration with Milight.  The other one was designed to use an external internet service, and the guy who wrote it shut it down.

sidoh, 2017-04-13:
I use Milight stuff with SmartThings as well with my [ha_gateway](https://github.com/sidoh/ha_gateway) project.

sidoh, 2017-04-14:
Added support for POST with c15ee923a4ac938cfbbdaa9818cf1c6e95b40ff7. Will be released with v1.2.0. Wrapping up a couple of features before releasing. Released [1.2.0-dev4](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.2.0-dev4) with this feature.

closed 2017-04-22 by sidoh