sidoh/esp8266_milight_hub #51: Support uPNP

fireboy1919, 2017-04-14:
It would be nice to have a discovery mechanism that works with other apps.

I suggest uPNP.

Here's an example of it in use for arduino.  Looks like it's relatively straightforward to add.
https://github.com/dannybackx/arduino-upnp

sidoh, 2017-04-14:
I was thinking about UPnP too. I happen to be pretty familiar with it because I wrote a [ruby upnp client](https://github.com/sidoh/easy_upnp).

I guess my hesitations with UPnP are:

1. It's pretty heavy. Take a look at [all of the supporting source](https://github.com/dannybackx/arduino-upnp/tree/master/libraries/UPnP) in the example you linked.
2. The service discovery part of UPnP (SSDP) is only small part of implementing a UPnP service, and I'm not sure how useful SSDP on its own is.

Probably important to clarify the goal here. What should we be enabling? 

To me, it seems like it'd be a lightweight, platform-independent way to discover the IP of the ESP. I feel like the rest should be handled by the REST API.

Does that sound right to you?

fireboy1919, 2017-04-14:
Yep.  That sounds about right.  That's pretty much it.

sidoh, 2017-04-15:
Cool, sounds good. 

I poked around a bit, and I don't think there's an awesome solution. 

It looks like SSDP as you described it actually will work for Windows. The device would show up in some network discovery window. I don't have a Windows machine, so it's a little hard for me to test. I might look into getting a VM.

For OS X, looks like mDNS/bonjour is the preferred solution.

For Linux, probably either of these things will work, but not out of the box.

Given that both SSDP and mDNS are baked-in to the ESP8266 Arduino platform, it's probably not hard to just do both. Gonna update the ticket title with that.

sidoh, 2017-04-16:
Merged support into the 1.2.0 branch.

@fireboy1919, if you were wanting to use this with your SmartThings app, it looks like you might be able to use mDNS:

https://github.com/rappleg/SmartThings/blob/master/smartapps/smartthings/mdns-discovery-example.groovy

I expect SSDP is roughly the same level of effort. Either should work.

fireboy1919, 2017-04-16:
That sounds like a good idea.  I'll go with mDNS.

Lstt2005, 2017-04-18:
>Get IP Address

Both mDNS and SSDP are supported.

OS X - you should be able to navigate to http://milight-hub.local.
Windows - you should see a device called "ESP8266 MiLight Gateway" show up in your network explorer.

On Windows - not work..

sidoh, 2017-04-18:
Can you provide more details? It's not showing up in Windows explorer? I tested it with a Windows 10 VM.

sidoh, 2017-04-20:
Here's what it looks like for me:

[image]

Lstt2005, 2017-04-20:
Ups..I sorry. My mistake - i search in  "Devices and Printers". It's work!

PR #53 (2017-04-16): Implement mDNS and SSDP
Fixes #51
closed 2017-04-22 by sidoh