sidoh/esp8266_milight_hub #56: Add support of static IP in WifiManager

WoodsterDK, 2017-04-19:
Option for setting static IP could be nice.

Lstt2005, 2017-04-20:
Yes,  I also agree with this.

sidoh, 2017-04-29:
Looks like the method `WiFiManager` exposes for a static IP is here:

https://github.com/tzapu/WiFiManager#custom-station-client-static-ip-configuration

You'd have to configure a static IP, gateway, and netmask. Is that along the lines of what you'd hope for?

sashsc, 2017-12-06:
That sounds good. Maybe it is also possible to specify a "fallback" WLAN. That means, no connection with the 1st WLAN, then please take the 2nd WLAN.

arjena, 2018-03-23:
Tried setting a static IP following the examples from tzapu, but no cigar. I must be doing something wrong, but I am not experienced enough to see what. I added the line `wifiManager.setSTAStaticIPConfig(IPAddress(192,168,0,99), IPAddress(192,168,0,1), IPAddress(255,255,255,0));` (IP's are correct for my network) after `wifiManager.autoConnect(ssid.c_str(), "milightHub");`. But since I don't understand what WifiManager does exactly it's kind of hard to troubleshoot... The esp keeps getting an IP address from the router.

bombcheck, 2018-03-23:
Look at the example here:
https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithStaticIP/AutoConnectWithStaticIP.ino
I think you must set the IP config BEFORE connecting.

arjena, 2018-03-23:
I'll try that and let you know if that works. Thanks!

arjena, 2018-03-23:
Yep, that was it. Now my bridge is connected on the IP I set it to. 
FWIW, I use a NodeMCU (V0.9) for the hub and had to use the 'block 1' method. 
Thanks again.

iMartyn, 2018-10-22:
I'll have a go at working up a pull request for this, as the dhcp server in my provider's router doesn't really do the "reserved dhcp" thing that I would otherwise use and I'm running domoticz in kubernetes and can't forward the avahi name into the k8s network.  No guarantees on speed though, as it'll be spare-time work.

iMartyn, 2018-10-22:
I'll try and work up a pull request for this as I want this functionality too.  No guarantees on speed as it's a spare-time thing.
FWIW my use-case is my home automation software (domoticz) runs in a different network (in kubernetes) and there's no way to send the avahi name through, so a static IP is about all I can do, and my router (half-braindead thing from my provider that I can't change) won't allow reservations in DHCP.

sidoh, 2019-03-31:
This is released in 1.9.0-dev11.  Thanks for @iMartyn for doing the hard part. :)

closed 2019-05-28 by sidoh