sidoh/esp8266_milight_hub #58: Allow webapp to run on Apache etc. with USB NRF24L01+

spmp, 2017-04-20:
Allow webapp to run on Apache etc. with USB NRF24L01+, such that any network connected device with USB could potentially become a gateway (rPi or other persistent server device on the network). Also reduces the WiFi device count 8)

sidoh, 2017-04-21:
While I love the idea, this project is pretty geared towards an ESP8266. 

One of my goals is to document the packet structure, which would make it easier for other projects pointed at different platforms :)

spmp, 2017-04-21:
I guess I am essentially asking for the two levels of separation within the project to be apparent:
1) A library that constructs the commands which drive the lights such that a standalone command may operate
2) The webservice that (may) use the library to control the lights
and what is now
3) A specific implimentation for esp8266

One day maybe...

In the meantime I will get soldering my esp12f's and use curl etc from the server to control the lights... 
Cant wait to try it 8)

sidoh, 2017-04-21:
I'm following :)

This project is very tied to the Arduino platform. I'm not sure there's a sane way to modularize components like this and have them still be usable with Arduino. If I was going to write some general library to translate commands into RF packets, I'd probably want to write it in something like ruby or python.

I think the general idea is great, but it'd probably have to be out of band from this project.

spmp, 2017-04-21:
Cheers,
 When I have had more of a play with the project as it is I can give you a hand to start on this. Surely seeing as Arduino is C with some + a C library that gets wrapped for Arduino is the way to go for the lower level stuff?
Anyway, nuff from me until I have been dirty in the code 8)

sidoh, 2017-04-21:
Haha, I'm sure it's possible. I guess I'm weighing it with the pains of gutting Arduino platform and library dependencies everywhere possible, and of maintaining a far more abstracted codebase. Honestly though (and I could be way wrong here), the biggest thing is that a C codebase seems like it's much harder than necessary to use if you want to run something on a (relatively) beefy piece of hardware like a Pi. If I were a user wanting to do that, I'd much prefer some ruby/python library.

It'd be quite easy to port the packet buffers to an interpreted language. One of the hardest parts of this project was cracking the scrambling used with the newer Milight bulbs, and that effort is completely translatable :)

spmp, 2017-04-21:
Yup, gotcha 8)

g-rom-info, 2017-04-25:
This is completely off topic but you say your code is intended for Arduino. Do you think that it is possible to execute your code on an Arduino Uno or Mega with an Ethernet shield (by removing of course the wifimanager)? To be honest with you I have not (yet) tested to compile :-)
The idea is to have a system that does not depend on the WiFi network.

WoodsterDK, 2017-04-25:
Without much hassle you can use a serial connection from the ESP8266 and attached it to an arduino, RPi if you want.

g-rom-info, 2017-04-25:
That's the other possibiliy that I think :-)  
However most automation solutions for Milight uses the IP stack (like domoticz) that's why Wired Ethernet will be a better choice.
When I will have sufficient time I will try to adapt an ENC28J60 ethernet controller to the ESP. [(see this](https://github.com/Cicero-MF/esp_enc28j60))

/OffTopic

sidoh, 2017-05-06:
Love the idea of supporting different kinds of use cases, but gonna close this one out as out of scope. This project is definitely pretty tied to the ESP8266.

To answer this question:

> Do you think that it is possible to execute your code on an Arduino Uno or Mega with an Ethernet shield (by removing of course the wifimanager)?

I don't think it'd work with other platforms without at least minor modifications. I don't know how much of the code is ESP8266 specific, but there is some amount of it (`ESP.restart()`, etc.). Also, a base install takes ~40KB of RAM, which is way more than fits on a ATmega. A lot of that is probably WiFi code, but I think it'd be a real challenge to strip it down enough to fit on a ATmega.

Using the ESP8266 ethernet hack you linked above sounds like a more feasible option.

closed 2017-05-06 by sidoh