sidoh/esp8266_milight_hub #62: Disco Mode Not working

g-rom-info, 2017-04-22:
Hi Chris,

As mentioned on other posts, i'm testing your 1.2.0 version.
The Disco mode does not work as expected comparing to the original bridge.

On the 7 bytes sequence, I think you have forgotten to increment the first byte on 0xB0 => 0xB8 range

(Although Disco mode works properly with RGB+CCT bulbs)

Rgds,
Jerome

sidoh, 2017-04-22:
Yeah I guess I was trying to avoid it keeping state, but might as well add it. You can set the mode directly.

g-rom-info, 2017-04-22:
Thank you for your reply. Moreover, Increment of Disco Mode does not work properly on the webui (my bulbs start over on the same disco mode...)

sidoh, 2017-04-23:
Should be fixed in [1.2.1](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.2.1). :)

g-rom-info, 2017-04-23:
I will test it tomorrow and give you a feedback !

g-rom-info, 2017-04-23:
Update : it's Ok from the webui. I will test it from UDP Gateway tomorrow :)

g-rom-info, 2017-04-25:
Unfortunately, it is not ok for Disco mode...

The Disco code stays to "0xB0"

```
[MiLightUdpServer port 8842] - Handling packet: 47 00
                                                      Sending packet: B0D24E0002058B
                                                                                    Elapsed: 671
[MiLightUdpServer port 8842] - Handling packet: 4D 00
                                                      Sending packet: B0D24E00020D8C
                                                                                    Elapsed: 669
[MiLightUdpServer port 8842] - Handling packet: 47 00
                                                      Sending packet: B0D24E0002058D
                                                                                    Elapsed: 670
[MiLightUdpServer port 8842] - Handling packet: 4D 00
                                                      Sending packet: B0D24E00020D8E
                                                                                    Elapsed: 669
[MiLightUdpServer port 8842] - Handling packet: 47 00
                                                      Sending packet: B0D24E0002058F
                                                                                    Elapsed: 669
[MiLightUdpServer port 8842] - Handling packet: 4D 00
                                                      Sending packet: B0D24E00020D90
                                                                                    Elapsed: 667
```

sidoh, 2017-04-29:
Thanks for catching that. Should be fixed in

https://github.com/sidoh/esp8266_milight_hub/tree/v5_udp_fixes

g-rom-info, 2017-05-01:
Yes it is working :)
However, bug fixes for RGBW all on/off backwards and CCT fixes had not been implemented to this branch.
Is it normal ?

sidoh, 2017-05-02:
Cool, thanks! 

Just needed master merged in.

sidoh, 2017-05-05:
Thanks for raising, @g-rom-info!

closed 2017-05-05 by sidoh