sidoh/esp8266_milight_hub #63: HTTP request in order to set specific disco mode

g-rom-info, 2017-04-22:
Hi Chris,

I suggest you an enhancement :-)

On the RGBW protocol, the first byte is the type of Disco mode (from 0xB0 to 0xB8)
Is it possible to implement a function in order to set a specific mode instead of classic mode (increment) ?

Moreover Is it possible to implement this mode for CCT bulbs ?

Best regards,
Jerome

sidoh, 2017-04-22:
It should already do this:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/RgbwPacketFormatter.cpp#L37
https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/RgbCctPacketFormatter.cpp#L67

The HTTP parameter is `mode`. Should be documented in the README.

g-rom-info, 2017-04-22:
My bad !
I confirm that it works properly !
curl --data-binary '{"mode":9}' -X PUT http://192.168.18.12/gateways/0xD24E/rgbw/0

Sorry for this undesirable ticket. As a punishment, I will wrote 100 times "RTFM"... :)
Regards,
Jerome

closed 2017-04-22 by g-rom-info

sidoh, 2017-04-23:
Haha no worries, it's kind of buried in there :)