sidoh/esp8266_milight_hub #65: UDP server "ALL GROUP ON/OFF" are backwards

phigmeta, 2017-04-24:
If you send the UDP signal for "ALL ON" (42 00 55 - All On) it turns everything off

If you Send the UDP signal for "ALL OFF" (41 00 55 - All Off) it turn everything on

Group UDP functions correctly:

45 00 55 - Zone 1 On
46 00 55 - Zone 1 Off
47 00 55 - Zone 2 On
48 00 55 - Zone 2 Off
49 00 55 - Zone 3 On
4A 00 55 - Zone 3 Off
4B 00 55 - Zone 4 On
4C 00 55 - Zone 4 Off

Version 5

Tested with app first, then with UDP Sender...

Used RGBW Bulbs with Zone 1 and RGBW LED Driver with Zone 2 

No further devices tested as I don't have any

g-rom-info, 2017-04-24:
According to [limitless documentation](http://www.limitlessled.com/dev/), you are right !

> LimitlessLED RGBW
> -------------------
> 41 00 55 - All Off
> 42 00 55 - All On
> 43 00 55 - Speed Down (One Step Slower Disco)
> 44 00 55 - Speed Up (One Step Faster Disco)
> 45 00 55 - Zone 1 On
> 46 00 55 - Zone 1 Off
> .........

It can be easily corrected by editing vars on V5MiLightUdpServer.h

Replace
```
  UDP_RGBW_ALL_ON            = 0x41,
  UDP_RGBW_ALL_OFF           = 0x42,

```
By
```
  UDP_RGBW_ALL_ON            = 0x42,
  UDP_RGBW_ALL_OFF           = 0x41,
```

sidoh, 2017-04-24:
Good find.

Happy to look at a PR if someone wants to toss one up. Otherwise I'll get to it over the next few days.

phigmeta, 2017-04-25:
PR sent as requested, would have done that earlier... but for some reason, I fail at git.

sidoh, 2017-04-25:
Thanks! Looks good. I'll test it with the bulbs I have later today.

g-rom-info, 2017-04-25:
Just tested it with my bulbs and this quick fix, it's OK :)

sidoh, 2017-04-29:
Gonna close this since it sounds like it's fixed. Thanks, @phigmeta!

closed 2017-04-29 by sidoh

g-rom-info, 2017-05-05:
@sidoh 1.2.3 version still have this issue :-(

sidoh, 2017-05-05:
Hmm... should've been fixed in 654b4f5380ae47ead5fe0d24cf7d0bba5a2b8e82. I'll test when I'm home. Maybe some later change swapped it back somehow.

g-rom-info, 2017-05-05:
Yes but not in the master branch :) [See here](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/Udp/V5MiLightUdpServer.h).

sidoh, 2017-05-06:
Argh. Crud. I must've done something goofy with git. I just manually reapplied it and released 1.2.4.