sidoh/esp8266_milight_hub #66: Sniffer mode not working on RGBW mode

g-rom-info, 2017-04-24:
Hi Chris,

It has a bug on the sniffer module.

After a cold start, when I start sniffing on rgbw mode, it does not receive any packet...
I need to start sniffing to another mode (CCT for example) and switch back to RGBW.

Tested with nRF24L01.

Best regards, 
Jerome

sidoh, 2017-04-25:
Hmm... seems to work okay for me. I think listening is sort of spotty in general. Resetting the NRF and/or the ESP usually seems to solve it. I'm hesitant to put too much effort into it because it's sort of a "use-once" feature to get the ID of a remote.

g-rom-info, 2017-04-25:
I understand, it is clearly not a priority :)

sidoh, 2017-04-29:
I'm wondering if this was caused by the same bug fixed in #73 (for the other issue you opened, #64). Could you try again after updating to 1.2.2?

g-rom-info, 2017-04-29:
Yes it's OK for the sniffing mode after a cold start :) (with my nRF24L01)

sidoh, 2017-05-05:
Gonna close since this sounds like it's fixed, but feel free to reopen if it's still an issue.

closed 2017-05-05 by sidoh