sidoh/esp8266_milight_hub #67: Multiple issues on CCT bulbs

g-rom-info, 2017-04-24:
Hi Chris,

Another feedback !
I just tested CCT bulbs and I have multiple issues with the V5 UDP Gateway:

Malformed packets on Brightness Up and Down commands:
```
Sending packet: 5A**FFFFFF**04CFCF
                              Elapsed: 128
Sending packet: 5A**FFFFFF**04D0D0
                              Elapsed: 116
Sending packet: 5AFFFFFF04D1D1
                              Elapsed: 129
Sending packet: 5AFFFFFF04D2D2
                              Elapsed: 121
Sending packet: 5AFFFFFF04D3D3
                              Elapsed: 125
Sending packet: 5AFFFFFF04D4D4
                              Elapsed: 117
Sending packet: 5AFFFFFF04D5D5
                              Elapsed: 115
```

And same thing for Color Temperature packets:
```
Sending packet: 5A**FFFFFF**0EDEDE
                              Elapsed: 132
Sending packet: 5A**FFFFFF**0EDFDF
                              Elapsed: 117
Sending packet: 5AFFFFFF0EE0E0
                              Elapsed: 112
Sending packet: 5AFFFFFF0EE1E1
                              Elapsed: 135
Sending packet: 5AFFFFFF0FE2E2
                              Elapsed: 114
Sending packet: 5AFFFFFF0FE3E3
                              Elapsed: 124
Sending packet: 5AFFFFFF0FE4E4
                              Elapsed: 115
```

Brightness and Color Temperatures are OK with the WebUI.

All On and All Off commands not recognized from UDP Gateway:
```
V5MiLightUdpServer - Unhandled command: 53
V5MiLightUdpServer - Unhandled command: 57
```
Same thing from the WebUI: malformed packets. (Groups 1 From 4 are OK.)

```
Sending packet: 5AD24E**0000**F0F0
                              Elapsed: 120
Sending packet: 5AD24E**0000**F1F1
                              Elapsed: 120
Sending packet: 5AD24E0000F2F2
                              Elapsed: 121
Sending packet: 5AD24E0000F3F3
                              Elapsed: 104
```
Expected Packets are:
ON ==> 5AD24E0009xyxy
OFF ==> 5AD24E0005xyxy

Best regards,
Jerome

sidoh, 2017-04-24:
Hmm... guessing it's a bug in commands with multiple packets. I'll take a look sometime soon.

sidoh, 2017-04-25:
Thinking this should fix it: f21ed220b5257a153c18dc8b01508a5966ab292c

sidoh, 2017-04-25:
And for not responding to all on/all off commands: 9b85dba4fdd25ce86a459c8ae4ce45eace93255a.

g-rom-info, 2017-04-25:
It's ok for color temperature and brightness :-)

However it's not OK for all groups on/off...

```
[MiLightUdpServer port 8842] - Handling packet: 39 00 55
                                                         Sending packet: 5AD24E00000303
       Elapsed: 133
[MiLightUdpServer port 8842] - Handling packet: 35 00 55
                                                         Sending packet: 5AD24E00000404

```

Moreover, I just see that the Night mode is not working too.

```
[MiLightUdpServer port 8842] - Handling packet: 39 00 55
                                                         Sending packet: 5AD24E00000202
       Elapsed: 134
[MiLightUdpServer port 8842] - Handling packet: B9 00 55
                                                         V5MiLightUdpServer - Unhandled command: 185
```

sidoh, 2017-04-25:
Thanks for catching that. Should be fixed in 7079031edeccadc21e2f6bfcb6733f4f8c798296.

g-rom-info, 2017-04-25:
Ur welcome !

It's OK for All groups On/Off (Bridge and WebUI)

For Night Mode it's ok only for all groups.
Apologize for this mistake, but night mode codes are different between all bulbs and groups...

> NIGHT MODE ALL		        0x39	100ms followed by:	0xB9
> NIGHT SAVER MODE GROUP 1	0x3B	100ms followed by:	0xBB
> NIGHT SAVER MODE GROUP 2	0x33	100ms followed by:	0xB3
> NIGHT SAVER MODE GROUP 3	0x3A	100ms followed by:	0xBA
> NIGHT SAVER MODE GROUP 4	0x36	100ms followed by:	0xB6

sidoh, 2017-04-29:
Ah. Think this is just a problem in v5 UDP. Should be fixed in this branch:

https://github.com/sidoh/esp8266_milight_hub/tree/cct_fixes

g-rom-info, 2017-05-01:
No it does not work :(
I'm unable to switch on/off lights with v5 UDP. (Web interface works)

Command packets are not OK with UDP, see below :

Switch On from WebUI : 5AD24E00054848
Switch Off from WebUI : 5AD24E00094949

Switch On from UDPv5 : 5AD24E00194A4A
Switch Off from UDPv5 : 5AD24E00194B4B

sidoh, 2017-05-02:
Can you include the UDP packets being sent? It looks like it's thinking they're nightmode.

g-rom-info, 2017-05-02:
Of course !

For Night Mode, 2 packets are sent (Off + Night), according to 2 UDP commands.

**On (All, 1, 2, 3, 4)**
```
[MiLightUdpServer port 8842] - Handling packet: 35 00 55
                                                         Sending packet: 5AD24E00190101
                                                                                       Elapsed: 661
[MiLightUdpServer port 8842] - Handling packet: 38 00 55
                                                         Sending packet: 5AD24E01080202
                                                                                       Elapsed: 662
[MiLightUdpServer port 8842] - Handling packet: 3D 00 55
                                                         Sending packet: 5AD24E02130303
                                                                                       Elapsed: 662
[MiLightUdpServer port 8842] - Handling packet: 37 00 55
                                                         Sending packet: 5AD24E031A0404
                                                                                       Elapsed: 662
[MiLightUdpServer port 8842] - Handling packet: 32 00 55
                                                         Sending packet: 5AD24E04020505
                                                                                       Elapsed: 663
```

**Off (All, 1, 2, 3, 4)**
```
[MiLightUdpServer port 8842] - Handling packet: 39 00 55
                                                         Sending packet: 5AD24E00190606
                                                                                       Elapsed: 663
[MiLightUdpServer port 8842] - Handling packet: 3B 00 55
                                                         Sending packet: 5AD24E011B0707
                                                                                       Elapsed: 663
[MiLightUdpServer port 8842] - Handling packet: 33 00 55
                                                         Sending packet: 5AD24E02130808
                                                                                       Elapsed: 664
[MiLightUdpServer port 8842] - Handling packet: 3A 00 55
                                                         Sending packet: 5AD24E030A0909
                                                                                       Elapsed: 663
[MiLightUdpServer port 8842] - Handling packet: 36 00 55
                                                         Sending packet: 5AD24E04060A0A
                                                                                       Elapsed: 661

```

**Night Mode (All)**
```
[MiLightUdpServer port 8842] - Handling packet: 39 00 55
                                                         Sending packet: 5AD24E00190C0C
                                                                                       Elapsed: 662
[MiLightUdpServer port 8842] - Handling packet: B9 00 55
                                                         Sending packet: 5AD24E00190D0D
                                                                                       Elapsed: 662
```

**Night Mode (Group 1)**
```
[MiLightUdpServer port 8842] - Handling packet: 3B 00 55
                                                         Sending packet: 5AD24E011B0F0F
                                                                                       Elapsed: 662
[MiLightUdpServer port 8842] - Handling packet: BB 00 55
                                                         Sending packet: 5AD24E011B1010
                                                                                       Elapsed: 663
```

**Night Mode (Group 2)**
```
[MiLightUdpServer port 8842] - Handling packet: 33 00 55
                                                         Sending packet: 5AD24E02131212
                                                                                       Elapsed: 663
[MiLightUdpServer port 8842] - Handling packet: B3 00 55
                                                         Sending packet: 5AD24E02131313
                                                                                       Elapsed: 663
```

**Night Mode (Group 3)**
```
[MiLightUdpServer port 8842] - Handling packet: 3A 00 55
                                                         Sending packet: 5AD24E030A1515
                                                                                       Elapsed: 664
[MiLightUdpServer port 8842] - Handling packet: BA 00 55
                                                         Sending packet: 5AD24E030A1616
                                                                                       Elapsed: 662
```

**Night Mode (Group 4)**
```
[MiLightUdpServer port 8842] - Handling packet: 36 00 55
                                                         Sending packet: 5AD24E04061818
                                                                                       Elapsed: 662
[MiLightUdpServer port 8842] - Handling packet: B6 00 55
                                                         Sending packet: 5AD24E04061919
                                                                                       Elapsed: 661
```

g-rom-info, 2017-05-02:
Expected packets (from the original bridge)

**On (All, 1, 2, 3, 4)**
```
Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 00
b2            : 05
b3            : 09
Sequence Num. : 8F

Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 01
b2            : 08
b3            : 0A
Sequence Num. : 94

Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 02
b2            : 0D
b3            : 0B
Sequence Num. : 9B

Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 03
b2            : 07
b3            : 0C
Sequence Num. : 97

Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 04
b2            : 02
b3            : 0D
Sequence Num. : 94
```

**Off (All, 1, 2, 3, 4)**

```
Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 04
b2            : 09
b3            : 18
Sequence Num. : A6

Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 04
b2            : 0B
b3            : 17
Sequence Num. : A7

Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 04
b2            : 03
b3            : 16
Sequence Num. : 9E

Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 04
b2            : 0A
b3            : 15
Sequence Num. : A4

Packet received (7 bytes):
Request type  : 5A
Device ID     : D24E
b1            : 04
b2            : 06
b3            : 14
Sequence Num. : 9F
```

Unfortunately I'm unable to sniff Night mode packets (I suppose Off and Night packets are sent to fast between us)

sidoh, 2017-05-03:
Ah. I'm wondering if it was a shitty order of operations bug: a71cf5f9c95c53e94e3c97a05d7afae706f615d6

Can you try the latest commit in that branch? I'm away from home at the moment and unable to test myself.

g-rom-info, 2017-05-04:
It's good ! 👍 This bug was not trivial...

sidoh, 2017-05-04:
Cool. Was that it on this one? If so I'll merge it in and release a new patch version

g-rom-info, 2017-05-04:
Yes it was this one (add parenthesis). You can merge it next release ;)

sidoh, 2017-05-04:
Cool! :D

sidoh, 2017-05-05:
Thanks for raising, @g-rom-info!

PR #77 (2017-05-05): Fix bugs in v5 UDP command handler
Fixes #67 and #62
closed 2017-05-05 by sidoh