sidoh/esp8266_milight_hub #70: Error in master brach?

Lstt2005, 2017-04-25:
I try work with LT8920, it's work very stable with old branch - esp8266_milight_hub-mqtt from 20.04.2017.
But when i try work with master branch from now time - LT8920 work VERY UNSTABLE with different settings RF packets etc...(((

sidoh, 2017-04-25:
Can you clarify what you mean by unstable? Would also be helpful if you could share the older revision you're using that works.

Lstt2005, 2017-04-25:
I mean, that my LED controller FUT035 work with master branch from time to time, I can not understand the patterns...with the different settings WEB UI, from web and from UDP Application..
https://cloud.mail.ru/public/4L1v/S7Z99tM6y

sidoh, 2017-04-25:
Are you able to look at what revision you're flashing it with? From my local version of the branch it looks like 38f8fce169edb0e1f4923ccf6d7b24b30a42e910.

sidoh, 2017-04-25:
There weren't many meaningful changes since the last commit to the mqtt branch as far as I can tell :\

https://github.com/sidoh/esp8266_milight_hub/network

Lstt2005, 2017-04-25:
I flashing library via Platformio to my D1_mini current brach from git, and flashing https://cloud.mail.ru/public/4L1v/S7Z99tM6y from you repo from 20.04.2017.

sidoh, 2017-04-25:
Obviously I trust that this is happening, I'm just a little confused as to why :)

Thanks for sharing the code you're flashing it with, but it'd be most helpful to get the revision. You can use `git rev-parse HEAD` to do that:

```
$ git rev-parse HEAD
38f8fce169edb0e1f4923ccf6d7b24b30a42e910
```

sidoh, 2017-04-25:
Could you also paste the output of `/about`? Should look something like this:

```
{"version":"1.2.1-1-gf21ed22","variant":"d1_mini","free_heap":18968}
```

Lstt2005, 2017-04-25:
Can you clarify the beginner when i do this in Windows :)) ?

sidoh, 2017-04-25:
Ah, are you using a Windows client? There's probably a "log" option or something and you'd just look at the most recent revision.

The output of `/about` might be enough, though.

Lstt2005, 2017-04-25:
Unfortunately, I trivially downloaded the library as a zip file, and not through the client .. (

sidoh, 2017-04-25:
Ah, gotcha. Guessing we're on the same branch then.

Can you paste the output of `/about`? Should look like this:

[image]

Lstt2005, 2017-04-25:
{"version":"UNKNOWN","variant":"d1_mini","free_heap":16416}

Lstt2005, 2017-04-25:
When i flash esp8266, i see in log:
esp8266_milight_hub-master> pio run --target uploadfs
[04/25/17 21:27:30] Processing d1_mini (build_flags: !python .get_version.py -Wl,-Tesp8266.flash.4m1m.ld -D FIRMWARE_VARIANT=d1_mini; lib_
deps: SPI, , RF24, WiFiManager, ArduinoJson, PubSubClient, https://github.com/ratkins/RGBConverter; platform: espressif8266; upload_speed:
 921600; board: d1_mini; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
Building SPIFFS image from 'data' directory to .pioenvs\d1_mini\spiffs.bin
/web/index.html
Looking for upload port...
Auto-detected: COM44
Uploading .pioenvs\d1_mini\spiffs.bin
Uploading 1028096 bytes from .pioenvs\d1_mini\spiffs.bin to flash at 0x00300000
................................................................................ [  7% ]
................................................................................ [ 15% ]

sidoh, 2017-04-25:
Yeah should be fine. I'll try to replicate in the next few days.

sidoh, 2017-04-29:
I've been playing around with the LT8900 again tonight, and it seems pretty reliable. I just switched my "production" milight hub over to an LT8900, so hopefully I'll notice any issues through daily use.

sidoh, 2017-06-03:
LT8900 seems pretty stable in my setup. I'm going to close this, but please feel free to re-open if I'm missing something.

closed 2017-06-03 by sidoh