sidoh/esp8266_milight_hub #72: Is it possible implement new OEM (partically Milight) protocol?

Lstt2005, 2017-04-27:
I recieve yesterday some RGBW OEM China LED plate - https://detail.1688.com/offer/522143371857.html?spm=a2615.2177701.0.0.hO79R9
When i sniff it, i see this syncword 0x0a0A 0x55AA and channels 0x04 (4) and 0x4A( 74), coomand with 7 byte payload, like 0xC2 0xDC 0xA0 0xAB 0x01 0xBE 0x00 (Power On All), 0xC2 0xDC 0xA0 0xAB 0x02 0xC1 0x00 (power Off ALL).
I believe that this is very similar to the description of the old MiLigh protocol - http://arduino-projects4u.com/milight-rf-control/
Can you help me for it's implemention?

sidoh, 2017-04-27:
Sure, if you can figure out the command structure I can take a look.

Lstt2005, 2017-04-27:
Ok, thanks. But only after the 2 of May..)

tbrasser, 2017-05-11:
it's probably one of the controllers on this page: http://www.futlight.com/productlist.aspx?typeid=146&page=1

itProfi, 2019-07-06:
Hi, Chris!
Yesterday, i again try investigate those packets with some modificatios on MilightRadioConfig.cpp

sidoh, 2019-07-06:
Compile with DEBUG_PRINTF on.  Listening with LT8900 is pretty broken honestly.  Would recommend using the nRF24.

itProfi, 2019-07-07:
I sniff all button from my remote via nRF24:
[image]
[oem_milight.txt](https://github.com/sidoh/esp8266_milight_hub/files/3365735/oem_milight.txt)
Please, help!!

itProfi, 2019-07-18:
Chris, help, please..)

sidoh, 2019-07-19:
This is pretty different.  Honestly don't think it makes sense to add support unless there is for some reason a lot of people with this same device.  Can offer some tips on decoding it and adding support in your own fork, though.

itProfi, 2019-07-20:
OK .. If you can, give advice on his support in my fork..

itProfi, 2019-07-25:
Chris?

sidoh, 2019-07-25:
sorry, not going to have time to give detailed instructions.  you can check out this PR adding support for a new remote type:

https://github.com/sidoh/esp8266_milight_hub/pull/490

you'll need to decode the packet structure yourself, though.

itProfi, 2019-07-25:
ok, thanks..i try..))

sidoh, 2019-07-27:
Nice.  It looked pretty similar to RGBW structure, for what it's worth.  Gonna close the issue, but happy to continue the discussion.  Good luck!

closed 2019-07-27 by sidoh

ZeMilkmann, 2020-12-05:
@itProfi Did you ever finish implementing this protocol? I have a RGB+CCT lamp with a remote I couldn't sniff with the default configuration but the lamp pairs and reacts to an old CCT remote I had laying around. 
Trying to sniff that gives me  what I have written below, so I assume we may have the same or a very similar protocol

```
cct packet received (7 bytes):
Request type  : 5A
```