sidoh/esp8266_milight_hub #75: Default value for "packet_repeats" and "http_repeat_factor"

WoodsterDK, 2017-05-03:
Might want to add some default values for these, as a complete fresh install cannot send, but only receive frames.

sidoh, 2017-05-03:
A complete fresh install with flash wiped should default to 10 and 5, respectively:

```
    packetRepeats(10),
    httpRepeatFactor(5),
```

Has this not been your experience?

WoodsterDK, 2017-05-03:
Strange, the fields were empty in the web app.

sidoh, 2017-05-03:
This is after clearing flash on another ESP:

[image]

It's possible older versions didn't play nicely with even older settings blobs, but I think the newest version should do what it's supposed to.

Are you able to replicate?

WoodsterDK, 2017-05-03:
Let's close it up... might be a bad flash routine.

closed 2017-05-03 by WoodsterDK

sidoh, 2017-05-03:
Cool. Thanks anyway for raising. :)