sidoh/esp8266_milight_hub #76: Support for `4CH WRGB Saturation`

spmp, 2017-05-04:
I have a semi experimental remote and controller sold to me by Limitless LED in New Zealand. The controller is still available [here](http://www.limitlessled.com/shop/rgbw-led-strip-mr16-gu10-controllers/) and is the _`4CH WRGB Saturation`_ type.
The remote is as:  [image]

I have managed to sniff commands only in `RGBW` mode. The standard `RGBW` controls from the web interface work, with the quirk that pressing `white` increases brightness
Below is a mix of the output from the web interface with debug output  together with the button pressed. The second line of `packet received`, `packet transformed` is from a second run through. 
I can confirm that sending the capture packets back produces the correct result.
I am quietly hoping that the R,G,B, and W can be specified as levels ie `{"r":255,"g":240,"b":255, "w":255}` or saturation + white level

```
Device ID: 0xAC34
Off
  Packet received (7 bytes):
  Request type  : B0
  Device ID     : AC34
  b1            : 71
  b2            : 04
  b3            : 0B
  Sequence Num. : FD
  Packet received: A4AE01D352C8E80408F6BC9F
  Packet transformed: 07B8AC34710102F1D693
  Packet received: A4AE01D352C682040C84D05F
  Packet transformed: 07B8AC3416040213B2A0
On
  Packet received (7 bytes):
  Request type  : B0
  Device ID     : AC34
  b1            : 71
  b2            : 04
  b3            : 0D
  Sequence Num. : FC
  Packet received: A4AE01D352C8E00804F9F8BF
  Packet transformed: 07B8AC34710001F2F9D1
  Packet received: A4AE01D352C6800802820B9F
  Packet transformed: 07B8AC3416000114049D
W+
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 04
  b3            : 0C
  Sequence Num. : FB
  Packet received: A4AE01D352C8E80C0CF3895F
  Packet transformed: 07B8AC34710103F31CA9
  Packet received: A4AE01D352C6880C0A887A7F
  Packet transformed: 07B8AC3416010315E1E5
W-
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 04
  b3            : 0A
  Sequence Num. : FA
  Packet received: A4AE01D352C8E80202FD509F
  Packet transformed: 07B8AC34710104F4AB90
  Packet received: A4AE01D352C688020684E59F
  Packet transformed: 07B8AC3416010416729A
R+
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 04
  b3            : 09
  Sequence Num. : F9
  Packet received: A4AE01D352C8E40A0AF79EEF
  Packet transformed: 07B8AC34710205F59E77
  Packet received: A4AE01D352C6840A0E8E2BEF
  Packet transformed: 07B8AC3416020517477D
R-
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 03
  b3            : 08
  Sequence Num. : F8
  Packet received: A4AE01D352C8E40606FB6F6F
  Packet transformed: 07B8AC34710206F66D6F
  Packet received: A4AE01D352C684060181BF5F
  Packet transformed: 07B8AC3416020618D8AF
G+
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 03
  b3            : 07
  Sequence Num. : F7
  Packet received: A4AE01D352C8EC0E0EF07BCF
  Packet transformed: 07B8AC34710307F7E03D
  Packet received: A4AE01D352C68C0E098AABFF
  Packet transformed: 07B8AC341603071955FD
G-
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 02
  b3            : 06
  Sequence Num. : F6
  Packet received: A4AE01D352C8EC0101FFB62F
  Packet transformed: 07B8AC34710308F8DF46
  Packet received: A4AE01D352C68C010586032F
  Packet transformed: 07B8AC341603081A064C
B+
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 02
  b3            : 05
  Sequence Num. : F5
  Packet received: A4AE01D352C8E20909FD143F
  Packet transformed: 07B8AC34710409F98BC2
  Packet received: A4AE01D352C682090D84A13F
  Packet transformed: 07B8AC341604091B52C8
B-
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 01
  b3            : 04
  Sequence Num. : F4
  Packet received: A4AE01D352C8E20505F1E5BF
  Packet transformed: 07B8AC3471040AFA78DA
  Packet received: A4AE01D352C68205038A169F
  Packet transformed: 07B8AC3416040A1C8596
S-
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 01
  b3            : 03
  Sequence Num. : F3
  Packet received: A4AE01D352C8E2030DF84F9F
  Packet transformed: 07B8AC3471040CFB219F
  Packet received: A4AE01D352C682030B83BCBF
  Packet transformed: 07B8AC3416040C1DDCD3
M
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 00
  b3            : 01
  Sequence Num. : F2
  Packet received: A4AE00D352C8E20B03F5534F
  Packet transformed: 07B0AC3471040DFCAA2C
  Packet received: A4AE00D352C6820B078CE64F
  Packet transformed: 07B0AC3416040D1E7326
S+
  Packet received (7 bytes):
  Request type  : B8
  Device ID     : AC34
  b1            : 71
  b2            : 01
  b3            : 02
  Sequence Num. : F1
  Packet received: A4AE00D352C8E20D0BFCF96F
  Packet transformed: 07B0AC3471040BFDF369
  Packet received: A4AE00D352C6820D0F854C6F
  Packet transformed: 07B0AC3416040B1F2A63
```

Please excuse me if this is all in hand and undersood.
There was some discussion [here](https://github.com/mwittig/node-milight-promise/issues/7)

sidoh, 2017-05-04:
I'm guessing this is just [FUT095](http://futlight.com/productdetails.aspx?id=173&typeid=143) with the labels changed. My milight wifi box has a remote that resembles yours, and it does look this way -- just FUT095 with the labels changed.

You should actually be able to use it already with the RGBW config. Group 1 on/off is white up/down, on/off for groups 2, 3, and 4 are red, green, and blue up/down respectively.

You should be able to set r/g/b directly, but it'll only match the hue. I don't think you can set saturation directly without ramping it down to the min/max and then ramping up to the desired value.

spmp, 2017-05-05:
Right on!
 That worked as you describe.
Repeats did not work tho.

I tried and utterly failed to understand how the code flow worked and where b1, b2, b3 etc. were formatted/extracted/rearranged to read or send packets. I was hoping as an excersize to implement this type as `wrgb` just to be confusing,  with the extra comands `red_up`, `red_down`, `green_up`... etc.
Was hoping that it could be a sub-class of `rgbw`.

I wonder if `saturation(x)` can be bodged in as `hue` ± `(r(y), g(y), b(y))` or where `y` is a function of `level`, `level_max`, and `x` where `y` is inc/dec with repeats... not pretty, but may work 8)

sidoh, 2017-05-05:
The code that formats the RF packets for RGBW remotes is in [RgbwPacketFormatter](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/RgbwPacketFormatter.cpp
). The handler for PUT-style JSON inputs is in [MiLightClient](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/MiLightClient.cpp#L224
).

I think it'd be easiest to just add commands to MiLightClient and use `rgbw` since there's nothing different about the RF config or packet structure.

I might be misunderstanding you, but I'm realizing also that I probably wasn't being clear. You can certainly determine hue and saturation from RGB values. In fact, that's done [here](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/MiLightClient.cpp#L260) in MiLightClient. But it doesn't look like there's a way with this remote's packet structure to set a value for saturation directly. It looks like you can only increase or decrease the current value. 

For reference, the older CCT-only bulbs are this way with both brightness and white temperature. To accomplish setting a value directly, they're decreased a number of times that guarantees they're at the minimum value, then increased a number of times to match the desired value. Code is [here](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/CctPacketFormatter.cpp#L16).

Snocksman, 2017-07-12:
I have five of this Controllers, too. I tried to use the rgbw setting for them but this didn´t really work. With the rgbw setting I´m able to change the color with the color picker, but i´m not able to change each color with the On / Off Button for the four Groups and I have no chance to switch the entire light off.
It would be great if you could add support for this type of Controller !!!
If you need further logs from this Controller type I can provide these for you.

sidoh, 2017-10-06:
Sort of lost context on this thread.  @Snocksman, do you have a remote that works with your controller?  Would be helpful to get packet captures.  The remote I was looking at is definitely just FUT095 with labels changed.

khmann, 2018-01-02:
The FUT028 "creative controller" responds to traditional RGBW remote commands (zone 1 = W+/W-, 2-4=RGB +/-).  It is "paired" using the all on/off keys.  In "color" mode, you can add/remove white (or other colors, for example to create a "cold" R+G+B white).  These adjustments are all relative.  I wonder if (since "zones" are not used) one regains those bits from the brightness register...

additionally this controller (advertised as "Transmitting Automatically", "Synchronizing Automatically") prioritizes a 4th frequency (2457MHz) for listening and also uses it for re-transmission of received packets

xbmcnut, 2023-09-10:
> It is "paired" using the all on/off keys.
> 
Thank you! I was struggling to get this to pair until I selected RGBW and all zones.