sidoh/esp8266_milight_hub #78: Fade In / Out mode

g-rom-info, 2017-05-05:
I suggest you a simple mod for bulbs.

When you turn off bulbs that are in maximum brightness, and then you want to re-light them, but at low light, they light up at full brightness and only then reduce their power.
This can be unpleasant when you are in the dark and you want to light the lights dimly.

It would seem that this is the nominal operation of the original Milight bridge: it is necessary to light the bulbs before adjusting their brightness.

Would it be possible to implement an optional function capable of systematically putting the lamps at minimum power before extinguishing them? As a result, the lamps will illuminate at low power before having the desired brightness.

What do you think ?

sidoh, 2017-05-05:
I like it. Could add commands:

* `fade_in` - turns on and sets brightness to max
* `fade_out` - sets brightness to min and turns off

is that what you'd want?

g-rom-info, 2017-05-05:
Initially, I thought this command could work.
```
curl --data-binary '{"level":10,"status":"off"}' -X PUT http://192.168.18.13/gateways/0xD24E/rgbw/0
curl --data-binary '{"status":"on"}' -X PUT http://192.168.18.13/gateways/0xD24E/rgbw/0
```
However it does not work... :(

As a consequence `fade_in` and `fade_out` commands can be a good solution.
However V5 or V6 UDP protocol usage does not have this ability natively (with domoticz for example)
Is it possible to add a check box to enable this feature at each UDP bridge?

sidoh, 2017-05-05:
Hmm... I'm worried this adds more complexity than I'm comfortable with.

How hard is it to integrate with MQTT or REST with Domoticz?

sidoh, 2017-05-05:
And regarding the example you posted not working, I think it's because status is always evaluated first. So the bulb is turned off, and will begin ignoring any packets that don't turn it on.

One thing that might make it work would be to set `"status":"off"` to lowest priority, and `"status":"on"` to highest priority. That'd make your example work without the command. Maybe that's preferable since the commands I suggested are redundant with existing functionality if priorities work that way. This wouldn't solve the problem with UDP, though.

g-rom-info, 2017-05-05:
I think it is possible to add a Scene in domoticz in order to sequence all of necessary UDP commands.
Scene 1 : Fade Out, then 1 seconds later Off
Scene 2 : On, then fade in.
However it is less user friendly because it is not listed as a unique Device but as 2 Scenes

The second solution is to add an HTTP request in order to put commands when lights on or off, however UDP commands will be sent before ==> the bulb will be switched off with UDP before sending requested HTTP commands...

I don't know if it is possible to catch commands and the replace it by http with an event script...

g-rom-info, 2017-05-05:
Yes, set a lowest priority to "off" command can be a good solution for the HTTP command.
I agree that fade_in and fade_out will be redundant :)

sidoh, 2017-05-06:
The on/off priority we discussed above should now be implemented in the [v1.3.0](https://github.com/sidoh/esp8266_milight_hub/tree/v1.3.0) branch.

I have no familiarity with Domoticz, but sounds like you have some good ideas. Do you think the priority change implemented for HTTP/MQTT will allow you to do what you want?

closed 2017-06-03 by sidoh