sidoh/esp8266_milight_hub #79: Import / Export configuration from the webui

g-rom-info, 2017-05-05:
I suggest you an another enhancement request : is it possible to add a simple function on the webui in order to backup / restore a configuration file ?

sidoh, 2017-05-05:
Good idea. Should be possible already if you're willing to do it without the UI. You can `GET /settings` to retrieve a JSON blob of the settings, and `PUT /settings` to patch in a JSON blob of settings.

g-rom-info, 2017-05-05:
Hum that's a very good alternative ! I think a simple import/export from/to .txt file will be more user friendly, but the /setting HTTP request is amply enough for the geek I am! ;-)

sidoh, 2017-05-05:
Yeah - a button to download the settings as a file, and to upload a settings snapshot is a good idea.

sidoh, 2017-05-06:
Added in v1.3.0 branch. The UI in that branch should work fine with 1.2.x. It's just using existing routes in the REST API.

closed 2017-06-03 by sidoh