sidoh/esp8266_milight_hub #81: Can't control FUT066, FUT015, FUT103. Sniffing works.

tbrasser, 2017-05-11:
I can't seem to control any of my RGB+CCT bulbs, that I currently control using bot the iBox1 bridge and a  FUT092 remote.

Here's what sniffing captures:

From the remote:
```
Packet received (9 bytes):
Raw packet: 3D 72 3F 6A 75 58 7C 77 AF 

Decoded:
Key      : 3D
b1       : 20
ID       : 0578
Command  : 01
Argument : 05
Sequence : D7
Group    : 00
Checksum : EF
Packet received (9 bytes):
Raw packet: 03 F1 C1 F8 27 86 A5 BC 82 

Decoded:
Key      : 03
b1       : 20
ID       : 0578
Command  : 81
Argument : 00
Sequence : D6
Group    : 00
Checksum : AB
```
From the v6 bridge using home assistant:
```
Packet received (9 bytes):
Raw packet: 78 63 5B ED EE 53 BD EB E4 

Decoded:
Key      : 78
b1       : 20
ID       : 0EBE
Command  : 01
Argument : 07
Sequence : 87
Group    : 02
Checksum : BD
Packet received (9 bytes):
Raw packet: BB 39 04 6E EF CB BD 06 EA 

Decoded:
Key      : BB
b1       : 20
ID       : 0EBE
Command  : 01
Argument : 07
Sequence : 86
Group    : 02
Checksum : 7B
```

Is there anything wrong with my config or are these bulbs not yet supported?

sidoh, 2017-05-11:
Nah, RGB+CCT is supported. I use them pretty heavily in my own setup with an ESP hub.

What are you trying to do with the ESP that isn't working? Are you setting the ID to match your remote or ibox, or are you trying to pair with a different ID?

Could you paste the output of `/settings`? Should look something like this:

```
$ curl -vvv 10.133.8.146/settings
* Hostname was NOT found in DNS cache
*   Trying 10.133.8.146...
* Connected to 10.133.8.146 (10.133.8.146) port 80 (#0)
> GET /settings HTTP/1.1
> User-Agent: curl/7.35.0
> Host: 10.133.8.146
> Accept: */*
>
< HTTP/1.1 200 OK
< Content-Type: text/html
< Content-Length: 290
< Connection: close
< Access-Control-Allow-Origin: *
<
* Closing connection 0
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"radio_interface_type":"LT8900","packet_repeats":100,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"10.133.8.11","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","device_ids":[]}%
```

tbrasser, 2017-05-11:
Here's my /settings:
```
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"radio_interface_type":"nRF24","packet_repeats":10,"http_repeat_factor":5,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","device_ids":[291,1400],"gateway_configs":[[291,8899,6]]}
```

sidoh, 2017-05-11:
Thanks for sharing that. Looks fine.

You're trying to control bulbs with IDs that bulbs are already paired with? Do you have another ESP+NRF you could install the firmware on and put in listen mode to verify RF packets are being sent? I assume you're sending commands via the UI?

The next thing I'd try is compiling with `-DDEBUG_PRINTF` (just add it to `build_flags` in [platformio.ini](https://github.com/sidoh/esp8266_milight_hub/blob/master/platformio.ini#L21)) and watching serial output while sending commands.

tbrasser, 2017-05-11:
I only have a bare ESP8266, not another NodeMCUv2, so I'll try the serial debug option first!

tbrasser, 2017-05-11:
I got it working by changing the cabling and trying (yet) another NRF24 module. Thanks for your help! (and sorry for bothering you when it was an issue on my end)

sidoh, 2017-05-11:
Glad you got it working! 

I saw WDT resets occasionally before I switched to soldered connections. I think the dupont wires were flakey enough to cause issues, especially when there was any movement involved.

tbrasser, 2017-05-11:
Thanks for the headsup, I'll solder them!

closed 2017-05-11 by tbrasser