sidoh/esp8266_milight_hub #83: Support for passive listening

sidoh, 2017-05-21:
The idea here would be to capture 2.4 GHz packets sent from other devices (e.g., remotes), and publish MQTT messages.

This will require:

* Scanning all known channels
* Listening by default. (i.e., in `loop`)

Probably blocked by #82.

sidoh, 2017-07-01:
Tried this using the most trivial approach, and it seems to work really reliably. #82 will be trickier.

Can also make the web UI listen mode a websocket without the need to select a mode or click on a button.

Arielpod, 2017-07-02:
So how can we get this functionality?

sidoh, 2017-07-02:
Still a work in progress, but it's working pretty well. Mostly just need to get documentation together. There's a WIP branch here:

https://github.com/sidoh/esp8266_milight_hub/tree/passive_listen

sidoh, 2017-07-02:
Going to give it a few days of testing, but for anyone that wants to try early:

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.4.0-dev1

SmilyOrg, 2017-07-03:
If I understand this right, it means that this hub will be able to pick up commands sent from a LimitlessLED / milight remote like this one?

[image]

If so, that would be really cool as I have one that's not really used for anything else right now.

Additionally, can this be used to detect when a bulb gets turned on at the wall switch? I've been trying to find out whether that is even possible (e.g. whether a bulb communicates with the hub when it turns on) for a long time.

Thanks ✌️

sidoh, 2017-07-03:
re: the remote, that's the idea. It reliably publishes updates to an MQTT topic. Unfortunately doing anything useful with that in HomeAssistant is proving more difficult than I thought it'd be. What I want to do is essentially forward messages from the updates topic to the command topic for multiple remote IDs. If I can't figure out an easy way to do it in HA, I'll cram something in my [ha_gateway](https://github.com/sidoh/ha_gateway) project.

Unfortunately bulbs don't send any data when they turn on. The only data I've ever seen bulbs send is when they're in one of the "disco modes." They seem to send out a packet to sync the timing with other bulbs. The only way I've thought to make that useful is to put bulbs in disco mode before they're turned off, but not sure if that'd negatively affect usability.

SmilyOrg, 2017-07-04:
#### Re: Remote
Very cool that you can do that with the remote! I might have to order me some radios then ;)

I'm not sure what you mean by forwarding messages? If I'm understanding you right you want to have multiple remotes control the same bulb(s)?

Maybe you can set up every remote as an [MQTT Light](https://home-assistant.io/components/light.mqtt/) and use the updates topic as the state topic for every parameter of the light. I'm not sure what the updates topic looks like, you might have to make use of the [JSON Light](https://home-assistant.io/components/light.mqtt_json/) or [Template Light](https://home-assistant.io/components/light.mqtt_template/) to properly parse out the right values for brightness, color, etc.

For some of the binary "buttons" on the remote you can probably also add additional [Binary Sensor](https://home-assistant.io/components/binary_sensor.mqtt/)s.

I guess then you would just set up automations to map the various light parameters and/or binary sensors to actual lights that might or might not be in the same system even. You could also loop back into the milight hub via the MQTT exposed lights (with their command topics set).

It seems like a lot of verbose configuration, but I think it would also be the most flexible and you don't have to do any hacking or additional scripting to do it. Maybe it would be useful to make a template of sorts for each remote type and you could probably even integrate it into HA as an "MQTT Limitlessled Remote" or something similar. In order of hassle ;)

#### Re: 🎶💡 Disco mode 🕺🚨
That's very interesting. So they do have the capability of sending back at least _something_. I think putting bulbs into disco mode before they're turned off is quite interesting. There would be an additional risk factor, as if the system malfunctions, you'll be left with disco lights and no easy way to change them (as the system malfunctioned!). Though it might turn into a spontaneous party, so maybe it's a feature.

I wonder how the sync packet works, is it sent on an interval i.e. every X seconds or is it sent at the start/end of a disco sequence? If the former, the usability probably depends on the interval. If the latter, maybe the frequency being as high as possible will make it send a sync packet faster.

If paired with an inoffensive disco mode (like a smooth white fade if that exists) it might not be that bad of a compromise to get state back into HA.

I wonder how hard it would be to update the firmware on the bulbs though... probably not worth the effort :)

Sorry for the wall of text and thanks for the info! 👍

Arielpod, 2017-07-04:
OK, so I have flashed 1.4.0-dev2. But I don't know how to configure MQTT to work. What have you got in the mqtt_topic_pattern? And in the configuration.yaml?

sidoh, 2017-07-04:
Yeah, I guess the idea would be to use a Milight remote in conjunction with HomeAssistant. Then you could control lights with the remote and have HA's state updated. You could also use the remote for totally different things.

The JSON state used by this project supports HomeAssistant's `mqtt_json` platform :)

What you said here:

> You could also loop back into the milight hub via the MQTT exposed lights (with their command topics set).

Is exactly what I was hoping to do! :) 

Think I managed to get it working. You can probably imagine how you'd expand this to control multiple groups of bulbs. Not sure if there's a nicer way of going about it:

```
$ cat automations/milight_forwarder.yaml
- alias: MiLight Forwarder
  initial_state: True
  trigger:
    platform: mqtt
    topic: milight/updates/0x1C8E/rgb_cct/+
  action:
    - service: mqtt.publish
      data_template:
        topic: "milight/0x118D/rgb_cct/{{ trigger.topic.split('/')[4] }}"
        payload_template: >
          {{ trigger.payload }}
```

I also figured out how to modify the trigger payload in JINJA2 templates, but it ended up not being useful.

> Re: 🎶💡 Disco mode :man_dancing:🚨
>
> That's very interesting. So they do have the capability of sending back at least something. I think putting bulbs into disco mode before they're turned off is quite interesting. There would be an additional risk factor, as if the system malfunctions, you'll be left with disco lights and no easy way to change them (as the system malfunctioned!). Though it might turn into a spontaneous party, so maybe it's a feature.
>
> I wonder how the sync packet works, is it sent on an interval i.e. every X seconds or is it sent at the start/end of a disco sequence? If the former, the usability probably depends on the interval. If the latter, maybe the frequency being as high as possible will make it send a sync packet faster.
>
> If paired with an inoffensive disco mode (like a smooth white fade if that exists) it might not be that bad of a compromise to get state back into HA.
>
> I wonder how hard it would be to update the firmware on the bulbs though... probably not worth the effort :)

A+ emoji game :)

I've not looked into it very carefully yet. I noticed it when I was working on reversing the RGB+CCT protocol.  I'm not sure if other bulb types do this.

They won't show up in sniffed packets right now because they're a different length and are getting parsed the wrong way, but if you compile with the `DEBUG_PRINTF` flag, you'll see it (I've done a bit of parsing here to strip out the LT8900 wrapper):

```
11 0D 06 00 02 06 C1
11 0D 06 00 02 09 C2
11 0D 06 00 02 06 C3
11 0D 06 00 02 01 C4
11 0D 06 00 02 06 C4
11 0D 06 00 02 04 C5
11 0D 06 00 02 09 C5
11 0D 06 00 02 00 C6
11 0D 06 00 02 03 C7
11 0D 06 00 02 08 C7
```

Here are some notes:

* bytes[0,1] is the remote ID. Actually different than my remote ID (0x118D), but assuming that's because there's something off about the RGB+CCT decoding parameters.
* bytes[2] seems to correspond with the ID of the mode the bulbs are in
* bytes[3,4] always 0002 as far as i can tell
* bytes[5] seems like this is maybe some kind of bulb ID. considering they're all pretty close to 0 it must not be assigned to the bulb, but negotiated somehow. This could be very interesting if that's true
* bytes[6] sequence id

Will probably be a while before I can do anything useful with this, but definitely shows promise.

sidoh, 2017-07-04:
@Arielpod,

There's a bit of documentation on this in the README. I use `milight/:device_id/:device_type/:group_id` for mqtt_topic_pattern, and `milight/updates/:device_id/:device_type/:group_id` for mqtt_updates_topic_pattern. Also make sure that listen_repeats is set to 3. Noticed default setting values aren't getting populated for existing configs.

SmilyOrg, 2017-07-04:
Oh it's cool that you can just forward the entire MQTT payload like that, I haven't though of that. It looks like the shortest automation you could do, since I imagine setting up the remote control as a bunch of sensors would easily take 100 lines. It might make it easier to reason with / debug though.

The disco packets look pretty cool, more info than I was expecting.

Btw which radio would you suggest me to order? NRF24L01 or LT8900 or both or neither?

sidoh, 2017-07-04:
I've had better luck with the NRF24L01. The LT8900 is more performant when it works, but I've had stability problems with it I've not been able to figure out. Could just be a bad connections, though. The one I ordered had 1.27mm pins and I had difficulty soldering something so tiny.

Whatever you order, I'd recommend getting a few of them. I've gotten a few bad NRFs and it took a while for me to realize what the problem was.

sidoh, 2017-07-05:
Here's a slightly more detailed writeup on how to set this up:

https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant

Will probably release v1.4.0 in the next few days.

SmilyOrg, 2017-07-05:
Very cool and useful! Nice giffage 🔘🔴💡👍 

I ordered some NRF24L01s, hopefully I'll get the chance to build a hub soon™️.

rhamblen, 2017-07-06:
Just as an FYI...

I have swapped from a NodeMCU V3 to a HiLetgo D1 Mini NodeMcu and I use the NRF240L1.  I've also now soldered the wires.  This new package is tiny! Half the box is the micro usb connector and the knot in the cable!  The box is 60x30x20 and I could have gone smaller with a Right angle micro usb.

It sits behind my Amp, in with all the electonics for the room (TIVO, FireTV, etc), and is near the AP for downstairs, so loads of potential interference.  Still can turn on all the lights downstairs and covers over 12m and goes through 2 brick walls.

Not a bad little toy.

sidoh, 2017-07-08:
Released in [1.4.0](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.4.0).

closed 2017-07-08 by sidoh