sidoh/esp8266_milight_hub #87: Old White Bulbs

rhamblen, 2017-06-21:
Is it possible to support the old white bulbs.
The structure is the same as the RGBW but uses different parameters.  Hex codes.
I have tested these codes on HA Bridge
udp://192.168.x.xxx:8899/0x380055  turns on the lights in zone 1

DIm is a challenge as it is just an increase or decrease, so may not work with smarthings. But it will be a huge help just to get the basic on off working.

R.

LimitlessLED White
——————
35 00 55 – All On
39 00 55 – All Off
3C 00 55 – Brightness Up
34 00 55 – Brightness Down (There are ten steps between min and max)
3E 00 55 – Warmer
3F 00 55 – Cooler (There are ten steps between warmest and coolest)
38 00 55 – Zone 1 On
3B 00 55 – Zone 1 Off
3D 00 55 – Zone 2 On
33 00 55 – Zone 2 Off
37 00 55 – Zone 3 On
3A 00 55 – Zone 3 Off
32 00 55 – Zone 4 On
36 00 55 – Zone 4 Off
B5 00 55 – All On Full (Send >=100ms after All On)
B8 00 55 – Zone 1 Full (Send >=100ms after Zone 1 On)
BD 00 55 – Zone 2 Full (Send >=100ms after Zone 2 On)
B7 00 55 – Zone 3 Full (Send >=100ms after Zone 3 On)
B2 00 55 – Zone 4 Full (Send >=100ms after Zone 4 On)
B9 00 55 – All Nightlight (Send >=100ms after All Off)
BB 00 55 – Zone 1 Nightlight (Send >=100ms after Zone 1 Off)
B3 00 55 – Zone 2 Nightlight (Send >=100ms after Zone 2 Off)
BA 00 55 – Zone 3 Nightlight (Send >=100ms after Zone 3 Off)
B6 00 55 – Zone 4 Nightlight (Send >=100ms after Zone 4 Off)

sidoh, 2017-06-21:
Sweet!

UDP is the first half of the puzzle. Will also need to reverse the RF packet structure. I may be able to do it with my Milight hub, but it'd be helpful to get some packet captures you know do what you want. Should be enough to just paste the output of listen mode from your ESP while you send commands to your milight hub.

rhamblen, 2017-06-21:
Listen mode doesn't work for me.
I tried it but couldn't get it to work.  Maybe it's me doing something wrong!

Happy to give you whatever is needed though.

R.

Rgds
Richard

rhamblen, 2017-06-22:
Doh - ok its obvious why it doesnt work now.  It ignores the packets which are not any of the default bulbs.
I guess I need to go into the ESP directly to sniff the packets... can you give me a steer and I'll do it for you.

sidoh, 2017-06-22:
Does it not show up when you listen on any of the existing radio configs (RGB, RGBW, etc.)?

It could be a different set of radio configs (channels + syncwords). If that's the case we'd probably have to do something more advanced. The radio settings that I have for the other bulb types were provided by brave spelunkers that pulled them by sniffing SPI on the PCB, which is not something I'm equipped to do currently. Would probably either have to find them on Google, or get someone who has as logic analyzer to help out.

rhamblen, 2017-06-22:
Ah that makes sense.  Things like HA Bridge talk to the existing hub using UDP commands.  the hub understands which lights the commands are for and then issues a rf signal to control the lights.  So the udp commands are the things you need to understand on the recieving side and the RF info is on the transmit side.  
I'll recheck the sniffer and see what happens and then I guess if we cant find the RF details we have a little problem!
Thanks and I do understand the problem now.

rhamblen, 2017-06-22:
hi 
just found this and it might be useful.
https://hackaday.io/project/5888-reverse-engineering-the-milight-on-air-protocol

sidoh, 2017-06-22:
I think there are some others who have opened issues on this project that have logic analyzers. I want to verify that this is what's going on, but I'll tag them to ask for help if so.

sidoh, 2017-06-25:
Alright, as far as I can tell this thing is using different radio configs. Dug up my old v1 wifi bridge to test, and none of the existing radio configs show anything.

@g-rom-info @Lstt2005 it sounded in previous tickets like you guys have logic analyzers (I don't, unfortunately). Would you be able to figure out the channels and syncword being used with the old plain white bulbs? I think it's this remote in the new milight app:

[image]

g-rom-info, 2017-06-25:
Hi all,

I don't understand what you mean... Because CCT white bulbs are already supported in [UDPv5](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/Udp/V5MiLightUdpServer.h).
UDP codes are the same as you said on your first post.
I have old white bulbs, and they work properly with my ESP.

About logic analyzer, I have already reversed RF Protocol with this method [see my blog](http://www.g-rom.info/2016/10/milight-multi-bridge-emulator-chapitre-3/) (in French) :-)
It uses a logic analyzer based on an Arduino Mega (cost ~15€)

I can try with this new app however I have not conviced that I will catch new informations.
Syncwords for CCT (white bulbs) are 0x050A 0x55AA
Channels are 4, 39, 74

sidoh, 2017-06-25:
I thought Milight made a plain white (i.e., fixed color temperature) bulb, but I'm having a hard time finding it. The remote I linked is [this](http://futlight.com/productdetails.aspx?id=125&typeid=146) one, I think, which is for a LED strip.

Cool! I had it in my head that logic analyzers were a few hundred bucks. Awesome to see you can get the job done with arduino-based hardware.

I didn't notice that the UDP codes supplied were for the CCT bulbs. I assumed they were new codes for one of the plain white remotes. Thanks for catching that.

@rhamblen, could you link to the bulbs you have? [futlight.com](http://futlight.com) has a pretty complete list of milight products.

rhamblen, 2017-06-25:
This is pre the new cut white lights... back in early days there was a white only light.
Often strip lights (which is where I'm using them) were white only.

Rgds
Richard

sidoh, 2017-06-25:
Just realized you're the same Richard I was corresponding with back in February, and that you also have a logic analyzer. Sorry for not catching that sooner!

I'm quite sure the remote type I was testing with on the v2 milight wifi box has different channels/syncwords. Not sure if it's the same one you're wanting support for, but would be very helpful to get the channels/syncwords.

rhamblen, 2017-06-25:
Nope not me.
I've only just found this project!

Rgds
Richard

sidoh, 2017-06-26:
Whoops. Sorry for the mixup!

Where did you find the UDP codes you provided? Could you take a screenshot of the screen in the app you're using to control the bulbs?

rhamblen, 2017-06-29:
HI Sorry been busy last few days with work and also re-learning about Microcontrollers and Arduino (been a few years since I did work with microcontrollers and they have moved forward a little!  C++ instread of machine code!!!! 

OK...
near the bottom of http://www.limitlessled.com/dev/ is the instruction set.  (after Examples of Python Code #2)
The controller for the ribbon lights is about 4 years old.  Was in the era of V4 hubs and the early days of RGBW lights.  WW and CW were separate bulbs you puchaced at the time.

The controller is NOT the one shownabove in the earlier post, it is different and as shown below.

[image]

Also do a quick search on 'milight white light controller'
https://www.google.co.uk/search?q=limitless+led+white+light+controller&client=firefox-b-ab&source=lnms&tbm=isch&sa=X&ved=0ahUKEwj17fO1vuPUAhVhBsAKHbcECG0Q_AUIDCgD&biw=1461&bih=787#tbm=isch&q=milight+white+light+controller&imgrc=xarPOyvpzv_uhM:

Hope this helps.
Richard

sidoh, 2017-06-29:
The LT8900 settings at the bottom of the limitlessled docs are for RGBW bulbs. Those are [already supported](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/MiLightRadioConfig.h#L49) by this project, and I don't think they work with the plain white remote. Otherwise packets from the white remote would show up when sniffing for RGBW like so:

[image]

rhamblen, 2017-06-29:
I was just having a play with the sniffer again.   AND I changed the batteries!
I managed to get it to work...

CCT lights
**all on and all off**

Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 00
b2            : 09
b3            : E8
Sequence Num. : BC
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 00
b2            : 05
b3            : E7

**group 1, 2 3 and 4 on of respectively**
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 04
b2            : 06
b3            : F0
Sequence Num. : C5
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 04
b2            : 02
b3            : EF
Sequence Num. : C0
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 03
b2            : 0A
b3            : EE
Sequence Num. : C6
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 01
b2            : 0B
b3            : EA
Sequence Num. : C1
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 01
b2            : 08
b3            : E9
Sequence Num. : BD

**bright/dim** (2x bright and 2x dim as two buttons do each action)

Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 04
b2            : 04
b3            : 30
Sequence Num. : 03
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 04
b2            : 0E
b3            : 2F
Sequence Num. : 0C
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 04
b2            : 0F
b3            : 2D
Sequence Num. : 0B
Packet received (7 bytes):
Request type  : 5A
Device ID     : 3634
b1            : 04
b2            : 0C
b3            : 2C
Sequence Num. : 07

rhamblen, 2017-06-29:
fyi, those notes worked for the white lights in HA_Bridge but i cant get HA_bridge to work with smartthings and your solution is way better and more fun!

rhamblen, 2017-06-29:
I can kinda get it to work by pairing as CCT light.
Dim is a bit funny but I expected that as it doesnt have levels.
Important thing is that I can turn on and off the lights.
Will try with Smartthings and Alexa later tonight.

thanks
R.

sidoh, 2017-06-29:
Oh interesting. Guess they just use the CCT protocol, then?

Yeah dimming to a particular level with this protocol is a little wonky. Basically it sends 10x down commands to guarantee brightness is at a minimum and then sends the appropriate number of up commands.

rhamblen, 2017-06-30:
Thanks for all the help.
I hadn't realised the batteries were low in the remote because the lights still worked and I initially had problems with the hub sniffing properly.
So pleased with the outcome.
Tested with smartthings and alexa and all works well.

Closure note for people:  old style white only lights use the ctt protocol and work.

sidoh, 2017-07-04:
@rhamblen do you have what you need for this issue? Is it okay to close it?

rhamblen, 2017-07-04:
The issue can be closed.
Many thanks.

closed 2017-07-04 by sidoh