sidoh/esp8266_milight_hub #88: Alexa integration

alexeinz, 2017-06-22:
I recently came across this and thought it could be really cool as an enhancement to be able to add it to Alexa as wemo emulated device.... 
https://bitbucket.org/xoseperez/fauxmoesp

sidoh, 2017-06-23:
Oh, nice! This looks significantly less weighty than the Hue API, which would enable the same thing (see: #49). The project you're linking is also written as a library, which should make it super easy to integrate with.

Thanks for sharing!

poudenes, 2019-02-11:
Will the Alexa Integration added in future? I love your script and run it for months now in combination with Domoticz.. But since couple of days have Alexa Echo Dot, want connect it will Alexa as well. Because found that the action with Alexa is faster then Domoticz... :D Keep up the good work!!

ml9907, 2019-02-11:
I use it with Alexa through HA bridge https://github.com/bwssytems/ha-bridge
Simple and straightforward with MQTT. I could give you details if needed.

haymaan, 2019-02-11:
Please share details thanks

poudenes, 2019-02-11:
I installed HA-Bridge on a RPi to play around with it.
Can you send me the details? Im now using http: links for the Domoticz connection.

MQTT makes it easier and faster

ml9907, 2019-02-11:
It's a perfect start. You should install MQTT broker as well. The simplest way is mosquitto.
You have to set up esp8266_milight_hub with appropriate MQTT IP, port and topic naming structure. After that you can add a HA-bridge MQTT device. Here is an example:
[{"item":"{\"clientId\":\"HAbridge\",\"topic\":\"milight/0xD65/rgbw/2\",\"message\":\"{status:on,hue:0,level:50}\"}","type":"mqttMessage"}]
This is an "onURL" in HA bridge for a device. It would send an MQTT message to the server with topic "milight/0xD65/rgbw/2" with payload "{status:on,hue:0,level:50}" It will turn on the 2nd bulb on milight-hub #0xD65, with red light at intensity 50%.
It works like a charm, fast and accurate.

poudenes, 2019-02-11:
Thanks for the info. Can you also explain how to do brightness and colour changing? In meanwhile I go play around with this and try to learn more about MQTT as well :)

poudenes, 2019-02-11:
I have on my RPi a MQTT server. Should I use that one as server on Milight Hub ? and then give it some  naming structure ?

ml9907, 2019-02-11:
Yes, you can.
brightness:
[{"item":"{\"clientId\":\"HAbridge\",\"topic\":\"milight/0xD65/rgbw/2\",\"message\":\"{level:${intensity.percent}}\"}","type":"mqttMessage"}]
You will use the ${intensity.percent} variable in HA bridge in dimURL.
Colour is somewhat tricky since HA bridge could not show devices as colour device to Alexa therefore I use a workaround and I've defined different colours as different devices. That way you can ask Alexa to "turn red light on" this case red light is a separate device turning a specific light on in red. The same for purple, etc.

poudenes, 2019-02-11:
Ok thanks... Maybe I will stick to the current configuration:

RPi:
Domoticz installed
Homebridge installed
- Homebridge-edomoticz
- Homebridge-alexa

Because with HomeBridge it import all devices I put in a room plan inside Domoticz and those devices are discoverable by Alexa, I also can turn on/off/dim and colour change with voice command...

Only nice thing about HA-Bridge will be the full MQTT support for fast actions.

poudenes, 2019-02-16:
Would be nice to see Alexa implementation inside this build. TO bad i'm not that geeky to try. I ordered a second wemos board with all the hardware to play around and see if i can build it my own, without mesh around with current working config.

csigabit, 2020-03-09:
My workaround for this:
I use nodered and https://flows.nodered.org/node/node-red-contrib-amazon-echo
to emulate philips hue bridbe and control milight, shades, garagedoor or anything with alexa.
The palette must run on port 80 and that's not free on my server, so I have a rpi dedicated for this alaxa flow (and zigbee2mqtt), I run nodered as root, and everything is in docker, just because.
I can attach the flow for the poeople who doesn't want the learning possibility, just let me know!