sidoh/esp8266_milight_hub #94: Zones again and python interface.

RvonMallesch, 2017-07-10:
It would appear that looking through the protocol, it should support 255 zones. Is there a reason why this hasn't been extended in this project? Does it simply not work or is it for compatibility reasons? If it is for the later, perhaps a compatibility warning could be shown if someone chooses a zone over 4?

I'm still awaiting my radio so I don't have my kit together yet to do any packet sniffing, but would someone kindly point out where the device id is specified in the following?

80 00 00 00 11 {WifiBridgeSessionID1} {WifiBridgeSessionID2} 00 {SequenceNumber} 00 {COMMAND} {ZONE NUMBER} 00 {Checksum}. 

I'm on holidays at the moment so while I have spare time I'm keen to do as much as I can and for many reasons I would like to use a modified version of of the python interface at https://github.com/happyleavesaoc/python-limitlessled to work with this controller. Being able to select from more than 4 zones would make this a much simpler task.

Thanks again, I'm very excited about this project and have big plans to automate all sorts of things! p.s. Sorry in advance  for any stupid questions.

sidoh, 2017-07-10:
I wasn't super thorough about it, but I did experiment a bit when I was working on the RGB+CCT protocol. The problem is with the way the RF protocol is designed. For example, with RGB+CCT, the `on` command ranges from 0-4 for groups 0-4, and `off` ranges 5-9. I forget exactly what happens when group is >4, but I seem to recall that bulbs weren't responsive to the commands. 

Other packet formats have similar problems. Basically command IDs for on/off are hardcoded, and values only exist for groups 0-4. In the case of the RGBW command, there are only 3 bits for the group ID.

If you wanted to experiment, an easy way might be to use the `/send_raw/:device_type` endpoint ([docs](https://github.com/sidoh/esp8266_milight_hub#rest-endpoints)). Note that if you're sending RGB+CCT packets, they'll need to be encoded with [this function](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/RgbCctPacketFormatter.cpp#L204). There's a ruby implementation of the encoding [here](https://github.com/sidoh/milight_decoding/blob/master/scripts/packet_transcoder.rb).

Would definitely be interested if more than four groups works. It'd be a big deal!

> I'm still awaiting my radio so I don't have my kit together yet to do any packet sniffing, but would someone kindly point out where the device id is specified in the following?
>
> 80 00 00 00 11 [...]

The device ID isn't specified in the UDP protocol. You'd normally be addressing a Milight device that's bound to a single device ID, so there's no need to input. The way ESPMH gets around this is by allowing you to run UDP servers on multiple ports, with each port bound to a specific device ID.

> I'm on holidays at the moment so while I have spare time I'm keen to do as much as I can and for many reasons I would like to use a modified version of of the python interface at [...]

I think the python interface should work as-is with the controler. The UDP protocol is designed to fool things into thinking it's an actual Milight device.

You might find it easier to use one of the other integrations. The UDP protocol is very clunky, and really only exists for ease of integration. If you're doing something fancy, you might consider using the REST or MQTT integrations.

> Thanks again, I'm very excited about this project and have big plans to automate all sorts of things! p.s. Sorry in advance for any stupid questions.

Awesome! I'll be curious to hear what you're planning. Nope - these are great questions :)

sidoh, 2017-07-10:
Actually I guess I can confirm that the on/off commands step on each other. I overran the group past 4 in the UDP protocol to assist with the decoding process.

I have a bunch of packet captures in my [milight_decoding](https://github.com/sidoh/milight_decoding/tree/master/packet_captures) repo.

Here's a sample from trying to turn group5 on:

```
$ head -n 2 packet_captures/sidoh_wifibox1/rgbcct_group5_on.txt | tr '\n' '\0' | xargs -0 -n 1 -I % bash -c 'echo "%" | ruby scripts/decode_packet.rb'
B4 20 11 8D 01 05 17 05 5A
31 20 11 8D 01 05 18 05 28
```

Notice the 5th byte has value `0x05`, which is the same command ID for group 0 off.

It's stupid that the RF packets are structured this way, but this is the language the bulbs speak, so it's hard to step around. Totally possible there's something there I'm missing. Maybe you can use groups 0-4, 9-255 or something like that.

RvonMallesch, 2017-07-11:
Interesting! Firstly, I must say that I am MOST impressed with your reverse engineering efforts and the skill and time you have applied to you project!. Kudos :-) 

From my analysis of your packet captures it appears that your only get the dreaded 0X05 for groups 5 and 6. I surmise there should be a simple work around for this and that would be to add 3 to any group over 4 prior to encoding and of of course reverse to the process during decoding. Of course it could be more complicated but I figure it could be worth a try. That would off course reduce the the total number of groups to 252, but but would be a large house indeed!

My plan with concern to lighting is to use a more attractive and user friendly lighting remote called an iDual2 by Jedi; The bulbs it supports are expensive and in any case I figure I can implement their button effects in a nicer way. I can have a unnoticeable white coloured IR receiver in the ceiling of each room  who's IR signals I can easily decode to an individual GPIO (so I know the source location). So, my aim is to have a zone for each room, but also have each light (over 70 downlights) and array of others with their own zone that I can create whole of room effects on, like rolling waves, a glinting pond, a dappled forest, and a fireplace etc. I also wish the default white colour to be in tune with f.lux. Then there is of course party modes of course in sync with music. 

I cannot find a reference to the 3 bit limit for RBGW lights, so I am a little confused by this. Can you provide a reference?

This is why I would like like to have more zones, One zone for each room and one zone for each light. The principle reason for adapting the the Python interface is that it has great support for staging and transitions. You can also pick your own disco mode (instead of the next one) one of which would match beautifully to a button on my remotes.

Since you asked, my other projects, involve voice recognition from google and wolfram alfa, programmable bedroom skylights, central air conditioning control from any room, and locks that open based on facial and vocal recognition.

Supreme respect,

Richard

sidoh, 2017-07-11:
Thanks, appreciate it :)

> From my analysis of your packet captures it appears that your only get the dreaded 0X05 for groups 5 and 6. I surmise there should be a simple work around for this and that would be to add 3 to any group over 4 prior to encoding and of of course reverse to the process during decoding. Of course it could be more complicated but I figure it could be worth a try. That would off course reduce the the total number of groups to 252, but would be a large house indeed!

Totally possible. I'd imagine you'd be limited to at most half that number of groups since you need an on/off command for each group (there's no good reason for this, it's just how the RF packets the bulbs expect are structured).

Milight also makes eight group remotes. Taking a look at the packet structure from those might be a good place to start.

Is there a reason you couldn't do what you want with many IDs with 4 groups instead of 1 ID with many groups?

> My plan with concern to lighting is to use a more attractive and user friendly lighting remote called an iDual2 by Jedi; The bulbs it supports are expensive and in any case I figure I can implement their button effects in a nicer way. I can have a unnoticeable white coloured IR receiver in the ceiling of each room who's IR signals I can easily decode to an individual GPIO (so I know the source location). So, my aim is to have a zone for each room, but also have each light (over 70 downlights) and array of others with their own zone that I can create whole of room effects such, like rolling waves, a glinting pond, a dappled forest, a starry sky, a candle lit room, and a fireplace etc. I also wish the default white colour to be in tune with f.lux. Then there is of course party modes which of course would also in sync with music.

Cool stuff! Have you used HomeAssistant? Some of this stuff might be doable with it. I've seen others that have the white temperature automation.

> I cannot find a reference to the 3 bit limit for RBGW lights, so I am a little confused by this. Can you provide a reference?

I documented this protocol here: https://github.com/sidoh/esp8266_milight_hub/blob/master/PROTOCOL.md. Note that this is the RF protocol, not the UDP protocol. 

> This is why I would like like to have more zones, One zone for each room and one zone for each light. The principle reason for adapting the the Python interface is that it has great support for [...]

Ah, I see.

I've noticed that transitions are pretty flaky with this thing. I think because UDP packets are sent too quickly and get dropped. Just a heads up to verify what you want works before you get too far :)

> Since you asked, my other projects, involve: voice recognition from google and wolfram alfa; programmable bedroom skylights: central air conditioning control from any room: and locks that open based on facial and vocal recognition.

Very cool stuff! :)

RvonMallesch, 2017-07-13:
Thanks, I'll take your wise advice. I'll get my kit together and may have some more questions. It's interesting the that the B8 takes 8 zones isn't it?

sidoh, 2017-07-13:
Yeah, pretty cool. Sorta bummed that FUT015 bulbs don't respond to it, though. :(

khmann, 2017-09-19:
I can confirm that RGBW bullbs can pair and operate with 7 zones.  The reason for the limitation is that register is shared with brightness (5 bits).  While you can turn zones 5-7 on just fine (all the "on" commands are equivalent), you can't turn them off other than "all off".  Bummer.

Old CCTv1 Dual-White bulbs work the same way but with a full 255 zones, but again, can't turn them off.

sidoh, 2017-10-06:
Thinking this is a duplicate of #95 ("B8 wall switch support").  @RvonMallesch does this sound right to you?  If so, the good news is B8/FUT089 support will be in v1.6 (the 1.6 dev releases already have support).

@khmann, this is interesting and quite surprising.  I'm quite sure that the old RGBW bulbs I have don't work with FUT092.  Maybe there's a newer revision with updated firmware or something?

sidoh, 2018-03-13:
Hey guys, gonna close this one out.  Please let me know if I'm missing something and it should be re-opened.

closed 2018-03-13 by sidoh