sidoh/esp8266_milight_hub #96: White temperatures for CCT-only bulbs are reversed

sidoh, 2017-07-13:


r4nd0mbr1ck, 2017-07-14:
Just adding in the pastebins from serial monitor for future reference. 

When set to warmest white in HASS:

https://pastebin.com/nuRdwmDR

When set to coolest white:

https://pastebin.com/XcTBJARN

A comment in one of the other github issues mentions a fix in HASS for this - https://github.com/sidoh/esp8266_milight_hub/issues/49#issuecomment-303866070. So maybe not a bug in the hub at all?

r4nd0mbr1ck, 2017-07-14:
Just adding in the pastebins from serial monitor for future reference. 

When set to warmest white in HASS:

https://pastebin.com/nuRdwmDR

When set to coolest white:

https://pastebin.com/XcTBJARN

A comment in one of the other github issues mentions a fix in HASS for this https://github.com/sidoh/esp8266_milight_hub/issues/49#issuecomment-303866070 - . So maybe not a bug in the hub at all? 

But this may not work with mqtt_json platform in HASS as the fix appears to be in the limitlessled component- https://github.com/home-assistant/home-assistant/pull/7359/files.

sidoh, 2017-07-14:
Yeah, they were definitely backwards for RGB+CCT. I think I have a CCT bulb somewhere, I'll test with it.

I think scales are flipped all over the place (in this project and elsewhere) because white temperature is a really confusing scale: "cold" whites have higher temperatures than "warm" whites. Also doesn't help that HASS uses mireds, which is inversely correlated with white temperature, and has the opposite (albeit more intuitive) scale.

sidoh, 2017-07-15:
Just tested with my CCT-only bulb. The temperature scales are correct both in the UI and with HomeAssistant v0.45.1. What version of HASS are you using?

r4nd0mbr1ck, 2017-07-16:
I am running 0.45.1 too. This is weird. 

The temperature works correctly in the UI. The issue is only when using through HASS and that happened with both mqtt_json and limitlessled platforms.

Are the packets sent by the hub correct for each case?

sidoh, 2017-07-16:
I've not tried the UDP integration in a while. Can you paste your HASS config?

r4nd0mbr1ck, 2017-07-16:
I've stopped using UDP based on your recommendation. Here's the HASS config for both UDP and mqtt_json

light:  
  - platform: limitlessled
    bridges:
      - host: 192.168.1.110
        version: 5
        port: 4000
        groups:
        - number: 1
          type: white
          name: Living Room MiLight
          fade: on

  - platform: mqtt_json
    command_topic: "milight/0xAAAA/cct/1"
    brightness: true
    color_temp: true
    name: LR MiLight

r4nd0mbr1ck, 2017-07-17:
The temperature calculation in MilightClient.cpp feels odd. I did some calculations on paper. 

For e.g, on the cool side, HASS sends 161. The calculation results in 96 or thereabouts which is actually warm. Similarly, for warm, HASS sends 300. The calculation results in 32 which tends towards cool.

Also, HASS appears to use a max mireds of 500 or so.

sidoh, 2017-07-17:
I feel like we're using different versions of ESPMH. When HASS sends 161, my bulbs definitely go to the coolest value.

I treated mireds as an absolute scale. It refers to a specific color temperature. The specs for my bulbs list the coolest temperature as 6500K, and the warmest as 2700K. This corresponds to 153 and 370 mireds. I basically just linearly map 153-370 mireds onto the relative color temperature scale 0-100 used by the bulbs.

The reason I did this is because it seemed useful to be able to know what actual color temperature you're setting the bulbs to. If it's just confusing and people would prefer to map (HASS_min, HASS_max) -> (0, 100), we can do that instead.

sidoh, 2017-07-17:
Forgot to include, can you paste the output you see when you go to `/about`?

r4nd0mbr1ck, 2017-07-17:
You know what - I think you are right. I don't know how but the code I have looks different to the one in GitHub. For e.g. I don't have the Units helper class at all. I only recently downloaded your project so not sure where I messed this up. 

I will try with the latest code again later tonight and see if it works any different.  Sorry about potentially wasting your time with this.

sidoh, 2017-07-17:
It's no problem at all. It's actually a pretty recent change. :)

r4nd0mbr1ck, 2017-07-18:
Not a bug.. The latest code works perfectly.

sidoh, 2017-07-18:
Awesome, glad to hear it. Sorry, I should've thought to verify that earlier. 1.4.0 was released just a few days before you opened the issue.

closed 2017-07-18 by sidoh