sidoh/esp8266_milight_hub #97: Speed (S+, S-) and Mode (M) buttons on MiLight remote don't publish unique events

kociubin, 2017-07-15:
I’m using the MQTT passive listening with the MiLight remote.  In 1.4.0 pressing Speed (S+, S-) and Mode (M) buttons on the remote don’t publish unique events.  All of them publish something like this:

{“device_id”:16184,”device_type”:”rgbw”,”group_id”:3,”state”:”ON”}

 
I would like them to publish an event that identifies that one of these buttons was pressed.  For example:

{“device_id”:16184,”device_type”:”rgbw”,”group_id”:3,”state”:”ON”, “last_btn”:”S+”}

{“device_id”:16184,”device_type”:”rgbw”,”group_id”:3,”state”:”ON”, “last_btn”:”M”}

sidoh, 2017-07-15:
It's less that unique events aren't published, and more that modes aren't supported right now. This isn't the first time I've made the mistake of thinking people don't use them. Easy enough to add support.

It'd probably look more like this, btw:

```
{..., "command": "mode_speed_up"}

{..., "command": "next_mode"}
``

sidoh, 2017-07-15:
@kociubin, could you try the [1.4.1-dev1 release](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.4.1-dev1)? Should address both this issue and the other one you opened ( #98 ).

kociubin, 2017-07-15:
Tested in 1.4.1.  It's working great!  Here is the MQTT output:
{"device_id":16184,"device_type":"rgbw","group_id":0,"command":"mode_speed_down","state":"ON"}
{"device_id":16184,"device_type":"rgbw","group_id":0,"command":"mode_speed_up","state":"ON"}
{"device_id":16184,"device_type":"rgbw","group_id":0,"mode":0,"state":"ON"}
{"device_id":16184,"device_type":"rgbw","group_id":0,"mode":1,"state":"ON"}

sidoh, 2017-07-15:
Awesome :)

sidoh, 2017-07-15:
Released 1.4.1. Thanks again for bringing these up.

closed 2017-07-15 by sidoh