sidoh/esp8266_milight_hub #98: Master ON and OFF on buttons on MiLight remote don't publish a correct MQTT event

kociubin, 2017-07-15:
Pressing the Master On or Off button on the MiLight remote publishes an event like this:
{"device_id":16184,"device_type":"rgbw","group_id":0,"state":"OFF"}

There is no way to distinguish the Master On/Off event from the group On/Off event.  They are the same.

It would be great if Master On/Off could publish an event that identifies that ALL groups are off (e.g. use -1 to identify the special "all" group):

{"device_id":16184,"device_type":"rgbw","group_id":-1,"state":"OFF"}

sidoh, 2017-07-15:
0 _is_ the special all on/off group. The other groups are 1, 2, 3, and 4:

```
milight/updates/0x2/rgbw/1 {"device_id":2,"device_type":"rgbw","group_id":1,"state":"OFF"}
milight/updates/0x2/rgbw/2 {"device_id":2,"device_type":"rgbw","group_id":2,"state":"ON"}
milight/updates/0x2/rgbw/3 {"device_id":2,"device_type":"rgbw","group_id":3,"state":"OFF"}
milight/updates/0x2/rgbw/4 {"device_id":2,"device_type":"rgbw","group_id":4,"state":"ON"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
```

kociubin, 2017-07-15:
Master on/off doesn't send the correct event.  Shouldn't it always send 0 for group?  Here is an example:

Press the Group1 Off Button.  Result (correct):
{"device_id":16184,"device_type":"rgbw","group_id":1,"state":"OFF"}

Press the Master Off Button.  Result (incorrect):
{"device_id":16184,"device_type":"rgbw","group_id":1,"state":"OFF"}

Expected:
{"device_id":16184,"device_type":"rgbw","group_id":0,"state":"OFF"}

sidoh, 2017-07-15:
Yeah, should definitely always be sending 0. Unfortunately I don't have a remote to test with, but it's really likely that the remote is just sending the wrong packets. The reason I say this is it works when the packets are sent from the ESPMH (this is how I'm getting the MQTT updates without a remote). The same code path handles packets sent from the ESPMH. Here's spamming group 0 on/off:

```
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"ON"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"ON"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"ON"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"ON"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"ON"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"ON"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"OFF"}
milight/updates/0x2/rgbw/0 {"device_id":2,"device_type":"rgbw","group_id":0,"state":"ON"}
```

You could verify by putting ESPMH in listen mode while you use the remote. Paste in the packet captures and we can take a look.

kociubin, 2017-07-15:
You're right.   The issue is with the remote.  There is a bug with the handling of the OFF buttons.  The remote is sending an incorrect group in some cases.  Strangely, the ON buttons are fine...  

Would you consider adding the ability to publish the ID of the last button press via MQTT?  That would solve the issue because the button ID is always sent correctly by the remote.  ( i.e. RGBW_ALL_ON  is always sending 0x01 and RGBW_ALL_OFF is always sending 0x02)

More details on the issue.  

Issue1:
-Reset the board (to clear any history)
-Press Group1 OFF
-MQTT reports group_id of 0 instead of 1: 
"device_id":16184,"device_type":"rgbw","group_id":0,"state":"OFF"
-Protocol Trace:
rgbw packet received (7 bytes):
Request type  : B8
Device ID     : 3F38
b1            : 5C
b2            : 00     <== looks like the remote is incorrectly sending group 0
b3            : 04
Sequence Num. : 9A

Issue2:
-Reset the board (to clear any history)
-Press the Group1 Off button
-Press the Master Off button
-MQTT reports group_id of 1 instead of 0: 
"device_id":16184,"device_type":"rgbw","group_id":1,"state":"OFF"
-protocol trace:
rgbw packet received (7 bytes):
Request type  : B8
Device ID     : 3F38
b1            : 5C
b2            : 01        <===WRONG.  Should be 0. 
b3            : 02
Sequence Num. : AF

sidoh, 2017-07-15:
Ah, interesting. This makes sense.

Given this, it probably makes more sense to determine the group from the button ID for on/off commands. I think that'd solve your problem.

Also happy to publish the button ID as well if it'd be helpful, but seems like this it'd do the trick.

kociubin, 2017-07-15:
Thank you for looking into it.  Either solution would work.  Publishing the button ID would be preferable because it would offer more flexibility.

sidoh, 2017-07-15:
btw, it's sending it as a command rather than the raw button ID, but I think it should be 1-to-1. I did add some logic to include the button_id if the packet wasn't handled, though.

kociubin, 2017-07-15:
Tested in 1.4.1.  Working perfectly!

sidoh, 2017-07-15:
Awesome! Thanks for bringing these things up :)

closed 2017-07-15 by sidoh