sidoh/esp8266_milight_hub #100: rgbw: No way to get back to white in HASS

r4nd0mbr1ck, 2017-07-17:
Not sure if I am missing something but using the HASS UI, there doesn't seem to be a way to switch a RGBW bulb back to white when it is currently in colour. 

I am thinking one way to achieve this maybe to hijack the color temperature setting (or white value?) and force any color temperature command to switch to white mode?

Note: this is using mqtt_json

sidoh, 2017-07-17:
Ah yeah, I think someone else brought this up in the comments of another ticket. Definitely not ideal. Thanks for raising.

The UDP integration handles this by special-casing the user selecting a color near white. This is not exactly the greatest in terms of user experience, but it feels like the "honest" way to do it.

r4nd0mbr1ck, 2017-07-18:
Can you point me to where in the code the UDP handles the near white? I couldn't figure it out.

How about using the effect payload from mqtt_json. It is a bit clunky because the user will have to configure the effects list exactly (say "set_white"). On the other hand, handling the effects payload opens up options for things like night_mode, disco_mode, etc. from within HASS.

sidoh, 2017-07-18:
It's [here](https://github.com/happyleavesaoc/python-limitlessled/blob/master/limitlessled/group/rgbw.py#L47), I think:

```python
    def color(self, color):
        """ Set group color.
        Color is set on a best-effort basis.
        :param color: RGB color tuple.
        """
        if color == RGB_WHITE:
            self.white()
            return
        self._color = color
        self.hue = hue_of_color(color)
```

Hijacking effects is interesting. Still need to play around with those in HASS. I'll take a look sometime soon.

bhaap, 2017-08-12:
Or just when you get a white value in the mqtt message you set the bulb into white mode? you will only get the white value in hass if you choose to change the white value slider.

This is the only thing thats holding me from switching from the official milight gateways to this instead.

sidoh, 2017-08-13:
Should just be adding some logic here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/MiLightClient.cpp#L264

Might be best to handle regardless of whether color is set via rgb or hue/sat, but not too tricky. 

I'm away for a few weeks, but if you're interested in contributing, happy to look at a PR in the meantime.

sidoh, 2017-09-10:
Fix added in v1.5.0 thanks to @r4nd0mbr1ck!

closed 2017-09-10 by sidoh