sidoh/esp8266_milight_hub #101: Saturation with RGBW?

sourgelockte, 2017-07-20:
I wonder if there is a way to control saturation on the RGBW bulbs?
In Home Assistant using the offical hubs i could mix in the white with the colours by choosing a brighter tint. These are the spot lights i am using. [image]
Thanks in advance  :+1:

sidoh, 2017-07-20:
My experience with RGBW bulbs (FUT016s) has been that the WW and RGB modes are strictly separately controlled. You're either in "white mode" or "RGB mode." Not sure if the bulbs you have are the same.

The RGB+CCT (FUT015) bulbs work exactly as you describe -- they mix in the CW/WW LEDs with the RGB LEDs to achieve saturation.

My guess is that although technically possible, the bulb firmware doesn't allow for it.

g-rom-info, 2017-08-01:
I acknowledge that : the RGBW bulb seems to be limited by its firmware. I suppose it is due to a limitation from the power supply (not sized for both RGB + White)

sidoh, 2017-08-01:
Ah, that makes a lot of sense.

sidoh, 2017-08-07:
Gonna close this one. Don't think there's anything that can be done to support saturation without changing the firmware on the bulbs, and given what g-rom-info said, probably wouldn't even want to do that.

closed 2017-08-07 by sidoh

milobt, 2018-10-27:
I have an issue related to saturation of colour with FUT103 (RGB+CCT) bulbs.
I have two official MiHubs controlling around 20 bulbs around my house and then came across your project and loved the thought of having one hub to control all groups.

I have subsequently added some LED strip controllers (LS2).

I have your hub setup on a NodeMCU(v3) and works with HA, using MQTT Json however, I cannot saturate colours unless I pick a saturated colour from the colour wheel - is this normal with this version of bulb?
If the bulb is set to a colour using the HA colour picker and then I try to vary the saturation bar, the bulbs flicker white and then back to the colour.
The same happens with Alexa, if I set the bulbs to a colour and then try and set them back to white, they flick to white and then back to the colour.

Here's my HA light config:
`  - platform: mqtt_json
    name: "Kitchen Spots"
    state_topic: "milight/states/0xAD9/rgb_cct/1"
    command_topic: "milight/cmds/0xAD9/rgb_cct/1"
    brightness: true
    rgb: true
    color_temp: true
    effect: true
    effect_list: [white_mode, night_mode, colorfade_slow, colorfade_fast, 4]
    optimistic: false
    qos: 0
    retain: true`

Are there particular 'Group state fields' that I should have selected in the Hub setup?

I also have another weird issue where not all bulbs in a group turn on or off at the same time, sometimes there is a delay or just don't come on/turn off.

Can you help?