sidoh/esp8266_milight_hub #103: level and brightness in MQTT publish messages

coehn1, 2017-07-25:
Hi,

First off, really great job!!!
I've got a question. When I change the brightness of a bulb, the MQTT published message contains the 'brightness  value, 0-255. And the 'state', ON or OFF. Is it possible to also send the 'level', 0-100 and the 'status' values? So the normal commands AND the redundant HomeAssistant's?

Kind regards,

Niels Coehorst
The Netherlands

sidoh, 2017-07-25:
Certainly possible. I was thinking it made sense to optimize more for the size of the messages. Probably not too hard to have something like a format string with sane defaults.

coehn1, 2017-07-25:
Sorry, I don't understand you.

Verstuurd vanaf mijn iPhone

sidoh, 2017-07-25:
Heh, sorry.

Guess I'm saying I'd prefer to not send all variants of each setting (brightness + level, status + state, hue + color, etc.) because the MQTT messages would start getting pretty big. 

I was proposing exposing a message format setting, something like:

```
{"state":"{{state}}", "level":"{{level}}"}
```

which you could fiddle with to get messages in the format you like.

coehn1, 2017-07-25:
Hi Chris,

Thanks, great idea!!

Verstuurd vanaf mijn iPhone

sidoh, 2017-11-08:
Another potential approach: list the keys you want to include in the state payload.  Might default to something like `state, brightness, color, color_temp, bulb_mode, mode`, but you could swap out `level` for `brightness` if you wanted.

This is less flexible, but I think would perform better, and is probably less confusing. 

@Skinah does this seem like it'd do the trick for you?

Skinah, 2017-11-08:
Sounds like a great idea as that way the packets can be made shorter to use less bandwidth. Level already works sending mqtt so it would be great to have it in the state like your idea details.

sidoh, 2017-11-11:
This has been added to [1.6.0-rc3](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.6.0-rc3).  Demo of UI:

[image]

sidoh, 2017-11-11:
Please feel free to re-open if this does not adequately address the issue.

closed 2017-11-11 by sidoh