sidoh/esp8266_milight_hub #107: Hyperion support

starkillerOG, 2017-08-27:
I have been looking to control my hyperion system with a physical hardware colorwheel remote for a while. Hyperion is an ambilight application that runs on an Raspberry Pi and can control lights based on the video images it grabs ([Hyperion site](https://hyperion-project.org/wiki/Main)).
When I watch movies the system performs outstanding, but I also want to use it as ambiant lighting if my TV is off, for lights behind the TV and around the room.

I found this github project and saw that  Milight makes nice 2.4GHz color wheel remotes that are sold separately and are fearly cheap. 

Great work on decifering the protocal, I saw that this code is able to listen to the remote signals using a cheap NodeMCU + ESP8266 and a NRF24L01+ and decode the signals and sent them to HomeAssistant. 

I think it would be fearly easy to add support for hyperion because hyperion has a good [JSON interface](https://hyperion-project.org/wiki/hyperion-remote-JSON-interface) to receive commands. The ESP8266 could receive the signals from the remote and send commands to hyperion to turn on/off, to change color and to change the brightness.

Unfortunately I am completely new to using electronic boards like esp's and NodeMCU's and am not familiar with how to program them.
Maybe there is someone who would like to give this a go?

sidoh, 2017-08-27:
I'm sort of keen on keeping this project lean and focused on milight hardware. But I also want there to be integration paths for these sorts of things.

You can forward commands intercepted from remotes to an MQTT topic. Seems likely that some glue like HASS could translate the messages into whatever format Hyperion is expecting, right?

Here's a HASS automation that forwards messages from one topic to another. Shouldn't be hard to adapt it to reformat the payload:

https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant#configure-hass

starkillerOG, 2017-08-29:
Thank you for the quick response.
I think it could be done through HASS, I do not know HASS but I will look at it.

Is this github project able to pass through the hue, saturation and luminocity (HSL) values?
In the demo i do not see the hue values beeing passed allong and it also states in the limitations saction that it is not able to keep track of the colors.

Is that still the case or is it now already able to pass through all the information required to reconstruct the RGB values of the color?

sidoh, 2017-09-07:
Sorry for the late reply here.

The passthrough events are direct translations of RF packets, which are basically delta commands of the form "set hue to X." In order to send the full state of the bulb, this project would need to keep track of state for every device ID and every group, and it doesn't currently do that. Certainly something on the roadmap, though.

starkillerOG, 2017-09-08:
Alright, I think I get it.
So If you would change the Hue, saturation and luminosity straight after each other on the remote, each command would be sepretly passed through to HASS and then HASS would have the full RGB data.
But if you only change the Hue, only the Hue gets updated on HASS and the old previous values that are still in memory of the HASS will be applied for saturation and luminosity.

Is that correct?

Thank you for all your help and explanation!

sidoh, 2017-09-08:
That's a pretty fair understanding. The only thing I'd add is that I'm not aware of a way to update Hue or Sat separately with HASS. You need to supply full color data. This is basically why esp8266_milight_hub would need to keep track of state itself in order to keep color state updated within HASS.

starkillerOG, 2017-09-08:
Alright, I get it.
So colors will not yet work with HASS.
Do you think eso8266_milight_hub will soon be able to keep track of states, or is this a long term project?

sidoh, 2017-09-10:
I think it's probably the next big feature to take on, but no idea when I'll get around to that. :)

starkillerOG, 2017-09-16:
Sorry for all the questions, but I am fairly new to programming boards.
I bought a "[CH340 NodeMcu V3 Lua, (ESP8266)](https://www.aliexpress.com/item/V3-4-M-bytes-32-Mbits-FLASH-Lua-NodeMcu-placa-de-desarrollo-de-Redes-WIFI-Basado/32711244582.html)".
Will the esp8266_milight_hub software run on that board?
I noticed to late (after I already bought it) that this is a NodeMcu V3 instead of a V2.
I installed PlatformIO and there I can only select "NodeMCU 0.9" or "NodeMCU 1.0".

sidoh, 2017-09-16:
It should work just fine. I've tested with a v3. As far as I can tell v3 is just a knockoff with a different form factor and a different UART module. I cover it in [the blog post](http://blog.christophermullins.com/2017/02/11/milight-wifi-gateway-emulator-on-an-esp8266/).

Just make sure you have the right USB drivers installed, and you should be able to use the precompiled nodemcuv2 image, or flash using platformio.

sidoh, 2017-09-19:
@starkillerOG, I'm going to close this as I think it makes sense for some other piece (like HomeAssistant) to handle an integration with Hyperion, but please feel free to re-open if you'd like to discuss further.

closed 2017-09-19 by sidoh

starkillerOG, 2017-11-14:
@sidoh finally back home and installed and tested. It works perfectly!
Thank you for this amazing project!
I ended up using Mosquitto as the MQTT broker and Eventghost to get the MQTT messages and execute some phython code to translate and sent the commands to Hyperion each time it receives an MQTT message.

I have it all working and I am verry pleased with the responsibility of the setup.
The only disadvantage is that I now need an extra device (in this case windows PC) that is running Mosquitto and eventghost.

I do not know how much work it would be to translate my phython code to integrate it in the esp8266_milight_hub, but it would be verry need to get rid of that extra device running Mosquitto and eventghost.

If your intrested in my phython code, just sent me a message and I will clean it up a bit and translate my comments to englisch.

Thanks for all your help!

Skinah, 2017-11-14:
I know a few people including myself that would love the python code, any chance of a copy or a github project?

starkillerOG, 2017-11-14:
Yes sure, I will start cleaning up and translating my comments, will post it here in I guess about 2 hours.

starkillerOG, 2017-11-14:
The phyton code I wrote to use in Eventghost:

```
#=====================================================================================================================================
# Settings section, the values in this section can be altered to personal prefrence and system configuration.
# at least the values under settings need to be set to your specific system.
#=====================================================================================================================================

# Settings
Hyperion_IP = '192.168.1.??'                # IP adress of the device running Hyperion (often an Rasberry Pi), syntax: '192.168.1.??'
Hyperion_port_list = ['19444','19446']      # Full port list of all Hyperion instances, each instance is a diffrent light group. Syntax for two lights: ['19444','19446']
Hyperion_mode_list = ['HDMI','Color']       # Define mode list for the mode button
Hue_corr = 9                                # Color wheel offset correction, to account for a remote apearing to have the color wheel rotated a bit with respect to the hue value.
ports_group_0 = ['19444','19446']           # The lights in group 0 (top ON/OFF button), a list of hyperion ports for the instances of hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []
ports_group_1 = ['19444']                   # The lights in group 1 (bottom ON/OFF button 1), a list of hyperion ports for the instances of hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []
ports_group_2 = ['19446']                   # The lights in group 2 (bottom ON/OFF button 2), a list of hyperion ports for the instances of hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []
ports_group_3 = []                          # The lights in group 3 (bottom ON/OFF button 3), a list of hyperion ports for the instances of hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []
ports_group_4 = []                          # The lights in group 4 (bottom ON/OFF button 4), a list of hyperion ports for the instances of hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []

# Check if all global variables exist that we need, these are the variables that are received from the Milight Remote
if not hasattr(eg.globals, 'Milight_Hue'):
    eg.globals.Milight_Hue = 240                        # Hue, default color: blue
if not hasattr(eg.globals, 'Milight_Sat'):
    eg.globals.Milight_Sat = 100                        # Saturation / Grey value, always 100 best.
if not hasattr(eg.globals, 'Milight_Lum'):
    eg.globals.Milight_Lum = 50                         # Luminosity, 0 to 50 is intensity, 50 to 100 goes to white.
if not hasattr(eg.globals, 'Milight_Power'):
    eg.globals.Milight_Power = 'ON'                     # Hyperion power status ON/OFF.
if not hasattr(eg.globals, 'Milight_Mode'):
    eg.globals.Milight_Mode = 'Color'                   # Hyperion mode, HDMI grabbing mode or solid color mode
if not hasattr(eg.globals, 'Hyperion_ports'):
    eg.globals.Hyperion_ports = ['19444']               # Hyperion instance that is controlled for this specific button press/command

#=====================================================================================================================================
# Import required modules & receive the command from the MQTT message as the payload that trigged this macro
#=====================================================================================================================================

# import the required modules
import ast
import sys
import socket
import json
import colorsys

# get the received command
try:
    mess = ast.literal_eval(eg.event.payload)
except:
    mess = {}
    print("Invalid payload/command")
    sys.exit(1)

#=====================================================================================================================================
# Overwrite the global variables that are received from the Milight remote, keep the rest of the global variables the same as the previous stored values.
#=====================================================================================================================================

# Button ID
ButtonID = mess.get("button_id", "")
if ButtonID == 0: # invallid button
    sys.exit(1)
    print("Invallid button")

# ON/OFF state
pow = eg.globals.Milight_Power # original power
eg.globals.Milight_Power = mess.get("state", eg.globals.Milight_Power)
if eg.globals.Milight_Power != pow: # If the power state changes, check if the saturation & luminosity are set to verry dim, if so, set back to default, to prevent a situation were the lights are so dim that the on/off button apears to do nothing.
    if (eg.globals.Milight_Sat < 30 or eg.globals.Milight_Lum < 10):
        eg.globals.Milight_Sat = 100
        eg.globals.Milight_Lum = 50

[... 85 lines cut ...]
        # initilize connection list
        eg.globals.Hyperion_s = []
[... cut ...]
    #res = s.recv(1024)     # the response from Hyperion
    #print(res)

#=====================================================================================================================================
# Diagnostics
#=====================================================================================================================================

#print(rgb_raw)
print("RGB: " + str(rgb))
#print(Hex)
#print(command)
#print("ButtonID: " + str(ButtonID))
#print("mode button" + str(mode))
#print(eg.globals.Milight_Mode)
```

starkillerOG, 2017-11-14:
I used [this MQTT plugin for Eventghost](http://eventghost.net/forum/viewtopic.php?f=2&t=6725) to receive the MQTT messages in Eventghost and trigger the execution of the phython script above.

starkillerOG, 2017-11-27:
I improved my initial python code to include an internal MQTT listener.
Therefore Eventghost is no longer needed for this Hyperion color wheel remote.
I have the python script running on the same rasberry pi that runs my hyperion instances and is running openelec.

The inproved code:
```

#=====================================================================================================================================
# Settings section, the values in this section can be altered to personal preference and system configuration.
# at least the values of the "MQTT listener Settings" section need to be set to your specific system.
#=====================================================================================================================================

# feedback
feedback = 0											# If executed in a terminal get messages about the color and ports on each message, syntax: 0=feedback off, 1=feedback on.
# MQTT listener Settings
broker_address = "192.168.1.000"						# IP address of the device running the MQTT broker (often an Raspberry Pi, an ESP8266 or an PC), syntax: "192.168.1.000"
broker_port = 1883										# Port of the MQTT broker (often port 1883), syntax: 1883
broker_username = "abcdef"								# User name of the MQTT broker, syntax: "abcdefg"
broker_password = "abcdef"								# Password of the MQTT broker, syntax: "abcdefg"
broker_topic = "milight"								# MQTT broker topic to subscribe to, syntax: "abcdefg"
Paho_module_path = "/storage/.config/modules-custom"	# Path to the Python Paho.mqtt.client module, because OpenElec does not support normal installation of python modules the path to the files of the module must be appended.
# Hyperion Settings
Hyperion_IP = '127.0.0.1'                				# IP address of the device running Hyperion (often an Raspberry Pi), syntax: '192.168.1.000'
Hyperion_port_list = ['19444','19446']      			# Full port list of all Hyperion instances, each instance is a different light group. Syntax for two lights: ['19444','19446']
Hyperion_mode_list = ['HDMI','Color']       			# Define mode list for the mode button
# Milight remote Settings
Hue_corr = 9                                 			# Color wheel offset correction, to account for a remote appearing to have the color wheel rotated a bit with respect to the hue value.
ports_group_0 = ['19444','19446']           			# The lights in group 0 (top ON/OFF button), a list of Hyperion ports for the instances of Hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []
ports_group_1 = ['19444']                   			# The lights in group 1 (bottom ON/OFF button 1), a list of Hyperion ports for the instances of Hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []
ports_group_2 = ['19446']                   			# The lights in group 2 (bottom ON/OFF button 2), a list of Hyperion ports for the instances of Hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []
ports_group_3 = []                          			# The lights in group 3 (bottom ON/OFF button 3), a list of Hyperion ports for the instances of Hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []
ports_group_4 = []                          			# The lights in group 4 (bottom ON/OFF button 4), a list of Hyperion ports for the instances of Hyperion that you want to control in this group, syntax for 1 light: ['1994'], syntax for 2 lights: ['19444','19446'], syntax to disable the group behaviour of this button: []

# Initialize all the global variables that we need, these are the global variables that are received from the Milight Remote.
global Milight_Hue, Milight_Sat, Milight_Lum, Milight_Power, Milight_Mode, Hyperion_ports
# set the defaults
Milight_Hue = 240                        				# Hue, default color: blue
Milight_Sat = 100                        				# Saturation / Grey value, always 100 best.
Milight_Lum = 50                         				# Luminosity, 0 to 50 is intensity, 50 to 100 goes to white.
Milight_Power = 'ON'                     				# Hyperion power status ON/OFF.
Milight_Mode = 'Color'                   				# Hyperion mode, HDMI grabbing mode or solid color mode
Hyperion_ports = ['19444']               				# Hyperion instance that is controlled for this specific button press/command
#=====================================================================================================================================
# Import required modules
#=====================================================================================================================================

import ast
import socket
import json
import colorsys
import sys
sys.path.append(Paho_module_path)		# add the path to the MQTT module
import paho.mqtt.client as mqtt 		# import the Paho MQTT module

#=====================================================================================================================================
# Define the code to execute on receiving an MQTT message
[... cut ...]
print("connecting to broker")
client.username_pw_set(broker_username, broker_password)	# set the username and password for the MQTT broker
client.connect(broker_address, port=broker_port) 			# connect to the broker
print("Subscribing to topic",broker_topic)
client.subscribe(broker_topic)								# subscribe to the broker topic

# Start the infinite loop to listen for messages, when an MQTT message with payload "stop" is sent the loop will exit and the script stops.
client.loop_forever(retry_first_connection=False) 
```

starkillerOG, 2017-11-27:
The only things needed now are:
- the physical color wheel remote.
- An ESP8266 with an receiver running the esp8266_milight_hub software
- An raspberry pi running Hyperion (and the python code above)
- An MQTT broker running on a device on the netwerk (I will use an second ESP8266 running [the martin-ger/esp_mqtt firmware](https://github.com/martin-ger/esp_mqtt), just ordered it).

For the script above to work it needs the python paho.mqtt.client module, and because you cannot just install python modules the normal way on openelec, I installed the python module on a diffrent computer, coppyed the files of the module to the raspberry pi running openelec and then added the path to that module in the code above (Paho_module_path variable) so it can be imported in the code above and used.

I am verry happy with the way it works now, and am happy to assist if someone else would like to implement this hyperion color wheel remote.

starkillerOG, 2017-11-27:
I run the python script on startup by adding the last line to my autostart.sh in the .config folder on the raspberry pi running openelec:

```
/storage/hyperion/bin/hyperiond.sh /storage/.config/hyperion.config.json > /storage/logfiles/hyperion.log 2>&1 &
sleep 10
/storage/hyperion/bin/hyperiond.sh /storage/.config/hyperion.configBank.json > /storage/logfiles/hyperionBank.log 2>&1 &
sleep 1
python /storage/.config/MilightMQTT.py
```

starkillerOG, 2017-11-27:
@Skinah @sidoh if you have any questions on how I manged to get it working, just ask.