sidoh/esp8266_milight_hub #109: Exception thrown on latest dev branch

s73v3r, 2017-08-27:
On dev 1.5, clicking the Pair, Unpair or White buttons on the web UI causes an exception to be thrown. 

```
Exception (3):
epc1=0x4021d61c epc2=0x00000000 epc3=0x00000000 excvaddr=0x4025eb38 depc=0x00000000

ctx: cont
sp: 3fff1890 end: 3fff1e10 offset: 01a0

>>>stack>>>
3fff1a30:  00000055 3fff452c 3fff4680 4021d617
3fff1a40:  feefeffe feefeffe 3fff4530 4020f2d7
3fff1a50:  feefeffe feefeffe feefeffe feefeffe
3fff1a60:  00000000 3fff1ac0 3fff1ad0 3ffe9cc4
3fff1a70:  401075dc 3fff4504 00000006 3fff3e9c
3fff1a80:  3ffe9cc0 00000001 00000008 00000001
3fff1a90:  00000000 00000000 00000000 3fff1a21
3fff1aa0:  00000002 feefeffe feefeffe 3fff457a
3fff1ab0:  3fff4530 3fff452c 00000001 4020ed69
3fff1ac0:  00000010 00000000 00000000 0000235c
3fff1ad0:  3fff3e9c 3fff452c 00000007 4020ed20
3fff1ae0:  3ffe97fc 00000001 3fff4504 4020dc93
3fff1af0:  3fff452c 4020ecf8 3fff4054 40210491
3fff1b00:  3fff65c8 3fff1b50 3fff4054 3fff4504
3fff1b10:  00000032 3fff406c 3fff4504 4020dd1c
3fff1b20:  3fff65b8 3fff1b50 3fff4504 4020df76
3fff1b30:  00000000 3fff536c 3fff4504 4020e05c
3fff1b40:  3fff65b8 3fff536c 3ffe9804 4020e3d8
3fff1b50:  3fff641c 0000000f 00000006 3fff1bc2
3fff1b60:  3fff25dc 00000001 3fff1b90 3fff1be0
3fff1b70:  000000ff 3fff536c 0000cd86 3fff1be0
3fff1b80:  00000001 0000cd86 3fff4504 3fff1be0
3fff1b90:  3fff25dc 3fff536c 0000cd86 4021a186
3fff1ba0:  3fff63ec 3fff536c 0000cd86 4021abaf
3fff1bb0:  77626772 3fff5300 3fff1c60 40202256
3fff1bc0:  3fff0031 000006c9 00000007 3fff1be0
3fff1bd0:  64637830 3f003638 00000000 4021aa67
3fff1be0:  00000000 00000000 00000000 000006c9
3fff1bf0:  00000020 3fff675c 3fff6604 3fff1c50
3fff1c00:  00000020 3fff1cbc 3fff1bb0 00000000
3fff1c10:  00000004 0000002c 00000004 3fff1bc0
3fff1c20:  00000000 00000001 3fff1c2c 00000001
3fff1c30:  3fff1bd0 00000000 00000006 0000002c
3fff1c40:  00000006 3ffe97f0 3fff1cb0 3fff65ac
3fff1c50:  00000200 3fff1d00 3fff1cb0 4020237d
3fff1c60:  3fff675c 0000000f 00000004 3fff5874
3fff1c70:  0000000f 00000001 3fff63ec 0000000f
3fff1c80:  00000006 3fff1d00 3fff1cb0 40100688
3fff1c90:  3fff5854 3fff5354 3fff1be0 3fff1bc0
3fff1ca0:  3fff65b8 3fff5874 3fff5854 40202180
3fff1cb0:  3fff1c90 00000000 00000017 3fff1d00
3fff1cc0:  3fff1d50 3fff655c 3fff60bc 4021ac28
3fff1cd0:  3fff1d50 00000001 3fff1d00 4021b594
3fff1ce0:  74616700 79617765 78300073 36386463
3fff1cf0:  62677200 00310077 3fff60bc 4021b53a
3fff1d00:  3fff1ce0 3fff1cf6 00000017 4020212f
3fff1d10:  00000015 3fff61b4 3fff1d00 40202205
3fff1d20:  3fff1ce0 3fff53ac 3fff1d50 4020230c
3fff1d30:  00000000 00000000 3fff536c 4021b504
3fff1d40:  3fff60bc 3fff53ac 3fff536c 402141f1
3fff1d50:  3fff655c 0000001f 00000017 00000000
3fff1d60:  00000000 00000000 00000000 402097fc
3fff1d70:  00000003 00000014 00000014 00000001
3fff1d80:  3fff6434 00000014 00000010 3fff0de8
3fff1d90:  00000001 3fff5390 3fff536c 4021b2eb
3fff1da0:  000001c8 3fff5390 3fff536c 4021b3ca
3fff1db0:  3ffe9180 00000000 000003e8 40220a04
3fff1dc0:  00000000 3fff511c 3fff5478 4020b18c
3fff1dd0:  3fffdad0 00000000 3fff5354 40219d24
3fff1de0:  00000000 00000000 3fff0de0 40220a43
3fff1df0:  3fffdad0 00000000 3fff0de0 40202da8
3fff1e00:  feefeffe feefeffe 3fff0df0 40204c3c
<<<stack<<<
```

sidoh, 2017-08-27:
Does this happen deterministically? Were earlier versions working for you?

I've been using this branch for weeks without any crashes, and I'm not able to reproduce this issue (pair/unpair/white commands for RGBW causing crashes).

There's also not a lot going on at the method in the top of your stack trace ([link](https://github.com/nRF24/RF24/blob/master/RF24.cpp#L742)).

I'd try:

1. Enabling some debug flags (see `platformio.ini`). `DEBUG_PRINTF` would probably be the most helpful.
2. Erasing the ESP and reflashing the firmware. Skip uploading SPIFFS -- flashing shouldn't cause problems, but it's also not doing any good.

sidoh, 2017-09-04:
Going to close because I can't reproduce, but please feel free to reopen if you're still having trouble.

grote-beer, 2017-09-23:
When compiling the 1.5 binary myself I get the same exception at RF24::stopListening(). When using the pre-compiled binary everything works fine. Can it be an issue with you using a different version of a library?

The serial output with DEBUG_PRINTF:

`0x4020dd14: MiLightClient::flushPacket() at ??:?
3fff1b30:  3fff6d80 3fff5374 3fff4504 4020deb5
0x4020deb5: MiLightClient::updateStatus(MiLightStatus) at ??:?
3fff1b40:  3fff6d80 3fff5374 3ffe99fc 4020e74c
0x4020e74c: MiLightClient::update(ArduinoJson::JsonObject const&) at ??:?
3fff1b50:  00000019 00000001 00000000 4020ecf9
0x4020ecf9: NRF24MiLightRadio::configure() at ??:?
0x4020db42: MiLightClient::switchRadio(MiLightRadioType) at ??:?

3fff1b70:  00000001 3fff5374 00001962 3fff1be0
3fff1b80:  00000001 00001962 3fff4504 3fff1be0
3fff1b90:  3fff25dc 3fff5374 00001962 402145b0
3fff1ba0:  3fff606c 3fff5374 00001962 40214fe8
3fff1bb0:  77626772 3fff5300 3fff1c60 40202256

3fff1bc0:  3fff0031 000007ec 00000007 3fff1be0
3fff1bd0:  38393436 3fff5300 00000000 40214e9e`

sidoh, 2017-09-23:
Interesting. Can you share what board you're using, and what command you used to compile/upload firmware?

grote-beer, 2017-09-23:
I use a Wemos D1, an Arduino uno form-factor board power by an ESP12.
To compile I use the command from the  README: 
```pio run -e esp12 -t upload```

sidoh, 2017-09-23:
Gotcha. And you were using the pre-compiled image for `esp12` as well?

grote-beer, 2017-09-23:
Yes but that one does work, it should give the same binary right?

sidoh, 2017-09-23:
Yes, I would think so. All I can figure is that some environment difference is causing an older RF24 library to get pulled in.

Could you see if the produced binaries are the same? Maybe check md5sum for the pre-compiled image and `.pioenvs/esp12/firmware.bin`

sidoh, 2017-09-23:
And could you paste the compile log with the verbose flag enabled?

```
pio run -e esp12 -v
```

grote-beer, 2017-09-23:
The compile log: https://pastebin.com/xE7UzQrg

The binaries are indeed different
md5sum of my own binary: f35b97ac5bb5c544d886b3d962c75f73
md5sum of precompiled binary: 61b73160e9b0a5e74d3e3591b2409269

sidoh, 2017-09-23:
Hmm... versions look the same as far as I can tell.  Do you know when you last installed platformio? Could you try running `pio upgrade` and `pio platform update`?

grote-beer, 2017-09-26:
I installed platformio that morning so running the update made no difference. I did fix my problem by manually reverting the [changes](https://github.com/nRF24/RF24/commit/012023916138a9685e4cf8c495e3c5d33e94a39a) you made in the NRF library. Now the esp doesn't crash anymore.

So the problem was most likely a different 'esp-arduino-core-library' version. If you can tell me how to find the version (if that is even possible) I would gladly provide you with this information.

sidoh, 2017-09-26:
Gotcha, sounds like that's the issue.  I probably missed some progmem cleanup.

the output of this command should provide the relevant info:

```
pio platforms show espressif8266
```

grote-beer, 2017-09-26:
pio platform show espressif8266:

```
espressif8266 ~ Espressif 8266
==============================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.

Version: 1.4.0

Home: http://platformio.org/platforms/espressif8266

Repository: https://github.com/platformio/platform-espressif8266.git

Vendor: https://espressif.com/

License: Apache-2.0

Frameworks: arduino, esp8266-rtos-sdk, simba

Packages
--------

Package framework-esp8266-rtos-sdk

----------------------------------
Type: framework

Requirements: >=1.5.0-beta

Installed: No (optional)

Package toolchain-xtensa

------------------------
Type: toolchain

Requirements: ~1.40802.0

Installed: Yes
Version: 1.40802.0

Original version: 4.8.2
Description: xtensa-gcc

Package framework-simba

-----------------------
Type: framework

Requirements: >=7.0.0

Installed: No (optional)

Package tool-esptool

--------------------
Type: uploader

Requirements: ~1.409.0

Installed: Yes
Version: 1.409.0

Original version: 4.9
Description: esptool-ck

Package tool-mkspiffs

---------------------
Type: uploader

Requirements: ~1.102.0

Installed: Yes
Version: 1.102.0

Original version: 1.2
Description: Tool to build and unpack SPIFFS images

Package tool-espotapy

---------------------
Type: uploader

Requirements: ~1.0.0

Installed: Yes
Version: 1.0.0

Original version: None
Description: This script will push an OTA update to the ESP

Package framework-arduinoespressif8266

--------------------------------------
Type: framework

Requirements: ~1.20300.1

Installed: Yes
Version: 1.20300.1

Original version: 2.3.0
Description: Arduino Wiring-based Framework (ESP8266 Core)

Package sdk-esp8266

-------------------
Requirements: ~1.10502.0

Installed: No (optional)

Boards
------
ID                    MCU            Frequency  Flash   RAM    Name

nodemcuv2             ESP8266        80Mhz     4096kB  80kB   NodeMCU 1.0 (ESP-12E Module)
wifinfo               ESP8266        80Mhz     423kB   80kB   WifInfo
esp01_1m              ESP8266        80Mhz     1024kB  80kB   Espressif Generic ESP8266 ESP-01 1M
d1_mini               ESP8266        80Mhz     4096kB  80kB   WEMOS D1 mini
espduino              ESP8266        80Mhz     4096kB  80kB   ESPDuino (ESP-13 Module)
esp12e                ESP8266        80Mhz     4096kB  80kB   Espressif ESP8266 ESP-12E
huzzah                ESP8266        80Mhz     4096kB  80kB   Adafruit HUZZAH ESP8266
phoenix_v1            ESP8266        80Mhz     1019kB  80kB   Phoenix 1.0
phoenix_v2            ESP8266        80Mhz     1019kB  80kB   Phoenix 2.0
d1                    ESP8266        80Mhz     4096kB  80kB   WEMOS D1 (Retired)
esp01                 ESP8266        80Mhz     512kB   80kB   Espressif Generic ESP8266 ESP-01 512k
esp07                 ESP8266        80Mhz     4096kB  80kB   Espressif Generic ESP8266 ESP-07
esp_wroom_02          ESP8266        80Mhz     4096kB  80kB   ESP-WROOM-02
sparkfunBlynk         ESP8266        80Mhz     4096kB  80kB   SparkFun Blynk Board
nodemcu               ESP8266        80Mhz     4096kB  80kB   NodeMCU 0.9 (ESP-12 Module)
d1_mini_lite          ESP8266        80Mhz     935kB   80kB   WEMOS D1 mini Lite
espresso_lite_v1      ESP8266        80Mhz     4096kB  80kB   ESPresso Lite 1.0
espresso_lite_v2      ESP8266        80Mhz     4096kB  80kB   ESPresso Lite 2.0
esp8285               ESP8266        80Mhz     423kB   80kB   Generic ESP8285 Module
espino                ESP8266        80Mhz     4096kB  80kB   ESPino
espinotee             ESP8266        80Mhz     4096kB  80kB   ThaiEasyElec ESPino
espectro              ESP8266        80Mhz     4096kB  80kB   ESPectro Core
thingdev              ESP8266        80Mhz     512kB   80kB   SparkFun ESP8266 Thing Dev
modwifi               ESP8266        80Mhz     2048kB  80kB   Olimex MOD-WIFI-ESP8266(-DEV)
thing                 ESP8266        80Mhz     512kB   80kB   SparkFun ESP8266 Thing
esp210                ESP8266        80Mhz     4096kB  80kB   SweetPea ESP-210
```

sidoh, 2017-09-26:
Actually those are all of the same versions I have... weird.

I noticed you were compiling from a Windows machine.  Do you have access to a Linux machine you could try on?  Just realized that the pre-compiled binaries are built via Travis on a clean-slate container, so platform versions probably can't be the issue if yours are recent.

I'll do another sweep of the code for weird/improper progmem usages. The problem was that I was trying to stuff everything into progmem, but because RF24 disabled all of the macros, any improper usages didn't cause issues because they were no-oped.

sidoh, 2017-09-26:
Could you try compiling from this tag?

https://github.com/sidoh/esp8266_milight_hub/tree/1.6.0-dev3

Found a progmem thing that shouldn't have been there.

grote-beer, 2017-09-29:
I will come back to this after the weekend, right now I don't have access to my esp board.

sidoh, 2017-10-02:
Great, thanks.

Actually if you could provide the firmware image you're using (both .bin and .elf) so I could tinker as well it'd be helpful.

grote-beer, 2017-10-08:
I finally tried the dev3 branch, unfortunately I still get a crash with my own binary. In the zip the esp12 folder from \.pioenv

https://www.dropbox.com/s/xox43nehch8buh0/esp12.zip?dl=0

sidoh, 2017-10-08:
Okay, not shocking, but it crashes for me too.

I guess it's sort of hard for me to iterate because my compiler isn't reproducing the issue.  But I think what I'd try next is yanking the variable RF24#stopListening is using out of progmem.  Committed this change to a branch on my fork of RF24 (https://github.com/sidoh/RF24/commit/e52d091912ba146f9089a69994fe1d0039b91863).

Can you try compiling with this change?  Just update your platformio.ini like so:

```
diff --git a/platformio.ini b/platformio.ini
index f96163c..fb9ef65 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -13,7 +13,7 @@ board_f_cpu = 160000000L
 lib_deps_builtin =
   SPI
 lib_deps_external =
-  sidoh/RF24
+  sidoh/RF24#remove_progmem
   WiFiManager
   ArduinoJson
   PubSubClient
```

sidoh, 2017-10-08:
Also, I'd think our versions are consistent, but can you double-check the version of your xtensa compiler?

```
$ /Users/cmullins/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-gcc --version
xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

grote-beer, 2017-10-09:
Success! 
With your latest branch of the RF24 library I managed to compile a working binary. Thanks for helping and if I can provide you with some more information just let me know

sidoh, 2017-10-09:
Okay, that's super weird.  The change I made removed what appeared to be entirely correct usages of progmem features.  I think this would basically only happen if your compiler didn't put this array in progmem, but the code accessing it was still expecting it to be in progmem.

This change might actually be a positive one: the progmem array is only a few bytes, but is accessed from some pretty important methods.  Entirely possible I'm missing something, but saving a couple of bytes at the cost of slowing down the radio does not seem like a good tradeoff.

I would like to at least be able to reproduce this myself.  Can you tell me more about your environment?  What version of Windows?

sidoh, 2017-10-09:
For my own notes: running with this change doesn't have material impact on transmit performance:

** With **
```
{:number=>66.0,
 :sum=>9099.0,
 :variance=>92.63292011019281,
 :standard_deviation=>9.624599737661448,
 :min=>"132",
 :max=>"181",
 :mean=>137.86363636363637,
 :mode=>"134",
 :median=>134.5,
 :range=>49.0,
 :q1=>134.0,
 :q2=>134.5,
 :q3=>138.75}
```

** Without **
```
{:number=>55.0,
 :sum=>7587.0,
 :variance=>89.90611570247931,
 :standard_deviation=>9.481883552463577,
 :min=>"133",
 :max=>"178",
 :mean=>137.94545454545454,
 :mode=>"134",
 :median=>135.0,
 :range=>45.0,
 :q1=>134.0,
 :q2=>135.0,
 :q3=>137.0}
```

grote-beer, 2017-10-09:
I'm compiling on Windows 10 pro x64 (build 15063). With a recently installed version of platformIO

sidoh, 2017-10-30:
I have no idea why the PROGMEM stuff causes issues, but I'm just going to roll with it for now:

https://github.com/sidoh/RF24/commit/da2075084c89f6d3a8318dfe16987db8b08a23ad

Figuring out the why would take expertise I don't have right now.

closed 2017-10-30 by sidoh