sidoh/esp8266_milight_hub #110: Web app not installed

kurniawan77, 2017-09-02:
flashed on Wemos D1 mini with PlatformIO i get the following when entered http://<ip_of_esp>:

[image]

platformio.ini:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[common]
board_f_cpu = 160000000L
lib_deps_builtin =
  SPI
lib_deps_external =
  RF24
  WiFiManager
  ArduinoJson
  PubSubClient
  https://github.com/ratkins/RGBConverter
build_flags = !python .get_version.py -DMQTT_MAX_PACKET_SIZE=200
# -D MQTT_DEBUG
# -D MILIGHT_UDP_DEBUG
# -D DEBUG_PRINTF

;[env:nodemcuv2]
;platform = espressif8266
;framework = arduino
;board = nodemcuv2
;build_flags = ${common.build_flags} -Wl,-Tesp8266.flash.4m1m.ld -D FIRMWARE_VARIANT=nodemcuv2
;lib_deps =
;  ${common.lib_deps_builtin}
;  ${common.lib_deps_external}

[env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini
build_flags = ${common.build_flags} -Wl,-Tesp8266.flash.4m1m.ld -D FIRMWARE_VARIANT=d1_mini
lib_deps =
  ${common.lib_deps_builtin}
  ${common.lib_deps_external}

;[env:esp12]
;platform = espressif8266
;board = esp12e
;framework = arduino
;build_flags = ${common.build_flags} -Wl,-Tesp8266.flash.4m1m.ld -D FIRMWARE_VARIANT=esp12
;lib_deps =
;  ${common.lib_deps_builtin}
;  ${common.lib_deps_external}

;[env:esp07]
;platform = espressif8266
;board = esp07
;framework = arduino
;build_flags = ${common.build_flags} -Wl,-Tesp8266.flash.1m64.ld -D FIRMWARE_VARIANT=esp07
;lib_deps =
;  ${common.lib_deps_builtin}
;  ${common.lib_deps_external}

```

sidoh, 2017-09-02:
That's expected. Prior to 1.5, the UI is not shipped with the firmware. It's stored in SPIFFS and must be uploaded separately. You can upload it with

```
pio run -e d1_mini --target uploadfs
```

Alternatively, use a RC release of [v1.5.0](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.5.0-dev3), which includes the UI in the firmware image.

sidoh, 2017-09-02:
https://github.com/sidoh/esp8266_milight_hub#setting-up-the-esp

kurniawan77, 2017-09-03:
Thanks for responding very fast! 
I have read that but i never knew where to tpe that in. Maybe i wasn't clear on what machine i was on.
I am using Atom/ PlatformIO (windows10) for the first time and found out that there is a "run other target option" (F7) which showed "upload SPIFFS image". It's all working perfect now.

[image]

Thanks! And my excuse for resonding very late.

closed 2017-09-03 by kurniawan77

sidoh, 2017-09-04:
glad it's working 👍 

Those are terminal commands. :)