sidoh/esp8266_milight_hub #112: ESPXXXX access point not showing up, can't setup WIFI

jacobtarter, 2017-09-14:
I successfully flashed the .bin file using esptool to my nodemcu controller. Everything appears to be good from the command line tool.

However, I'm not seeing an ESP wireless device that I can connect to. I tried reflashing the device, and resetting it, and I'm still not seeing anything I can connect to under networks on my Linux machine, my Mac, or my Android phone.

sidoh, 2017-09-14:
I'd recommend:

* Erasing all flash with esptool: `esptool.py --port /dev/ttyUSB0 erase_flash`. Previous WiFi configs are saved in a reserved area of flash that isn't touched when you flash firmware.
* Looking at the serial logs to see what's going on. The baudrate is 9600. If you're using platformio, you should be able to do `pio device monitor -b 9600`.

jacobtarter, 2017-09-14:
I did the erase flash command, and then reflashed the bin.  Now the light on the esp doesn't come on at all (before it was solid).  Still no network visible, and when I do the pio device monitor command I see nothing being logged at all. 
Not really sure what's going on - I don't know if it's just a faulty board or what, it's new, I just got it.

sidoh, 2017-09-14:
Gotcha.

If you're not seeing anything in the serial console, something is preventing the ESP from booting. Try disconnecting any hardware (like the NRF) you've got hooked up.

jacobtarter, 2017-09-14:
It doesn't currently have anything attached to it. Just the bare board, freshly flashed with the milight hub bin.  Once I ran the erase_flash command, the light went out on the board. 
When I first got it, the light would flash when powered up. When I first flashed the milight hub bin, the light would be solid. 
Now, the light blinks while flashing (I tryed reflashing a few times), and if I hit the reset button it flashes very briefly once. But other than that no light.

At this point I'm thinking it may be a hardware issue, and likely not a problem with your software. Is there something that gets erased when you run the erase_flash command that needs to be replaced (outside of flashing your binary)? It's almost like that wiped out something essential and now the thing won't boot up at all. 

Maybe I just got unlucky and got a bad board.

jacobtarter, 2017-09-14:
I reflashed NodeMCU firmware, and still the board seems dead.
Pretty sure I just got a dud.
I highly doubt this has anything to do with your software, so you can probably just close this

sidoh, 2017-09-14:
Sounds like some pin is tied low or high when it shouldn't be. Could it be that either the flash or reset button is stuck? I don't think you'd be able to flash if the board were completely dead. 

I'll close since it seems like a hardware problem, but happy to help out if I can.

closed 2017-09-14 by sidoh