sidoh/esp8266_milight_hub #115: RGBW All On command via UDP Gateway sends CCT protocol

BrizzzUK, 2017-09-22:
Hi Chris,

First off, I want to congratulate you on a truly awesome piece of work!

I'm having an issue since upgrading from v1.3.1 to v1.5 with the RGBW "ALL ON" command (42,00,55). Since upgrading, the hub appears to be sending out a CCT protocol packet instead of the correct RGBW protocol packet.

This is what I see if I sniff the traffic sending 42,00,55 to the gateway ...
cct packet received (4 bytes):
Request type  : 5A
Device ID     : BCDD
b1            : 04
b2            : 02
b3            : 79
Sequence Num. : 79

Sending the ALL OFF command (41,00,55) still works correctly however...
rgbw packet received (4 bytes):
Request type  : B0
Device ID     : BCDD
b1            : 00
b2            : 04
b3            : 02
Sequence Num. : 3C

I've temporarily reverted back to 1.3.1 for now as a workaround.

Cheers,
Brian

sidoh, 2017-09-22:
Thanks for catching! I'll either release a patch for 1.5 or work this into 1.6 depending on how much work the features I have for 1.6 end up being.

sidoh, 2017-10-07:
Should be fixed in 1.6.0-dev7:

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.6.0-dev7

Let me know if you notice any other issues.

closed 2017-10-07 by sidoh