sidoh/esp8266_milight_hub #117: Optimize nRF radio client

sidoh, 2017-09-26:
Some context here:

https://github.com/sidoh/esp8266_milight_hub/issues/91#issuecomment-332080168

khmann, 2017-09-30:
I'm still new to git, but consider this component A: https://github.com/khmann/esp8266_milight_hub/commit/e3600cef75b102ff3be51a7afdb55ab7460fe712#diff-abacac63ba0406afaae4d03dac9589ec

I took an openmilight_pi, copied in your PL1167 file (made it work) then tweaked the packet framing.  Because your library is slightly than mine (and the existing code makes bad choices) there may be new bugs.

Component B: verify operation of RX, optimize startup and mode switching.  optimize "reverse_bits" with a lookup table?

optimize CRC with lookup table?  I've seen other applications of this, but I haven't yet got it working.  I wonder if the RGBCCT encryption could be accelerated through 32bit lookup tables and xor operations?

sidoh, 2017-09-30:
This is great.  I really like how much simpler the result is.  Love seeing this much red in a diff :)

I'll try to make time to test it out this weekend.

Basically nothing in `PL1167_nRF24.cpp` is optimized.  I didn't make a thorough attempt to understand what it was doing because it works well enough, and it wasn't what I was focused on.  But seems clear you're very familiar with what's going on, and I'm more than happy to incorporate any improvements you provide.

> (I still hate that function - don't do all the math and reprogram the syncword EVERY time you change channel!!!)

Totally open to suggestions here.  I'm not super familiar with the nRF or RF communications in general, so any high-level (or low-level, as the case may be here :) input you have is more than appreciated.

khmann, 2017-09-30:
I guess you could integrate this code as a new radio option, but only because I haven't tested with 1st generation products (my order was canceled for a second time, I just can't get them).  I send a shorter preamble (1.5 bytes instead of 3) than the original implementation but everything else should be the same in the air.

I haven't really worked on the RX code, but clearly I'm suspicious of the line commented HACK HACK HACK, though since my hardware is different I didn't want to mess with it yet (and also I'm still nervous about moving those setup functions out of recalculate_parameters, put them back if you have problems).  Also I didn't have a remote with me to test, but this was working in a different project.

I think overall there's still too much reprogramming and mode switching, when the radio should always default in RX mode and it is the responsibility of the TX routine to do any switching, however I haven't really looked at how all this is being called.

note that the old driver can't receive my packets because (contrary to proper L2 protocol design) it's trying to match the long preamble.  but you're not supposed to match a preamble (because it's a preamble), so it's their bug.

If you really want to take the radio "to the next level", we need a (priority) timer interrupt that can re-schedule 4ms into the future every time there is a TX, and so can yield() until packet is done then automatically drop back to RX mode.

As a side note, the radio can be programmed to RX two independent syncwords at once so if you want to listen for multiple remotes, all you have to do is change channels (maybe the timer).  Sometimes you don't even need to change channels because the signal bleeds over ; ) This will be more practical as I clean up some functions.

for my own stuff I want to get that reverse_bits function optimized and I assume a lookup-table will help.  And then the CRC function as well.  To optimize the rgbcct protocol 0x2x encryption, I thought about tables but my MCU doesn't have enough RAM.  For now, I control those awesome FUT105 bulbs unencrypted over rgbw protocol 0xb0 and it is good enough; 11 steps saturation and temperature.

sidoh, 2017-10-01:
Just tested this out, and it works beautifully.  Looks like roughly a 20% speed increase.  Test conditions:

* 50 repeats
* Sending 0x20 protocol RGB+CCT packets (encrypted)
* 100 samples of the same command

Without your changes, each command took an average of 180ms.  With, they were down to 140ms.  I'll work these changes (and whatever else you cook up :) into the v1.6 release.

sidoh, 2017-10-02:
I think the lookup tables for the 0x2* encoding would be huge.  There might be a much cleverer way to do it, but I think the naive method would require something like 200 KB (one table for each of 8 encoded bytes, 128 possible keys, 255 possible values).

Seems like lookup table for bit reversal saves another 10ms (5%).  Not positive it's worth the 256 bytes of RAM.

khmann, 2017-10-02:
re: bit reversal performance gain, boo... maybe if you forced the compiler to store it in internal RAM (if you have such a thing), but it'd have to be a pretty amazing gain to justify *that*.  My application may warrant it, I'm hoping to tightly sync a bunch of units (to music or whatever) so I don't plan to transmit the same packet more than once - at the very least I increment the sequence and thus recalculate CRC.

For this project "fully optimized" is probably just to cache the desired "raw bits" output and retransmit that; right now CRC and bits are recomputed each time an identical packet is retransmitted.  I'll make one pass through this file in an attempt to optimize how often the radio is reprogrammed then figure out how to retransmit better (I don't want to affect compatibility with real PL1167).

The encryption has to be "easy" somehow, unless they switched to a new MCU in the new bulbs.  My guess it was done to foil clones- I've found at least one generic product series that worked by cloning codes from the real remotes.  I wonder if anyone has tried to get into the STM8 micro using the SWIM debug interface?  A dump would reveal all... and I'd love to flash the WRGB "Saturation controller" firmware to my old bulbs.

sidoh, 2017-10-02:
> re: bit reversal performance gain, boo... maybe if you forced the compiler to store it in internal RAM (if you have such a thing), but it'd have to be a pretty amazing gain to justify that. My [...]

There's a way to put static blocks of memory in a reserved area of flash, but I'm guessing that'd defeat the point of the lookup table.  You can only access things a word (32 bits) at a time.

> For this project "fully optimized" is probably just to cache the desired "raw bits" output and retransmit that; right now CRC and bits are recomputed each time an identical packet is retransmitted. [...]

You're referring to the raw bits and CRC surrounding the raw milight packets, correct?  Makes sense to me.  Definitely happy to look at whatever you come up with. :)

> The encryption has to be "easy" somehow, unless they switched to a new MCU in the new bulbs. My guess it was done to foil clones- I've found at least one generic product series that worked by [...]

The model I imposed via the reverse engineering is pretty simple (not that I found it simple to actually reverse... -_-): compute a key `k` from the 0th byte, then the other bytes `i` are "encrypted" with the function `f_i(b) = (b + c_i(b_0)) xor k`, where `c_i(b)` is one of four seemingly random constants for each byte location depending on the remainder of the first byte mod 4.  You only need ~32 bytes of RAM if you do these computations.

It's totally possible that the real scrambling is done with a simpler function.  This is just the pattern I was able to see.

For _encoding_ you could actually have a much smaller lookup table (something on the order of 1KB) because you can always choose the same key to encode.

I haven't actually tested how much overhead this adds, but I'm guessing it's dominated by what's going on in the radio handler.

khmann, 2017-10-02:
> There's a way to put static blocks of memory in a reserved area of flash, but I'm guessing that'd defeat the point of the lookup table. You can only access things a word (32 bits) at a time.

Mm, I'd not thought of minimum access width, better sort the table right.  I probably will use a flash table for my first pass at encryption but let's give up on lookup tables here for now.  Milight has to be doing somethings easy.  Like straight-dumping the receive packet buffer into a math-co and iterating an operation... some sort of matrix and/or shift, I doubt it's actually calculated 8 bits at a time.  Also I don't think anyone has officially ruled out that it could be a PL1167 "scrambling" function.  I guess I need to build an SPI interface, verify the IC used in the RGBCCT bulb and read the STM8/PL1167 documentation.

> You're referring to the raw bits and CRC surrounding the raw milight packets, correct? Makes sense to me. Definitely happy to look at whatever you come up with. :)

Ok I'll do it.  I'm going to confine my change as much as possible to PL1167_nRF24.cpp so that these optimizations can benefit the whole community.  It probably makes the most sense to separate 'generating the bits' and 'transmitting the bits' into separate functions but maybe there's a native PL1167 function that I should emulate and we could see an improvement on both fronts.

khmann, 2017-10-09:
updated, about as optimized as it can be without splitting up transmit() into "makeBits() and sendBits()".  It'll be a minute before I have time for that... so let's try to roll this to 1.6.0

https://raw.githubusercontent.com/khmann/esp8266_milight_hub/master/lib/MiLight/PL1167_nRF24.cpp

sidoh, 2017-10-09:
Awesome!  Thanks for taking the time to look into this.  Did you happen to test with the FUT092 ("RGB+CCT") protocol?  My bulbs aren't responding after applying this change.

Also, very possible I'm missing something, but it actually seems slower (nearly 2x) with these changes:

**With**: 
```
{:number=>20.0,
 :sum=>5093.0,
 :variance=>2227.1275,
 :standard_deviation=>47.19245172694464,
 :min=>"206",
 :max=>"400",
 :mean=>254.65,
 :mode=>"207",
 :median=>248.0,
 :range=>194.0,
 :q1=>226.0,
 :q2=>248.0,
 :q3=>262.25}
```

**Without**:
```
{:number=>22.0,
 :sum=>3118.0,
 :variance=>319.38016528925624,
 :standard_deviation=>17.87121051549828,
 :min=>"132",
 :max=>"207",
 :mean=>141.72727272727272,
 :mode=>"135",
 :median=>135.0,
 :range=>75.0,
 :q1=>134.0,
 :q2=>135.0,
 :q3=>137.0}
```

khmann, 2017-10-09:
If net performance is 2X as slow, that's actually still good because this is sending 3X the packets.

A lot is wasted in reconfiguring the radio and doing the same math over.  The nRF FIFO allows 3 packets to be queued at once.  Not really knowing the library and hardware, I figured it would be really "inexpensive" to just TX the same packet 3X but apparently the "cost" is still high.  I was hoping to retain performance equal to the original implementation (yet with 3X packets), then just set retries=/2 at a higher level and still be a net win.

Do you know offhand about the SPI setup?  I think it should be set to either 2mbps (more stable with MCU?) or 4mbps (+15% faster).  8mbps has been reported as problematic...

Also I commented all your yield()s... maybe this is just blocking something above?  Maybe we should after every _radio.write(tmp, outp) ?

Not working with RGB+CCT is a bigger disappointment though.  I assume you were attempting to TX to real bulbs (remember the old code can't hear us ;), and that RGBW was working.  Did you try RX?  That's been the problem... because of mis-matched radio chips results can vary by syncword (PL1167 "trailer processing"), so different results are possible for TX/RX of CCT (5A), RGB (A4), RGBW (Bx), and RGB+CCT (20).

Hopefully it's "easy" stuff like order-of-operations, or I should always call "set RF channel" for example... like I make bad assumptions of the calls from above...

the only real change that should be related to "RGB+CCT not working but RGBW is" would have been the addition of
"_nrf_pipe[ --nrf_address_pos ] = reverse_bits( ( (_syncword3 >> 12) & 0x0f ) + 0xa0 );"
which was required to make some RX work.  Or maybe I'm not setting the packet length correctly.  I'll look at it again, maybe add some #DEBUG stuff.

sidoh, 2017-10-09:
That's a huge improvement!  Nearly 40%.  Definitely makes sense to push repeats to a lower level whenever possible.

> Do you know offhand about the SPI setup? I think it should be set to either 2mbps (more stable with MCU?) or 4mbps (+15% faster). 8mbps has been reported as problematic...

I'm not sure how the Mbps rate translates into SPI clock speed, but the ESP8266 has a 80 MHz CPU that can be overclocked to 160 MHz at essentially no cost other than increased power draw.  I'd imagine the rate at which it can drive the SPI clock is limited by that.

> Also I commented all your yield()s... maybe this is just blocking something above? Maybe we should after every _radio.write(tmp, outp) ?

Not yield()ing where necessary would cause watchdog timer crashes, but I didn't see any WDT resets with your changes, so think it's fine.  Guessing that either RF24 is yield()ing internally, or each transmit is fast enough that the yield is unnecessary.

> Not working with RGB+CCT is a bigger disappointment though. I assume you were attempting to TX to real bulbs (remember the old code can't hear us ;), and that RGBW was working. Did you try RX? That's been the problem... because of mis-matched radio chips results can vary by syncword (PL1167 "trailer processing"), so different results are possible for TX/RX of CCT (5A), RGB (A4), RGBW (Bx), and RGB+CCT (20).
> Hopefully it's "easy" stuff like order-of-operations, or I should always call "set RF channel" for example... Or maybe I'm not setting the packet length correctly. I'll look at it again, maybe add some #DEBUG stuff.
>
> the only real change that should be related to "RGB+CCT not working but RGBW is" would have been the addition of
"_nrf_pipe[ --nrf_address_pos ] = reverse_bits( ( (_syncword3 >> 12) & 0x0f ) + 0xa0 );"
which was required to make some RX work.

Correct, was trying to send to FUT015s.  I did not test RGBW actually, but I could try if it'd be helpful.  I did not try RX, but can do that later today.

khmann, 2017-10-25:
Just wanted to pop in and say I haven't forgotten this, but I don't see any issues with the library on my Pi and it's a little esoteric to try and get others to troubleshoot.  I've given in and ordered a NodeMCUv2, will spend some time on it next week.  Preoccupied currently with a halloween lighting project: 30 milight bulbs and the theme "epileptics' worst nightmare" ;)  It's pretty ridiculous... I'll post a video and that code soon

sidoh, 2017-10-29:
Great!  No rush.  I might release 1.6.0 in the next day or so depending on how productive I'm feeling, but can definitely do a patch release with any additional improvements you come up with. :)

RvonMallesch, 2017-12-02:
Would you be so kind as to explain why so many multiple packets must be sent to globes? I find the number of 50 to be extremely strange, but necessary. Do the globes have to be sent the same packet a number of times to accept a command to accept it? Is it a question signal strength? Would you also be so kind as to explain the performance difference between the nRF24 and the LT8900. I purchased a LT8900 but no matter how hard I try, I can't get it to work? Does anyone have explicit instructions on the pin config for a nodemcuv2 and a LT8900? If it is more performant, then I would prefer to use it. I would dearly like to implement transitions between colours but I'm dubious that I can do that given the huge number of packets that need to be sent per command. I have roughly 75 bulbs. I don't want to waste my time attempting if it is not accomplishable. Thank you in advance for any insights you may provide.

Sincere regards,

Richard

sidoh, 2017-12-02:
Honestly this part of the stack isn't something I have a ton of experience with.  There are others who have contributed to the project that I think have a lot more expertise with radio stuff (eg @khmann).  But if I had to guess, the the reason so many repeats are needed is some combination of:

1. Lots of interference on 2.4 GHz
2. Bulbs scan across three different channels
3. Goofy radio handling on bulbs

espMH also sends copies of each packet on each of the three channels, which slows things down.  I've not experimented how necessary this is.

If you haven't already, read through the [Performance Tuning](https://github.com/sidoh/esp8266_milight_hub/wiki/Performance-Tuning) article on the wiki.

I use an nRF24.  The LT8900 has slightly higher packet throughput, but I experienced better stability with the nRF24.  Hookup instructions are in the [README](https://github.com/sidoh/esp8266_milight_hub#lt8900), but assuming you've seen that.  LT8900 also does not support passive listening in case that matters.

What kind of throughput are you looking for?  Sending 75 commands within a couple of seconds is likely gonna be really tough.

khmann, 2017-12-11:
The biggest reason the packets have to be sent so many times is the bulb's frequency hopping- you have no control over wether or not they're actually going to be listening when you start to TX.  My 2016-era 6W RGBW bulbs actually hop over (at least) 9 frequencies while listening for transmissions.  This is the basis for my "frequency hopping is stupid" argument because there's really no guarantee you actually stand a better chance of catching the bulb's attention jumping all over the place yourself.  Better to just stand still at let the nRF24 hardware hammer out packets at 1mbps.

re: updating 75 bulbs at once... maybe you can "cheat" by making pairing some extra common zones or a future controller will "interleave" commands to multiple devices

sidoh, 2017-12-12:
This makes a lot of sense.  I'll start a todo list on this ticket.

khmann, 2018-01-07:
I spent some time hooking the SPI on a number of receivers to look at frequency hopping timing.
current (2016+) RGBW strip controller FUT037/038 dwell 6ms through 9 frequencies
current (2017+) WW/CC strip controller FUT035 dwell 6ms through 6 frequencies
current (2014+) "Creative Controller" dwell 6ms through 6 frequencies
current (rev B) FUT015 bulb dwell 12ms through 3 frequencies
current LS2 Controller dwell 6ms through 6 frequencies

I've not yet taken apart my FUT105 or FUT014 (6W RGBW that also speak RGBCCT) bulbs but I assume the FUT105 will be the same as the LS2 (dual-protocol) and the FUT014 the same the FUT037 (tri-protocol).  I'm probably never going to buy an FUT039 (I hate RGB+CCT protocol 20) so that'll have to remain shrouded in mystery.

All of these products hop "downward" in frequency.  The "old" stuff hops a lot slower and is less consistent, so let's just ignore it for now.

So... everything modern thus far except for the "tri-protocol (A4/Bx/20)" RGBW visits a given frequency once every 36ms.

The iBox sends 1 packet about every 1ms (depending on packet length) and then hops, upward.  So, the reality if you're doing fixed frequency, is you need to transmit the same packet at least once every 6ms for a total of 36ms to guarantee reception for the "common" dual-protocol devices.  3-protocol would obviously need 50% longer.

If you extend this "fixed frequency" concept and start using a timer interrupt on the MCU, we can switch to using the nRF's hardware re-transmit which will get 4ms of continuous transmit (16-20 packets) if you follow the datasheet precisely (or more, which seems to work just fine).  I think the net is more retries with less CPU = overall win.

sidoh, 2018-01-08:
This is really good info.  Thanks very much for digging in, @khmann.  Super cool stuff!  :)

Sounds like the next step is to experiment with some changes that transmit on a fixed frequency, but leverage the nRF's native retransmit functionality.

Should be easy enough to skip the software retransmits.  I think you had some changes to the nRF handler with the relevant changes a while back.  Is it in here?

https://raw.githubusercontent.com/khmann/esp8266_milight_hub/master/lib/MiLight/PL1167_nRF24.cpp

closed 2019-09-01 by sidoh