sidoh/esp8266_milight_hub #120: Should "Clear Wifi Config" bring WIFIManager back?

speendo, 2017-10-04:
Hello!

I have my esp8266 milight hub working and connected to a wifi.

Now I want to connect it to a different wifi, while the old wifi is still present.

I thought the button "Clear Wifi Config" in the web interface would clear the old wifi settings so that I could enter new settings, once the esp8266 has rebooted.

Apparently this is not the case for me. When I press the "Clear Wifi Config" button, the esp seems to reboot, but it does not go to Access Point Mode. Instead it seems to reconnect to the old wifi.

Is this happening on purpose? Otherwise there might be a bug.

sidoh, 2017-10-04:
Yes, it's supposed to.  Looks like it's a common problem with the EPS8266 SDK:

https://github.com/esp8266/Arduino/issues/1494

Sounds like the SDK puts wifi credentials from RAM into flash when restarting, which completely undoes `ESP.eraseConfig()`.

I probably won't have a chance to look into this for a little while, but if you feel like looking into it yourself, the relevant code is [here](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/WebServer/MiLightHttpServer.cpp#L109) (search for `clear_wifi_config`).

Some things to try:

1. Use `ESP.reset()` instead of `ESP.restart()`
2. Force a crash ("best" way is probably by triggering a watchdog reset with `while(true);`)

If you're just looking for a quick solution, clearing flash with esptool.py will do the trick:

```
$ esptool.py --port /dev/cu.SLAB_USBtoUART erase_flash
```

speendo, 2017-10-05:
Thank you, sidoh!

I have to state that I have never done any ESP programming yet (some Arduino, though).

However, I looked at this issue and didn't really appreciate the two solutions you stated so much. It seemed to me, both of them are more hacks (provoking some behavior that would lead to the ESP forgetting the WiFi as a side effect) than actual solutions.

So I digged deep into the [ESP Arduino Core](https://github.com/esp8266/Arduino) and found the [disconnect method](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#disconnect) in the STA class.

So I added

    #include <ESP8266WiFi.h>

to the head of `MiLightHttpServer.cpp` and rewrote line 114 to this

    //ESP.eraseConfig();
    WiFi.disconnect(false);

It seems like this did the trick.

I can provide a fork for your project tomorrow (today I have go to sleep, I'm tired ;-)).

However, I am still not all too happy with that solution. I feel like a method to disconnect from the WiFi should be provided by [WiFiManager](https://github.com/tzapu/WiFiManager). It should not be necessary to import ESP8266WiFi in this project, as WiFiManager is a wrapper for that already...

I am interested in your opinion because as I said I have no experience with ESP8266 programming yet. However, I wonder if I should suggest to add a wrapper for `WiFi.disconnect(false)` in WiFiManager.

sidoh, 2017-10-05:
Interesting.  This sounds like exactly the right solution.  disconnect clears the credentials from flash and from the registers, which is exactly what's desired here.

I'm not at all worried about using ESP8266WiFi directly for this.  This is a great solution.  Thanks very much.

Happy to look at a PR!  Thanks for looking into this :)

speendo, 2017-10-06:
Thank you! After asking a lot of annoying questions about this project on reddit, I am happy to be finally able to contribute to your project a little bit!

Checking [WiFiManager](https://github.com/tzapu/WiFiManager), I found out that there is a wrapper method for `WiFi.disconnect()` already. Find it [here](https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L335). However, as far as I can see this method seems to be undocumented. I asked the developer(s) of WiFiManager to add this method to the documentation.

speendo, 2017-10-06:
Oh, my bad. Actually the method is documented in the [API reference](https://github.com/tzapu/WiFiManager/wiki/API-reference#void-wifimanagerresetsettings). I'd rather use this method instead of `WiFi.disconnect()`.

I mean, practically they do the same, but once you dedicated your project to a WiFi manager, it feels cleaner to stick to that and not using workarounds (as long as it is possible).

I will try and check this solution later that day and then prepare a pull request.

sidoh, 2017-10-06:
Sounds good 👍

closed 2017-10-29 by sidoh